neufieldrobotics/spinnaker_sdk_camera_driver #5: not achieving device's max fps

vannem95, 2018-07-31:
Hi @shahvi ,

Thanks for building this package. 
I am currently using a black fly s. The acquisition frame rate on spinview is 200 and the image resolution was 1440x1080.

I ran your launch file and tried to get 200 fps but only getting a max of 47fps(33fps in color). The image resolution i'm getting is 720p. When i change the soft_framerate to anything higher than 47 i get this error:
[image]

How do i get 200 fps and 1080p image using your package?

Thanks in advance.

shahvi, 2018-07-31:
In my experience the most likely cause of your error is that the camera doesn't have enough power. Are you using a laptop?

shahvi, 2018-07-31:
Also have you tried actually saving data at the 200 fps in Spinview?

vannem95, 2018-07-31:
I'm using a pc.

How do i save data using spinview?

shahvi, 2018-07-31:
The point is that Spinview doesn't really do much with the 200 fps. It can't show it to you on the screen at 200. 

I'd suggest trying auxiliary power to the camera or using a powered USB 3 hub. 

You could also try running the Spinnaker acquisition example to see if it your hardware can deliver the 200 fps. 

Also I think pushing 200 fps through ROS would be difficult since it all has to go through the network interface. In this package try making to_ros false and save BMP format directly. I have found that to be the quickest.

vannem95, 2018-07-31:
Used the to_ros and save features. Still getting the same error if i have the soft_framerate higher than 47

shahvi, 2018-07-31:
Just to try, if you have binning of 2 x 2 turned on.  Can you go higher?

Also, have you tried making the frame-rate 0.  What does it deliver then?

vannem95, 2018-07-31:
1. I couldn't increase the binning past 4. Even then the max fps is 45. 
2. When i put it to 0, it gives a runtime error

vannem95, 2018-08-01:
I came across [this](https://github.com/miks/spinnaker-fps-test) fps test package and tried it. I'm getting 226fps on this.

shahvi, 2018-08-01:
The package you are referring to is doing something similar to what Spinview does.  It is getting the pointer from the Spinnaker SDK and then dumping it.  So there is no real activity going on.

Frankly, if you are looking at the high frame rates I don't think sending it through ROS is the way to go, unless you are using nodelets which this code isn't doing currently.  

The last thing I can think of is to try the max_rate_save flag, it is multi threaded and should run faster, however all it can do for you is save files to disk, it won't publish the messages in ros.  Again recommend using the bmp format to save.  While you are running this, keep an eye on your processor usage and memory to see if it is filling up.

shahvi, 2018-08-01:
@arazdan would you have any recommendations for @vannem95

shahvi, 2018-08-03:
@vannem95 Any luck?

vannem95, 2018-08-03:
@shahvi I am trying to run the node with no publishing and just get_mat_images in the run_soft_trig loop and i still can't set an fps higher than 47.

closed 2019-02-25 by shahvi

ancabilloni, 2019-08-23:
Late reply but hope this may help the next person.

I also faced the issue of restricted frame rate in the 40~ fps. After playing around with setting, I figured the reason to this was due to the ExposureAuto is set to Continuous. When set the ExposureAuto to Off and set the exposure time value lower than the default value of Continuous, the camera can get faster frame rate. 

The default exposure time for Continuous mode can be found in SpinView, and assign `exposure_time` value other than 0 in acquisition launch file would automatically set the ExposureAuto to Off.

ZXY1231, 2024-09-04:
I turned off the Gain auto and the frame rate increased.