neufieldrobotics/spinnaker_sdk_camera_driver #7: fatal error: spinnaker_sdk_camera_driver/spinnaker_image_names.h: No such file or directory

Rockyzyk, 2018-08-13:
hi, Mr shahvi,

When I Install the driver and run "catkin_make",
The following error occurred:
fatal error: spinnaker_sdk_camera_driver/spinnaker_image_names.h: No such file or directory

How can I find the "spinnaker_image_names.h" file?
Would you please help me? Thanks

shahvi, 2018-08-13:
I have seen that sometimes when it generates tries to use the messages before generating then. Try running `catkin_make` a couple more times and it should compile.

Rockyzyk, 2018-08-13:
Thanks for help! I think the mean reason is my python module empy missing.
And I refer the following link and fix it! 
https://answers.ros.org/question/257331/python-module-empy-missing-tutorials/

run the cmd:
`pip uninstall em`
`pip install empy`

shahvi, 2018-08-14:
I haven't tried this code with the Ethernet cameras and I don't have one to test.  However, I can guide you through a few things to try.  It looks like the system is detecting the camera, so we might have a chance.

The first thing to try would be to comment out the ROS_ASSERT at [line 59 in camera.cpp](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/e8cb78a655d755e91d0250996a13dee49b23a234/src/camera.cpp#L59).  Then recompile by running `catkin_make`.

Rockyzyk, 2018-08-14:
@shahvi after try to comment out the ROS_ASSERT, the error is changed to follow:
[image]
The problem is same with https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/issues/5
so I changed the FPS to 4, to make sure. But the error still occurs.  Would you please check it? Thanks!

Rockyzyk, 2018-08-14:
By the way, I use the virtual box, and spinView is worked for showing images of camera.

shahvi, 2018-08-14:
Also have you tried this package http://wiki.ros.org/spinnaker_camera_driver

Rockyzyk, 2018-08-14:
I try this package and it succeeds! Thanks for helping!

closed 2018-08-15 by Rockyzyk