neufieldrobotics/spinnaker_sdk_camera_driver #8: Problem with reading list of cameras from test_params.yaml file

daobilige-su, 2018-10-03:
Hi,

Thanks for providing the driver. Very useful.

I had a problem with reading list of cameras from test_params.yaml. I followed you documentation and set the yaml file as follows:
`cam_ids: 
- 18285082
cam_aliases: cam0
master_cam: 
- 18285082
skip: 20
delay: 0`

Unfortunately, it is not working. When I examined it carefully, I found that the following line in capture.cpp is doing nothing.
`ROS_ASSERT_MSG(nh_pvt_.getParam("cam_ids", cam_id_vec),"If cam_aliases are provided, they should be the same number as cam_ids and should correspond in order!");`
This results in `cam_id_vec.size()` equals to 0.
I further found that `ROS_ASSERT_MSG(CONDITION, MSG)` is doing nothing. e.g. I put the following code in capture.cpp,
`ROS_ASSERT_MSG(true,"true !!");
ROS_ASSERT_MSG(false,"false !!");`
It does not give me any error during the launch.

To make things get working. I ended up changing the code into:
`//ROS_ASSERT_MSG(nh_pvt_.getParam("cam_ids", cam_id_vec),"If cam_aliases are provided, they should be the same number as cam_ids and should correspond in order!");
int cam_id_vec_int;
	nh_pvt_.getParam("cam_ids", cam_id_vec_int);
	cam_id_vec.push_back(cam_id_vec_int);`
...
`//ROS_ASSERT_MSG(nh_pvt_.getParam("master_cam", mcam_int),"master_cam is required!");
    nh_pvt_.getParam("master_cam", mcam_int);
    master_cam_id_=to_string(mcam_int);`
and the yaml file into:

`cam_ids: 18285082
cam_aliases: cam0
master_cam: 18285082
skip: 20
delay: 0`

However, the code only runs one camera in this way. 

So, I'd like to if the raw code is working properly on your machine? Can you suggest me what might be the problem on my side?

Thanks.

Su

shahvi, 2018-10-03:
Take a look at the multi_camera example showed in the params folder. Also this code is setup for hardware triggering of cameras, so you will need to use GPIO cables to hook up the cameras together.

shahvi, 2018-10-03:
see https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/master/README.md#multicamera-master-slave-setup

closed 2018-10-12 by shahvi