neufieldrobotics/spinnaker_sdk_camera_driver #9: published image resolution and framerate is lower than my camera config

daobilige-su, 2018-10-03:
Hi,

Just one more question.

I found that the published framerate (~20Hz) is lower than my camera's max frame rate (35Hz), even if I set **fps** and **soft_framerate** to be 35Hz in **acquisition.launch** file.

Also, the resolution (1224x1024) of published image is exactly half of the max resolution of my camera (2448x2048). 

Is there any way to tune it to be max? Thanks.

Cheers,
Su

daobilige-su, 2018-10-03:
Just a update on this issue, the full resolution is achieved by setting binning to 1. 

But I still cann't get 35 Hz full frame rate. After having full resolution, the frame rate actually decreased from 20 Hz to 8Hz.

shahvi, 2018-10-03:
That must be because of limits of the USB 3 bus on your hardware. Do you have a different machine you can try?

closed 2018-12-06 by shahvi