neufieldrobotics/spinnaker_sdk_camera_driver #12: Failed to run ROS node

mfehr, 2018-11-28:
Hi

I was trying to install the driver and ROS node for a Blackfly S 0.4MP (Mono) camera, but the ROS node keeps failing with an obscure error message:
```bash
process[acquisition_node-1]: started with pid [18446]
[ INFO] [1543406066.566989925]: [ OK ] USB memory: 1000 MB
20181128[ INFO] [1543406066.567830177]: *** PARAMETER SETTINGS ***
[ INFO] [1543406066.567851121]: ** Date = 20181128
[ INFO] [1543406066.568256178]:   Save path set via parameter to: /home/mfehr
[ INFO] [1543406066.568295076]:   Camera IDs:
[ INFO] [1543406066.568700824]:     18286011
[ INFO] [1543406066.569064274]:   Camera Aliases:
[ INFO] [1543406066.569085450]:     18286011 >> cam0
[ INFO] [1543406066.569885286]:   Unique time stamps for each camera: 0
[ INFO] [1543406066.570257958]:   color set to: 0
[ INFO] [1543406066.570630975]:   Exporting images to ROS: 1
[ INFO] [1543406066.570990423]:   Showing live images setting: 0
[ INFO] [1543406066.571327265]:   Showing grid-style live images setting: 0
[ INFO] [1543406066.571658724]:   Max Rate Save Mode: 0
[ INFO] [1543406066.571938841]:   Displaying timing details: 0
[ INFO] [1543406066.572210547]:   No. of images to skip set to: 20
[ INFO] [1543406066.572479189]:   Init sleep delays set to : 20 sec
[ WARN] [1543406066.572758069]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1543406066.573029590]:   'exp'=0, Setting autoexposure
[ INFO] [1543406066.573316150]:   Binning set to: 2
[ INFO] [1543406066.573605354]:   Using Software rate control, rate set to: 20
[ INFO] [1543406066.573913936]:   Saving images set to: 0
[ INFO] [1543406066.574420132]:   Camera coeffs not provided correctly, camera info messges will not be published.
[ INFO] [1543406066.574460727]: Creating system instance...
[ INFO] [1543406066.575081079]: Retreiving list of cameras...
[ INFO] [1543406068.494115871]: Numer of cameras found: 1
[ INFO] [1543406068.494220409]:  Cameras connected: 1
[ INFO] [1543406068.494301892]:   -18286011
[ INFO] [1543406068.499358346]: *** FLUSH SEQUENCE ***
[ INFO] [1543406068.499439460]: Initializing cameras...
[ INFO] [1543406068.841084258]: Deinitializing cameras...
[ INFO] [1543406068.846022099]: All cameras deinitialized.
[ INFO] [1543406068.846102392]: Initializing cameras...
[ INFO] [1543406069.091411564]: *** ACQUISITION ***
[FATAL] [1543406071.120527337]: Some exception occured. 
                                                         Exiting gracefully, 
  possible reason could be Camera Disconnection...
[ INFO] [1543406071.556492237]: Deinitializing cameras...
[ INFO] [1543406071.561361527]: All cameras deinitialized.
[ INFO] [1543406071.561383029]: Clearing camList...
[ INFO] [1543406071.561416981]: Releasing camera pointers...
[ INFO] [1543406071.561442979]: Releasing system instance...
terminate called after throwing an instance of 'std::bad_cast'
  what():  std::bad_cast
[acquisition_node-1] process has died [pid 18446, exit code -6, cmd /home/mfehr/spinnaker_ws/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/mfehr/.ros/log/a20abe0a-f2e9-11e8-b469-507b9dc5144a/acquisition_node-1.log].
log file: /home/mfehr/.ros/log/a20abe0a-f2e9-11e8-b469-507b9dc5144a/acquisition_node-1*.log
```

The camera application that came with the driver `spinview` works fine and successfully displays the camera image.

My System:
Ubuntu 18.04, ROS melodic, I installed the driver (followed the instructions in the README) and pulled current master from this repo.

I tried searching online and also ran gdb, but it's an error from inside the driver, not the ROS node itself. 

Could you give me a hint how I can fix this issue?

vik748, 2018-11-28:
First off, I have only tested the node on Ubuntu 16.04 so there could be something going on about that. 
I have observed this error when the USB connection is flaky and the camera doesnt get enough power. In your case are you running on a Virtual Machine by any chance?
I would suggest trying to reduce the frame rate and see if that helps.

mfehr, 2018-11-28:
@vik748 thank you for your prompt reply.
I'm not running a VM, I've also tried different USB ports on my laptop and docking station.
The camera works just fine with the spinview, it's just the rosnode that crashes. I also tried lowering the FPS to 10, no change.

shahvi, 2018-11-28:
@mfehr Another thing to try would be to remove that particular try / catch [block ](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/2743f06530a37f93a136f9959321c9935e315949/src/capture.cpp#L817) and seeing if that helps you understand the issue.

mfehr, 2018-11-28:
@vik748 thanks for the hint. I actually just remembered that I forgot to add this to the description, I changed the try-catch block to actually output the exception and the result was this error:
```
Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]
```
I did find this issue [here](https://github.com/ros-drivers/flir_camera_driver/issues/8). But I'm not sure how to use this, since I'm not triggering the camera.

Also the gdb bt is:
```
#0  0x00007ffff3b5ee97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff3b60801 in __GI_abort () at abort.c:79
#2  0x00007ffff3f9d8b7 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff3fa3a06 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff3fa2a69 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff3fa3418 in __gxx_personality_v0 () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff76cea9c in __libunwind_Unwind_RaiseException () at /usr/lib/x86_64-linux-gnu/libunwind.so.8
#7  0x00007ffff3fa3c67 in __cxa_throw () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff3fa29a2 in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff7972bfd in Spinnaker::CameraPtr::operator=(int) () at /home/mfehr/spinnaker_ws/devel/lib/libacquilib.so
#10 0x00007ffff7998ece in acquisition::Camera::~Camera() () at /home/mfehr/spinnaker_ws/devel/lib/libacquilib.so
#11 0x00007ffff798e87d in void std::_Destroy<acquisition::Camera>(acquisition::Camera*) ()
    at /home/mfehr/spinnaker_ws/devel/lib/libacquilib.so
#12 0x00007ffff798b7fb in void std::_Destroy_aux<false>::__destroy<acquisition::Camera*>(acquisition::Camera*, acquisition::Camera*) () at /home/mfehr/spinnaker_ws/devel/lib/libacquilib.so
#13 0x00007ffff7986df2 in void std::_Destroy<acquisition::Camera*>(acquisition::Camera*, acquisition::Camera*) ()
    at /home/mfehr/spinnaker_ws/devel/lib/libacquilib.so
#14 0x00007ffff79818df in void std::_Destroy<acquisition::Camera*, acquisition::Camera>(acquisition::Camera*, acquisition::Camera*, std::allocator<acquisition::Camera>&) () at /home/mfehr/spinnaker_ws/devel/lib/libacquilib.so
#15 0x00007ffff797a483 in std::vector<acquisition::Camera, std::allocator<acquisition::Camera> >::~vector() ()
    at /home/mfehr/spinnaker_ws/devel/lib/libacquilib.so
#16 0x00007ffff795b66c in acquisition::Capture::~Capture() () at /home/mfehr/spinnaker_ws/devel/lib/libacquilib.so
#17 0x000055555555cd49 in main ()
```

shahvi, 2018-11-28:
@mfehr we are in fact using a software trigger in our version. Have you added your camera serial number as the 'master' because that is how the camera is triggered? Also why is the initial sleep at 20 secs can you lower that to like 2.

shahvi, 2018-12-06:
@mfehr  did you have any luck with your issue?

mfehr, 2018-12-07:
@shahvi  I tried with a lower skip and I added the serial number in master, no luck. I'm probably configuring sth wrong. 

I ended up using [this](https://github.com/ethz-asl/flir_camera_driver) ROS node for now. But I might try again with this one, since the other one is quite heavy in terms of dependencies that we don't really need.

Since this is probably a very specific issue with how I configure things, I will close this issue and reopen it if I end up trying again. Thanks a lot for your support.

closed 2018-12-07 by mfehr

alankwok97, 2019-07-19:
@mfehr Can I have some instructions on installing the package referenced in your comment as the README file doesn't seem very useful (or it could be me misunderstanding some of the concept in the README file).

Many thanks.

mfehr, 2019-07-19:
If I remember correctly it's mostly installing the Spinnaker sdk and then building the catkin package. It depends on some msgs from the pointgrey_camera_driver package so that should be present in the workspace or system as well.

alankwok97, 2019-07-23:
@mfehr thanks for your reply. I keep getting the following errors and I don't know how to solve it.

-- ~~  traversing 2 packages in topological order:
-- ~~  - flir_camera_driver (metapackage)
-- ~~  - spinnaker_camera_driver
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'flir_camera_driver'
-- ==> add_subdirectory(flir_camera_driver-kinetic-devel/flir_camera_driver)
-- +++ processing catkin package: 'spinnaker_camera_driver'
-- ==> add_subdirectory(flir_camera_driver-kinetic-devel/spinnaker_camera_driver)
-- Could NOT find image_exposure_msgs (missing: image_exposure_msgs_DIR)
-- Could not find the required component 'image_exposure_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "image_exposure_msgs" with any of the following names:

    image_exposure_msgsConfig.cmake
    image_exposure_msgs-config.cmake

  Add the installation prefix of "image_exposure_msgs" to CMAKE_PREFIX_PATH
  or set "image_exposure_msgs_DIR" to a directory containing one of the above
  files.  If "image_exposure_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  flir_camera_driver-kinetic-devel/spinnaker_camera_driver/CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/dell8116/spinnaker_ws2/build/CMakeFiles/CMakeOutput.log".
See also "/home/dell8116/spinnaker_ws2/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Many thanks.

longningbo, 2020-07-29:
> @mfehr thanks for your reply. I keep getting the following errors and I don't know how to solve it.
>
> -- ~~ traversing 2 packages in topological order:
> -- ~~ - flir_camera_driver (metapackage)
> -- ~~ [...]

> @mfehr thanks for your reply. I keep getting the following errors and I don't know how to solve it.
>
> -- ~~ traversing 2 packages in topological order:
> -- ~~ - flir_camera_driver (metapackage)
> -- ~~ [...]

Hi,I also encountered this problem, and I use Ubuntu 16.04, So, how did you solve this problem？
THank you very much !

vik748, 2020-07-30:
Can you try installing `sudo apt install ros-kinetic-image-exposure-msgs `