neufieldrobotics/spinnaker_sdk_camera_driver #15: Fix the '' terminate called after throwing an instance of 'std::bad_cast' " error on exit.

shahvi, 2018-12-06:


4artit, 2019-03-04:
when will this bug be fixed? I am trying to use 4 cams using hardware trigger but I usually see this error.

shahvi, 2019-03-04:
This bug doesnt seem to affect functionality so its not high priority. Are you able to use the 4 cameras together?

4artit, 2019-03-05:
I could not say I can do that. Sometimes It works but I should try several times before. I am using official hardware trigger circuit. 4 cams can not initialize at once.

shahvi, 2019-03-05:
Can you paste the output so we can look at it.

4artit, 2019-03-05:
One is when it works well and the other is not.
[image]
[image]

shahvi, 2019-03-05:
This could he related to cameras not getting enough power. How are you powering them? Just plugging into IUB ports or powered USB hub or external power?

4artit, 2019-03-06:
I connected master(primary) camera directly into computer and the others are connected to USB hub(this one do not have external power). I ordered powered usb hub. So when after I get that usb hub, I will test again.
Thanks for your help. I will upload that result too.

4artit, 2019-03-07:
When I use powered usb hub, another error is occurred.
[image]

shahvi, 2019-03-07:
Not a lot of information to go by there. Have you increased your USB buffer as suggested during installation.

4artit, 2019-03-07:
Thanks it fixed! previously I set use buffer as 1000 but now I increase 1500. It works very well.
I followed this instruction.
[https://importgeek.wordpress.com/2017/02/26/increase-usbfs-memory-limit-in-ubuntu/](https://importgeek.wordpress.com/2017/02/26/increase-usbfs-memory-limit-in-ubuntu/)

closed 2020-06-29 by shahvi