neufieldrobotics/spinnaker_sdk_camera_driver #18: image_Tranport_test branch: multiple cams trigger, error on timeout

mithunvarma, 2018-12-21:
code works for single camera

when multiple cameras are plugged in, following error occurs

> terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]

**gdb output:**

[New Thread 0x7fff84ff9700 (LWP 21638)]
[New Thread 0x7fff667fc700 (LWP 21639)]
[Thread 0x7fff667fc700 (LWP 21639) exited]
[Thread 0x7fff84ff9700 (LWP 21638) exited]
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]

Thread 1 "acquisition_nod" received signal SIGABRT, Aborted.
0x00007ffff3c29428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) where
#0  0x00007ffff3c29428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff3c2b02a in __GI_abort () at abort.c:89
#2  0x00007ffff404d84d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff404b6b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff404b701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff404b919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff4db77be in ?? () from /usr/lib/libSpinnaker.so.1
#7  0x00007ffff4dae655 in ?? () from /usr/lib/libSpinnaker.so.1
#8  0x00007ffff4de5816 in ?? () from /usr/lib/libSpinnaker.so.1
#9  0x00007ffff4ea8b61 in Spinnaker::CameraBase::GetNextImage(unsigned long, unsigned long) () from /usr/lib/libSpinnaker.so.1
#10 0x00007ffff7b90813 in acquisition::Camera::grab_frame() () from /home/mithun/apps/ws_catkin/devel/lib/libacquilib.so
#11 0x00007ffff7b91477 in acquisition::Camera::grab_mat_frame() () from /home/mithun/apps/ws_catkin/devel/lib/libacquilib.so
#12 0x00007ffff7b4981d in acquisition::Capture::get_mat_images() () from /home/mithun/apps/ws_catkin/devel/lib/libacquilib.so
#13 0x00007ffff7b4a05a in acquisition::Capture::run_soft_trig() () from /home/mithun/apps/ws_catkin/devel/lib/libacquilib.so
#14 0x00007ffff7b4e3c2 in acquisition::Capture::run() () from /home/mithun/apps/ws_catkin/devel/lib/libacquilib.so
#15 0x00000000004073c2 in main ()
(gdb) 

Tried with Powered USB hub, so it's not power related error either.

AcquisitionMultipleCamera binary, can save images from both the cameras.

The GetNextImage error occurs for the cam1 (for master cam, it is successful )before entering the loop in run_soft_trig()

closed 2018-12-21 by mithunvarma

angegu, 2019-03-25:
@mithunvarma Hi, what was your solution to this?

themightyoarfish, 2020-02-12:
Would also be interested in the resolution.

irispete, 2020-04-08:
Was there a solution for this problem? I am encountering this same error when acquiring from multiple cameras

shahvi, 2020-04-08:
@irispete please start a new issue and fill out the details in the Bug Report so we can advise accordingly.

themightyoarfish, 2020-04-08:
Don't know if it's applicable in this case, but this also happens when the available bandwidth is not enough. So if multiple cameras go over one 1gbit connection, you would need to set a non-debayered pixel format and debayer on the host, or retrieve a smaller sensor region.

themightyoarfish, 2020-04-08:
Generally I've found this issue happens a lot when multiple cams are involved, even when they don't share a link, and one always has to permit retries, a timeout in GetNextImag() doesn't seem ro always work.

irispete, 2020-04-08:
I created a new issue:
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/issues/75

irispete, 2020-04-08:
@themightyoarfish  forgive my ignorance, how do you permit retries?

themightyoarfish, 2020-04-09:
By catching the exception and also calling GetImageStatus() and IsIncomplete() on the resulting frame. Sometimes, GetNextImage() succeeds but the frame is not useable. 

In any event, I've had good experience with the FLIR support, though we are a business, so I can't say if they are this responsive for private individuals.