neufieldrobotics/spinnaker_sdk_camera_driver #20: Fail to read single camera info from test_params.yaml

BobbleLaw, 2019-02-25:
Hi, 

The camera I am using is GS3-U3-41C6C, under Ubuntu 16.04 LTS, ROS Kinetic. 
I modified the test_params.ymal as the tutorial suggested, but encountered two problems.

1. cam_ids and cam_aliasis can't be read into the process.

process[acquisition_node-1]: started with pid [8704]
[ INFO] [1551087121.707963015]: [ OK ] USB memory: 1000 MB
[ INFO] [1551087121.708028347]: *** PARAMETER SETTINGS ***
[ INFO] [1551087121.708038749]: ** Date = 20190225
[ INFO] [1551087121.708396940]:   Save path set via parameter to: /home/robotarm
[ INFO] [1551087121.708416384]:   Camera IDs:
[ INFO] [1551087121.708731826]:   No camera aliases provided. Camera IDs will be used as names.
[ INFO] [1551087121.709056883]:   Unique time stamps for each camera: false
[ INFO] [1551087121.709383283]:   color set to: false
[ INFO] [1551087121.709693204]:   Exporting images to ROS: true

2. always crush at *** FLUSH SEQUENCE *** stage

[ INFO] [1551087015.778780291]: Creating system instance...
[ INFO] [1551087015.779031366]: Retreiving list of cameras...
[ INFO] [1551087016.720772197]: Numer of cameras found: 1
[ INFO] [1551087016.720851572]:  Cameras connected: 1
[ INFO] [1551087016.720918757]:   -16400214
[ INFO] [1551087016.723558542]: *** FLUSH SEQUENCE ***
[ INFO] [1551087016.723630000]: Initializing cameras...
[acquisition_node-1] process has died [pid 8550, exit code -11, cmd /home/robotarm/catkin_ws/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/robotarm/.ros/log/51d851bc-38db-11e9-8ecd-e0d55e613c3c/acquisition_node-1.log].
log file: /home/robotarm/.ros/log/51d851bc-38db-11e9-8ecd-e0d55e613c3c/acquisition_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done.

I have tried to modify the source code as in Problem #8 , and still couldn't solve it.
Any help would be appreciated.

shahvi, 2019-02-25:
Is this a USB camera?  Are you able to run it using SpinView?

BobbleLaw, 2019-02-25:
Yes, this is a USB camera, and I can run it using SpinView, and it also work well with my other codes (not related to ROS).

shahvi, 2019-02-25:
Can you paste or attach a copy of your `test_params.yaml`

BobbleLaw, 2019-02-25:
case 1:

cam_ids:
- 16400214
cam_aliases:
- cam0
master_cam: 16400214
skip: 20
delay: 0

process ends at *** PARAMETER SETTINGS ***. 

process[acquisition_node-1]: started with pid [26993]
[ INFO] [1551096095.203821717]: [ OK ] USB memory: 1000 MB
[ INFO] [1551096095.203881366]: *** PARAMETER SETTINGS ***
[ INFO] [1551096095.203892166]: ** Date = 20190225
[ INFO] [1551096095.204209130]:   Save path set via parameter to: /home/robotarm
[ INFO] [1551096095.204234059]:   Camera IDs:
[ INFO] [1551096095.204508368]:   Camera Aliases:
[acquisition_node-1] process has died [pid 26993, exit code -11, cmd /home/robotarm/catkin_ws/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/robotarm/.ros/log/51d851bc-38db-11e9-8ecd-e0d55e613c3c/acquisition_node-1.log].
log file: /home/robotarm/.ros/log/51d851bc-38db-11e9-8ecd-e0d55e613c3c/acquisition_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Case 2:

cam_ids: 16400214
cam_aliases: cam0
master_cam: 16400214
skip: 20
delay: 0

process end at *** FLUSH SEQUENCE ***

...
[ INFO] [1551103305.179697840]: Creating system instance...
[ INFO] [1551103305.179969102]: Retreiving list of cameras...
[ INFO] [1551103306.124703100]: Numer of cameras found: 1
[ INFO] [1551103306.124781966]:  Cameras connected: 1
[ INFO] [1551103306.124852922]:   -16400214
[ INFO] [1551103306.127361662]: *** FLUSH SEQUENCE ***
[ INFO] [1551103306.127433808]: Initializing cameras...
[acquisition_node-1] process has died [pid 29347, exit code -11, cmd /home/robotarm/catkin_ws/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/robotarm/.ros/log/51d851bc-38db-11e9-8ecd-e0d55e613c3c/acquisition_node-1.log].
log file: /home/robotarm/.ros/log/51d851bc-38db-11e9-8ecd-e0d55e613c3c/acquisition_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

sinanmut, 2019-02-27:
Hi,
I also have the same problem with my USB camera. It's model: Blackfly S BFS-U3-16S2M
Did you find any solution for it?
Best,

shahvi, 2019-06-22:
Guys did you have any luck or have any update on the issue?

closed 2019-06-26 by shahvi