neufieldrobotics/spinnaker_sdk_camera_driver #22: how to set the parameters of the Blackfly S camera ?

passion3394, 2019-03-15:
such as White Balence

vik748, 2019-03-15:
We haven't implemented this in our driver.  You should be able to implement it easily by using modifying the section at https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/fd3d35e43ad590c4ef42937083317c56d5459386/src/capture.cpp#L489

You should be able to use either `cams[i].setIntValue` or `cams[i].setEnumValue` to set the white balance.  A good way to check the values and variable name it will accept is by opening the camera in SpinView and looking at the White Balance options there.

If you do implement this, please submit a pull request for the others.

passion3394, 2019-03-17:
@vik748 thanks for your suggestion, I will consider it.

jianpursuit, 2019-06-14:
> @vik748 thanks for your suggestion, I will consider it.

Have you implemented this setting? I also want to set the parameters of the Blackfly S camera.

shahvi, 2019-06-14:
In the latest version of the code, we have implemented settings for Target grey value which can be set using Dynamic reconfigure. White balance should be very similar to that.

closed 2019-06-26 by shahvi