neufieldrobotics/spinnaker_sdk_camera_driver #23: met a problem

passion3394, 2019-05-05:
PARAMETERS
 * /acquisition_node/binning: 2
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [18285550]
 * /acquisition_node/color: True
 * /acquisition_node/delay: 0
 * /acquisition_node/exp: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 18285550
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: png
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 10
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    acquisition_node (spinnaker_sdk_camera_driver/acquisition_node)

auto-starting new master
process[master]: started with pid [7179]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to de373cdc-6f35-11e9-8464-9c305be96d8d
process[rosout-1]: started with pid [7193]
started core service [/rosout]
process[acquisition_node-2]: started with pid [7209]
[ INFO] [1557061277.057179656]: [ OK ] USB memory: 1000 MB
[ INFO] [1557061277.057291773]: *** PARAMETER SETTINGS ***
[ INFO] [1557061277.057315139]: ** Date = 20190505
[ INFO] [1557061277.057781627]:   Save path set via parameter to: /home/tarena
[ INFO] [1557061277.057815296]:   Camera IDs:
[ INFO] [1557061277.058256414]:     18285550
[ INFO] [1557061277.058709557]:   Camera Aliases:
[ INFO] [1557061277.058726610]:     18285550 >> cam0
[ INFO] [1557061277.059670647]:   Unique time stamps for each camera: false
[ INFO] [1557061277.060148378]:   color set to: true
[ INFO] [1557061277.060596604]:   Exporting images to ROS: true
[ INFO] [1557061277.061059862]:   Showing live images setting: false
[ INFO] [1557061277.061548876]:   Showing grid-style live images setting: false
[ INFO] [1557061277.061953093]:   Max Rate Save Mode: false
[ INFO] [1557061277.062347908]:   Displaying timing details: false
[ INFO] [1557061277.062833572]:   No. of images to skip set to: 20
[ INFO] [1557061277.063281499]:   Init sleep delays set to : 0.00 sec
[ WARN] [1557061277.063725959]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1557061277.064149388]:   'exp'=0, Setting autoexposure
[ INFO] [1557061277.064549205]:   Binning set to: 2
[ INFO] [1557061277.064986304]:   Using Software rate control, rate set to: 10
[ INFO] [1557061277.065412179]:   Saving images set to: 0
[ INFO] [1557061277.067137108]:   Camera coeffs not provided correctly, camera info messges will not be published.
[ INFO] [1557061277.067168683]: Creating system instance...
[ INFO] [1557061277.068513767]: Retreiving list of cameras...
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Parameter is already set. argument was expected to be NULL. [-1009]
[acquisition_node-2] process has died [pid 7209, exit code -6, cmd /home/tarena/spinnaker_ws/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/tarena/.ros/log/de373cdc-6f35-11e9-8464-9c305be96d8d/acquisition_node-2.log].
log file: /home/tarena/.ros/log/de373cdc-6f35-11e9-8464-9c305be96d8d/acquisition_node-2*.log

vik748, 2019-05-05:
What camera are you using?  Can you see images via Spinview?

passion3394, 2019-05-06:
@vik748 yes, the spinview software could read the camera and display the real-time image info. The camera is Blackfly S BFS-U3-51S5C.

[image]

vik748, 2019-05-06:
We have used that camera model extensively.  I don't have any ideas as to what the issue might be.  
The error code is from spinnaker, it relates to settings the camera parameters which is done around:
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/fd3d35e43ad590c4ef42937083317c56d5459386/src/capture.cpp#L491

Once thing to try would be setting the camera to factory defaults incase some setting in it is causing a conflict.

passion3394, 2019-05-06:
thanks for you tip, I will try to find out the reason

passion3394, 2019-05-06:
I have tried some methods:
(1) use the default yaml, no use
(2) catkin_make the ros package, it works

I don't know why.

passion3394, 2019-05-06:
but ,there are some warnings:

[ WARN] [1557114815.016727545]: Image incomplete with image status 9!
[ WARN] [1557114815.076058986]: Frames are being skipped!
[ WARN] [1557114855.005014838]: Image incomplete with image status 9!
[ WARN] [1557114855.087386516]: Frames are being skipped!
[ WARN] [1557114923.290210464]: Image incomplete with image status 9!
[ WARN] [1557114923.348592768]: Frames are being skipped!

passion3394, 2019-05-06:
after some minutes, an error occurs:

[FATAL] [1557115060.136949739]: Excption: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]

shahvi, 2019-05-06:
Before catkin_make how were you building it?

shahvi, 2019-05-06:
I think the incomplete message might be related to USB throughput. A good
test would be to reduce the framerate and see if this becomes stable.

shahvi, 2019-06-22:
were you able to resolve your issue?

closed 2019-06-26 by shahvi

andreamassetti, 2025-07-25:
Having the same problem regarding the status 9 error, using only PySpin and spinnaker for arm version 4.0.0, writing here to see if anyone has solved this