neufieldrobotics/spinnaker_sdk_camera_driver #24: Node crashes in Jetson tx2

cpaniagua1, 2019-05-22:
Hello,
I have a Blackfly S USB 3 camera connected to a Jetson Tx2. I'm trying to use this node but it crashes without information about the reason. Using the spinview application I can see the image. 
Could anybody help me?

nvidia@tegra-ubuntu:~$ roslaunch spinnaker_sdk_camera_driver acquisition.launch 
... logging to /home/nvidia/.ros/log/af4f800e-7c7e-11e9-af01-00044bc57968/roslaunch-tegra-ubuntu-3080.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:46778/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 2
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [18382800]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 0
 * /acquisition_node/exp: 0
 * /acquisition_node/fps: 30
 * /acquisition_node/frames: 3400
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 18382800
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 20
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    acquisition_node (spinnaker_sdk_camera_driver/acquisition_node)

auto-starting new master
process[master]: started with pid [3090]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to af4f800e-7c7e-11e9-af01-00044bc57968
process[rosout-1]: started with pid [3106]
started core service [/rosout]
process[acquisition_node-2]: started with pid [3110]
[acquisition_node-2] process has died [pid 3110, exit code -6, cmd /home/nvidia/worspaces/crv_ws/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/nvidia/.ros/log/af4f800e-7c7e-11e9-af01-00044bc57968/acquisition_node-2.log].
log file: /home/nvidia/.ros/log/af4f800e-7c7e-11e9-af01-00044bc57968/acquisition_node-2*.log

vik748, 2019-05-22:
Try removing libunwind from CMakelist

cpaniagua1, 2019-05-22:
Thank you! It worked!

closed 2019-05-24 by cpaniagua1