neufieldrobotics/spinnaker_sdk_camera_driver #26: camera_info & camera/set_camera_info

picatostas, 2019-05-24:
Is there any idea to implement these things ?
When listing topics, I can see that Camera_info has been initialized but doesnt show any info when using rostopic echo

vik748, 2019-05-24:
You will need to specify the camera matrix and distortion coefficients.  You can add that to the config file and it will be published.

picatostas, 2019-05-26:
Hi,
I was asking this because I wanted to use http://wiki.ros.org/camera_calibration for calibrating the camera but when running it, the service set_camera_info was not detected. In the end I did it "manually". Once I have the camera matrix & distortion coeficients where should I put them in the config file ?

vik748, 2019-05-27:
Checkout the multicam_example at https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/fd3d35e43ad590c4ef42937083317c56d5459386/params/multi-cam_example.yaml#L26

closed 2019-06-26 by shahvi