neufieldrobotics/spinnaker_sdk_camera_driver #27: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]

cpaniagua1, 2019-05-27:
Hi!
I'm using a GigE BSF camera and get this error:

[FATAL] [1558940122.459818739]: Excption: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]

I can see the image for a few seconds but then the error comes. 

Could somebody help me?

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 2
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [18434477]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 0
 * /acquisition_node/exp: 0
 * /acquisition_node/fps: 10
 * /acquisition_node/frames: 3400
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 18434477
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 2
 * /acquisition_node/soft_framerate: 10
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    acquisition_node (spinnaker_sdk_camera_driver/acquisition_node)

auto-starting new master
process[master]: started with pid [8248]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4f9f595a-804c-11e9-8439-e4e7491b4d31
process[rosout-1]: started with pid [8261]
started core service [/rosout]
process[acquisition_node-2]: started with pid [8278]
log4cxx: Could not read configuration file [/home/cpaniagua/workspaces/crv_ws/src/spinnaker_sdk_camera_driver/cfg/atd_console.conf].
[ INFO] [1558940086.138683281]: [ OK ] USB memory: 1000 MB
[ INFO] [1558940086.138787367]: *** PARAMETER SETTINGS ***
[ INFO] [1558940086.138810996]: ** Date = 20190527
[ INFO] [1558940086.139269778]:   Save path set via parameter to: /home/cpaniagua
[ INFO] [1558940086.139311425]:   Camera IDs:
[ INFO] [1558940086.139714906]:     18434477
[ INFO] [1558940086.140151610]:   Camera Aliases:
[ INFO] [1558940086.140177757]:     18434477 >> cam0
[ INFO] [1558940086.140980758]:   Unique time stamps for each camera: false
[ INFO] [1558940086.141366739]:   color set to: false
[ INFO] [1558940086.141805404]:   Exporting images to ROS: true
[ INFO] [1558940086.142242177]:   Showing live images setting: false
[ INFO] [1558940086.142679324]:   Showing grid-style live images setting: false
[ INFO] [1558940086.143168229]:   Max Rate Save Mode: false
[ INFO] [1558940086.143602762]:   Displaying timing details: false
[ INFO] [1558940086.143983993]:   No. of images to skip set to: 2
[ INFO] [1558940086.144361464]:   Init sleep delays set to : 0.00 sec
[ INFO] [1558940086.144787032]:   Master cam fps set to : 10.00
[ INFO] [1558940086.145149309]:   'exp'=0, Setting autoexposure
[ INFO] [1558940086.145520790]:   Binning set to: 2
[ INFO] [1558940086.145921243]:   Using Software rate control, rate set to: 10
[ INFO] [1558940086.146331736]:   Saving images set to: 0
[ INFO] [1558940086.147707064]:   Camera coeffs not provided correctly, camera info messges will not be published.
[ INFO] [1558940086.147741762]: Creating system instance...
[ INFO] [1558940086.149169264]: Retreiving list of cameras...
[ INFO] [1558940087.619172667]: Numer of cameras found: 1
[ INFO] [1558940087.619209955]:  Cameras connected: 1
[ INFO] [1558940087.619242041]:   -18434477
[ INFO] [1558940087.621849166]: *** FLUSH SEQUENCE ***
[ INFO] [1558940087.621879806]: Initializing cameras...
[ INFO] [1558940088.928264000]: Deinitializing cameras...
[ INFO] [1558940089.138336741]: All cameras deinitialized.
[ INFO] [1558940089.138539016]: Initializing cameras...
[ INFO] [1558940090.279638149]: *** ACQUISITION ***
[FATAL] [1558940122.459818739]: Excption: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]
[ INFO] [1558940122.540482320]: Deinitializing cameras...
[ INFO] [1558940122.748654234]: All cameras deinitialized.
[ INFO] [1558940122.748729271]: Clearing camList...
[ INFO] [1558940122.748754686]: Releasing camera pointers...
[ INFO] [1558940122.748779764]: Releasing system instance...

shahvi, 2019-06-22:
any update on the issue?  Unfortunately we don't have any Gig-e cameras in our lab, so we can't test this.

closed 2019-06-26 by shahvi

ParthAD3, 2021-05-19:
Hi,

Did you find any solution?

vik748, 2021-05-19:
@ParthAD3 given that this is a 2yr old issue, pls open a new issue and fill out the bug report template, so we have all the information to look at.