neufieldrobotics/spinnaker_sdk_camera_driver #32: Flip image

picatostas, 2019-06-12:
In which part of the node is it possible to flip the image both vertically and horizontally ? In spinview its ReverseX and ReverseY but I dont really know which part of the API corresponds to those flags and in which part of the adquisition procedure should I change it

shahvi, 2019-06-12:
All the camera settings are done at https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/e1ff42b4e7777a8654d8d1b8501d2e1991cce7d2/src/capture.cpp#L476

We should have all the required enumerations in the camera objects.  If this requires a boolean, you should be able to do similar to:
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/e1ff42b4e7777a8654d8d1b8501d2e1991cce7d2/src/capture.cpp#L518

If you can get the desired behavior please consider submitting a pull request.

picatostas, 2019-06-12:
thanks a lot, I will let you know

mithunvarma, 2019-06-20:
@picatostas a New Feature to flip image both horizontally and vertically has been implemented in pr #41 and merged to dev branch in commit f16e6c0e705df44c80413786a5d890bccbe3286d 

Take a look

closed 2019-06-20 by mithunvarma