neufieldrobotics/spinnaker_sdk_camera_driver #44: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]

liuxiangdi, 2019-07-09:
Hi, I use two camera (BFS-U3-16S2C-CS), and I connect this two camera to my laptop's two different USB3 port. Then I follow your document rewrite the "test.yaml"
`
cam_ids:
- 18575250
- 18012381
cam_aliases:
- cam0
- cam1
master_cam: 18575250
skip: 20
delay: 1.0
` 
and I get this Error
`
 INFO] [1562686267.400490846]: [ OK ] USB memory: 1500 MB
[ INFO] [1562686267.400648559]: *** PARAMETER SETTINGS ***
[ INFO] [1562686267.400675644]: ** Date = 20190709
[ INFO] [1562686267.401506943]:   Save path set via parameter to: /home/lxd
[ INFO] [1562686267.401574484]:   Camera IDs:
[ INFO] [1562686267.402252559]:     18575250
[ INFO] [1562686267.402289423]:     18012381
[ INFO] [1562686267.402929153]:   Camera Aliases:
[ INFO] [1562686267.402959661]:     18575250 >> cam0
[ INFO] [1562686267.402992996]:     18012381 >> cam1
[ INFO] [1562686267.404053349]:   Unique time stamps for each camera: false
[ INFO] [1562686267.404784677]:   color set to: false
[ INFO] [1562686267.405329043]:   Exporting images to ROS: true
[ INFO] [1562686267.405853776]:   Showing live images setting: false
[ INFO] [1562686267.406456399]:   Showing grid-style live images setting: false
[ INFO] [1562686267.407043810]:   Max Rate Save Mode: false
[ INFO] [1562686267.407514213]:   Displaying timing details: false
[ INFO] [1562686267.408018557]:   No. of images to skip set to: 20
[ INFO] [1562686267.408574815]:   Init sleep delays set to : 1.00 sec
[ WARN] [1562686267.409098651]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1562686267.409576668]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1562686267.410089067]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1562686267.410597933]:   Binning set to: 2
[ INFO] [1562686267.411152542]:   Using Software rate control, rate set to: 20
[ INFO] [1562686267.412026190]:   Saving images set to: 0
[ INFO] [1562686267.412867108]:   Camera Intrinsic Paramters:
[ INFO] [1562686267.413047461]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1562686267.413095712]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1562686267.414018941]:   Camera Distortion Paramters:
[ INFO] [1562686267.414089253]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1562686267.414128283]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1562686267.415671602]:   Camera Rectification Paramters:
[ INFO] [1562686267.416153002]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1562686267.416241537]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1562686267.418700859]:   Camera Projection Paramters:
[ INFO] [1562686267.418818628]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1562686267.418866921]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1562686267.418891334]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1562686267.418948098]: Creating system instance...
[ INFO] [1562686267.420515779]: Retreiving list of cameras...
[ INFO] [1562686268.391111021]: Numer of cameras found: 2
[ INFO] [1562686268.391183358]:  Cameras connected: 2
[ INFO] [1562686268.391275448]:   -18575250
[ INFO] [1562686268.391328738]:   -18012381
[ INFO] [1562686268.469329701]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1562686268.470869437]: *** FLUSH SEQUENCE ***
[ INFO] [1562686268.470903906]: Initializing cameras...
[ INFO] [1562686269.084765526]: Deinitializing cameras...
[ INFO] [1562686269.094301500]: All cameras deinitialized.
[ INFO] [1562686271.094662583]: Initializing cameras...
[ INFO] [1562686271.662068798]: *** ACQUISITION ***
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]
[acquisition_node-2] process has died [pid 12802, exit code -6, cmd /home/lxd/rosworkspace/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/lxd/.ros/log/91b2573c-a25e-11e9-a6d0-14abc578a8ca/acquisition_node-2.log].
log file: /home/lxd/.ros/log/91b2573c-a25e-11e9-a6d0-14abc578a8ca/acquisition_node-2*.log
`
I want to use this two camera for stereo camera, and I can use this package to get image from camera respectively now.
So what problems I met, must i follow the document "Muticamera Master-Slave Setup"?

shahvi, 2019-07-09:
Yes, for multicam this package has to be setup as Master-slave,  otherwise there is no way to guarantee the sync between cameras.

samanfahandej, 2019-07-10:
Asked about this error from FLIR support team, and this is their answer:
error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]
"This error occurs when trying to retrieve an image from the buffer(RAM) while the buffer is empty."

liuxiangdi, 2019-07-10:
> Asked about this error from FLIR support team, and this is their answer:
> error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]
> "This error occurs when trying to retrieve an image from the buffer(RAM) while the buffer is empty."

thanks a lot

liuxiangdi, 2019-07-10:
> Yes, for multicam this package has to be setup as Master-slave, otherwise there is no way to guarantee the sync between cameras.

thanks for your reply.
I try to use spinview for this two cameras at the same time, but I failed. If I do not use this package, writting code with official SDK, Is using Master-slave mode still the only way for stereo camera?

shahvi, 2019-07-10:
This code uses a software trigger for the cameras, so technically if you wanted you could modify this driver to do a software trigger on both your cameras.  You'd have you dive in a bit to see whatall might to be changed. A couple lines to help might be:

https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/15803f032859341b2d1740dbce5300726b2ac8b8/src/capture.cpp#L894

and 

https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/15803f032859341b2d1740dbce5300726b2ac8b8/src/capture.cpp#L648

liuxiangdi, 2019-07-10:
I tried another package[ https://github.com/ros-drivers/flir_camera_driver ](url) and got two videostream at same time (with out adapter).
Thanks again for giving me suggestion and that help a lot  :)

alankwok97, 2019-07-18:
@liuxiangdi I am new to ROS language. How did you install the flir_camera_driver because I can't seem to find the instructions in the README file.

Many thanks.

atyshka, 2019-08-13:
@shahvi I know you do not have a GigE camera to test with, but could we at least add a try-catch for this? Spinview gracefully catches incomplete frames and ignores them and we should expect the same behavior here.

shahvi, 2019-08-13:
@atyshka could you please submit a PR.

atyshka, 2019-08-14:
So... Tried adding a try-catch to this but there's another problem: after this error occurs, video cuts out for about a second. This isn't a good fix then and we'll have to get to the root of the problem. Spinview does not have this issue.

harsco-jfernandez, 2020-05-25:
This worked for me.

if ( cam_img->GetImageStatus() == IMAGE_NO_ERROR )

orange2xiaoyu, 2021-07-01:
Hi, I use one camera (BFS-PGE-31S4C-C), the same error occurred.

**System Description**
Camera Name and Model: BFS-PGE-31S4C-C
Operating System: Ubuntu 18.04
Spinnaker Version: 2.2.0.48
ROS Version: Melodic

**Computer details**
Nvidia Jetson Xavier Development Kit
Processor: ARM 64 bit (v8.2)
RAM: 32 GB

**Do the cameras work with SpinView?**
Yes

**Describe the bug**
when I run `roslaunch spinnaker_sdk_camera_driver acquisition.launch binning:=2` using root  shell, log file as follows:

 `... logging to /root/.ros/log/bd1bdf3c-da39-11eb-a182-48b02d3c8d77/roslaunch-nvidia-desktop-4901.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia-desktop:44361/

SUMMARY

PARAMETERS
 * /acquisition_node/binning: 2
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [17165439]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/external_trigger: False
 * /acquisition_node/flip_horizontal: [False]
 * /acquisition_node/flip_vertical: [False]
 * /acquisition_node/frames: 3400
 * /acquisition_node/gain: 0
 * /acquisition_node/image_height: 1080
 * /acquisition_node/image_width: 1440
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 17165439
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/region_of_interest/height: 0
 * /acquisition_node/region_of_interest/width: 0
 * /acquisition_node/region_of_interest/x_offset: 0
 * /acquisition_node/region_of_interest/y_offset: 0
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/tf_prefix: 
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
[... 71 lines cut ...]
[ WARN] [1625122718.782643776]: Frames are being skipped!
[ WARN] [1625122725.305636864]: Image incomplete with image status 3!
[ WARN] [1625122725.352558720]: Frames are being skipped!
[ WARN] [1625122729.982655232]: Image incomplete with image status 3!
[ WARN] [1625122730.016548160]: Frames are being skipped!
[ WARN] [1625122731.239272608]: Image incomplete with image status 3!
[ WARN] [1625122731.281747616]: Frames are being skipped!
[ WARN] [1625122733.739670720]: Image incomplete with image status 3!
[ WARN] [1625122733.781980384]: Frames are being skipped!
[ WARN] [1625122735.541918240]: Image incomplete with image status 3!
[ WARN] [1625122735.573182560]: Image incomplete with image status 3!
[ WARN] [1625122735.615300672]: Frames are being skipped!
[ WARN] [1625122736.570218208]: Image incomplete with image status 3!
[ WARN] [1625122736.615274848]: Frames are being skipped!
[ WARN] [1625122738.442516416]: Image incomplete with image status 3!
[ WARN] [1625122738.484362464]: Frames are being skipped!
[FATAL] [1625122750.521860640]: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]
 Likely reason is that slaves are not triggered. Check GPIO cables

[vision_nodelet_manager-2] process has died [pid 4929, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/root/.ros/log/bd1bdf3c-da39-11eb-a182-48b02d3c8d77/vision_nodelet_manager-2.log].
log file: /root/.ros/log/bd1bdf3c-da39-11eb-a182-48b02d3c8d77/vision_nodelet_manager-2*.log
[acquisition_node-3] process has finished cleanly
log file: /root/.ros/log/bd1bdf3c-da39-11eb-a182-48b02d3c8d77/acquisition_node-3*.log

`

supertoto29, 2021-09-01:
@atyshka We had the same issue with a setup for 2 cameras. Both cameras were initialized but after a second, the node would crash. 

We have made the following modifications in this pull request : https://github.com/sonia-auv/provider_vision/pull/22 This fixed the issue.

Note : we are running an older version (April 2020) of this driver.

ana-lan, 2023-08-23:
> Asked about this error from FLIR support team, and this is their answer:
> error: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]
> "This error occurs when trying to retrieve an image from the buffer(RAM) while the buffer is empty."

How to solve this issue? I am new to Ros and couldn't find a solution for this anywhere.

Harsh188, 2024-07-29:
> > Yes, for multicam this package has to be setup as Master-slave, otherwise there is no way to guarantee the sync between cameras.
> 
> @shahvi: Yes, I am also using two cameras, one RGB and IR. But, I am using this package only for RGB. I am running IR from an another package. Yet, I am getting the same error as this post. Any suggestions?

Hi,

I've used a hardware trigger by connecting the sync wires between my IR and RGB camera. I'm trying to retrieve images from both cameras using a modified version of the `Trigger.py` example. However, my code runs into an error when trying to `GetNextImage()` I'm not sure what the issue is.

maninka123, 2025-02-03:
The images are too large to stream. To fix this, downsample the images by increasing the image_format_x_binning and image_format_y_binning values. Also, reduce the image_format_roi_width and image_format_roi_height.

You should then see the image output.