neufieldrobotics/spinnaker_sdk_camera_driver #45: Excption: Spinnaker: Could not write Port at address: 000C8824

ancabilloni, 2019-07-11:
Hi. 

I'm using Blackfly S USB and running tiny yolo for object detection on Jetson Xavier. A few times, I run into this error and the node shuts down. 

I use the `acquisition.launch`, color image at  with usbfs memory 1000-2048. Can you help to clarify what this error means? Does it relate to memory issue?

`[FATAL] [1562808193.910065924]: Excption: Spinnaker: Could not write Port at address: 000C8824 with length: 00000004
 [-1010]
`

shahvi, 2019-07-12:
Are you able to run the camera with SpinView?  I have not seen this error before but looks like it might be on the Spinnaker side.

ancabilloni, 2019-07-12:
SpinView does not work for me on Jetson Xavier Ubuntu 18.04 (on both Spinnaker version 1.23 and 1.21). I verify the camera working because this ros driver can acquire the frames. In the case of memory goes low (about .9 gb available), this error happens much sooner. In case I leave 3gb available gb, the longest I can run was about 24 mins for frame acquisition node + darknet node.

oysteinvolden, 2019-10-29:
Did you figure it out? I'm struggling with the same issue.

oysteinvolden, 2019-10-31:
For me, it seems to help to reset to factory settings. I did this via Spinview in the User Set Control. Then I set binning horizontal/vertical to 2 to lower resolution (also in Spinview). I got at least a more stable connection when I launch the acquisition file. But to be honest, its hard to say what the real problem actually is. As you say, it may be something with the linux kernel. I also use jetpack 4.2.2 for my Nvidia Xavier. For my case, the cameras is working better with this ros camera driver compared to spinview.

closed 2020-06-29 by shahvi