neufieldrobotics/spinnaker_sdk_camera_driver #46: Qt plugin issue after spinnaker SDK installation

m0oz, 2019-07-15:
**Setup:**
Ubuntu 16.04.1, kernel 4.15.0-29
ROS kinetic
FLIR Spinnaker SDW v1.21.0.61

**Description:**
After installing the current version of the Spinnaker SDK (1.21.0.61) from FLIR's website many Qt base ros tools such as rviz, and all rqt_* variants crash on launch.
I did not go much into detail, so I do not have a fix yet. Of course, this is not related to this ROS driver but rather the under-lying SDK. but I guess this is relevant for anyone who wants to use this driver alongside ROS kinetic.

**Steps to Reproduce:**

- Download Spinnaker SDK (1.21.0.61) from https://flir.app.boxcn.net/v/SpinnakerSDK
- Install dependencies as described in the SDKs README
- Install SDK using install_spinnaker.sh
- Invoke ```rosrun rviz rviz```.
Output:
```
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".
```
-Invoke  ```rosrun rqt_image_view rqt_image_view```
Output: 
```
Segmentation fault (core dumped)
```

m0oz, 2019-07-15:
Tried  ```export QT_DEBUG_PLUGINS=1``` and relaunch rviz to get more info about the cause.
Out of all directories that are checked for plugins it seems like the issue is with the files located in /opt/ros/kinetic/bin
I suspect that either something about the Qt plugin path or the Qt version gets messed up by the spinnaker install script or that the system Qt version is somehow modified ...

```
QFactoryLoader::QFactoryLoader() checking directory path "/opt/ros/kinetic/bin" ...

QFactoryLoader::QFactoryLoader() looking at "/opt/ros/kinetic/bin/binvox2bt"
"Plugin verification data mismatch in '/opt/ros/kinetic/bin/binvox2bt'" 
         not a plugin
QFactoryLoader::QFactoryLoader() looking at "/opt/ros/kinetic/bin/bt2vrml"
"Plugin verification data mismatch in '/opt/ros/kinetic/bin/bt2vrml'" 
         not a plugin
...
```
And many more

m0oz, 2019-07-15:
Btw if you want to fix your ros qt applications you can use the SDKs remove_spinnaker.sh script to undo the changes that mess with ros' qt plugins

m0oz, 2019-07-15:
Modifying install_spinnaker.sh so that only libspinnaker-*.deb is installed but all other packages are skipped fixes the issue. Be aware that this means that the SpinView and SpinUpdate GUI are not installed.

Modified section of  install_spinnaker.sh:
```
sudo dpkg -i libspinnaker-*.deb
# sudo dpkg -i libspinvideo-*.deb
# sudo dpkg -i spinview-qt-*.deb
# sudo dpkg -i spinupdate-*.deb
# sudo dpkg -i spinnaker-*.deb
```

closed 2019-07-15 by m0oz

shahvi, 2019-07-15:
@m0oz its always great when issues get addressed on their own.  On a different note, we have just been using Spinnaker v1.13.0.31 on our systems running Ubuntu 16.04.  I personally haven't been able to make any of the newer versions work. I suspect that Point Grey have targeted the newer versions for Ubuntu 18.04.  Is there any feature that you need from the newer versions?

m0oz, 2019-08-11:
@shahvi No I don't need any specific feature from the newer versions.
It's just that FLIR is not hosting any legacy versions on there site so you always only get a download link to the newest version (here)[https://www.flir.com/support-center/iis/machine-vision/downloads/spinnaker-sdk-and-firmware-download/]
Where would you get the older versions?

shahvi, 2019-08-11:
I have added the 1.13 for AMD64 on ubuntu 16.04 here https://drive.google.com/drive/folders/1V6UFuDH38p-AX-verc-Rrpv88MM3fgbr?usp=sharing

ykk136, 2020-12-15:
Thank you very much for solving this issue which has confused me for long time. However, I recently found that there is no such issue if you install spinnaker-2.0.0 on Ubuntu 18.04LTS (link is here: https://flir.app.boxcn.net/v/SpinnakerSDK/folder/74729115388). And this spinnaker-2.0.0 would not destroy system QT software.

Therefore, there is no need for us to modify CMakeLists.txt to find Spinnker.h any more.