neufieldrobotics/spinnaker_sdk_camera_driver #48: Problem with roslaunching the acquisition.launch using melodic

alankwok97, 2019-07-18:
Hi there,

I am currently working on a new project using ROS to get data from a FLIR Blackfly camera (I am still quite new to this language). I followed the instructions written on the ROS wiki web page (including changing the serial number) but still couldn't launch the acquisition.launch file. Here are the FATAL errors that I have:

[FATAL] [1563449758.815448351]: Error: Spinnaker: GenICam::OutOfRangeException= Value = 1 must be equal or greater than Min = 2. : OutOfRangeException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 79) [-2002]
[ WARN] [1563449758.815547185]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1563449758.934733561]: Failed to load nodelet '/acquisition_nodeof typeacquisition/capture_nodeletto managervision_nodelet_manager'
terminate called after throwing an instance of 'Spinnaker::Exception'
what(): Spinnaker: Camera is not started. [-1002]
[vision_nodelet_manager-2] process has died [pid 31124, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/dell8116/.ros/log/3345535e-a950-11e9-b8e9-c8f750384496/vision_nodelet_manager-2.log].
log file: /home/dell8116/.ros/log/3345535e-a950-11e9-b8e9-c8f750384496/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 31125, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/dell8116/.ros/log/3345535e-a950-11e9-b8e9-c8f750384496/acquisition_node-3.log].
log file: /home/dell8116/.ros/log/3345535e-a950-11e9-b8e9-c8f750384496/acquisition_node-3*.log

And here is the text in test_params.yaml:
cam_ids:

19138453
cam_aliases:
cam0
master_cam: 19138453
skip: 20
delay: 1.0
Assign all the follwing via launch file to prevent confusion and conflict
#save_path: ~/projects/data
#save_type: .bmp #binary or .tiff or .bmp
#binning: 1 # going from 2 to 1 requires cameras to be unplugged and replugged
#color: false
#frames: 50
#soft_framerate: 20 # this frame rate reflects to the software frame rate set using ros::rate
#exp: 997
#to_ros: true #When to_ros is not selected, but live is selected, pressing 'space' exports single image to ROS

#Camera info message details
distortion_model: plumb_bob
image_height: 1536
image_width: 2048
distortion_coeffs:

[-0.021141875266089592, -0.3733872931278025, 2.385982550579459, 3.2824571732099725]
#specified as [fx 0 cx 0 fy cy 0 0 1]
intrinsic_coeffs:

[1886.9232141485886, 0.0, 604.7214878709341, 0.0, 1886.6668765711668, 493.47726714719823, 0.0, 0.0, 1.0]
rectification_coeffs:

[1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_coeffs:

[913.700317, 0.000000, 953.448302, 0.000000, 0.000000, 1063.296631, 777.871993, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]

Currently I am using ros melodic (Ubuntu 18.04 Bionic) and I know that the package is written for kinetic but are there ways to fix these problems and get it working? It would mean so much to me.

Many thanks.

shahvi, 2019-07-18:
Try setting binning = 2 in launch file to see of that helps.

alankwok97, 2019-07-18:
Hi, thanks for your reply. Different errors arise with the change.

[FATAL] [1563462420.179655764]: ASSERTION FAILED
file = /home/dell8116/spinnaker_ws/src/spinnaker_camera_driver/src/capture.cpp
line = 214
cond = numCameras_
message = 
[FATAL] [1563462420.179762676]: No cameras found!
[FATAL] [1563462420.179800067]: 

[FATAL] [1563462420.302010849]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 5859, exit code -5, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/dell8116/.ros/log/afd0bd06-a96d-11e9-9a62-c8f750384496/vision_nodelet_manager-2.log].
log file: /home/dell8116/.ros/log/afd0bd06-a96d-11e9-9a62-c8f750384496/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 5860, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/dell8116/.ros/log/afd0bd06-a96d-11e9-9a62-c8f750384496/acquisition_node-3.log].
log file: /home/dell8116/.ros/log/afd0bd06-a96d-11e9-9a62-c8f750384496/acquisition_node-3*.log

alankwok97, 2019-07-18:
Changing back binning to 1 gives:

[FATAL] [1563462632.939289247]: ASSERTION FAILED
file = /home/dell8116/spinnaker_ws/src/spinnaker_camera_driver/src/camera.cpp
line = 59
cond = frameID_ == 0
message = 
[FATAL] [1563462632.939392615]: First frame ID was not zero! Might cause sync issues later...
[FATAL] [1563462632.939430534]: 

[vision_nodelet_manager-2] process has died [pid 6013, exit code -5, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/dell8116/.ros/log/2cc0aae2-a96e-11e9-9a62-c8f750384496/vision_nodelet_manager-2.log].
log file: /home/dell8116/.ros/log/2cc0aae2-a96e-11e9-9a62-c8f750384496/vision_nodelet_manager-2*.log

shahvi, 2019-07-18:
Is this a USB cam or ethernet?

alankwok97, 2019-07-18:
ethernet

shahvi, 2019-07-18:
We dont have any ethernet cameras in our lab to test, but someone else has been able to get past a similar issue here  at #10

alankwok97, 2019-07-19:
Hi, thanks for your help. I managed to get it working on melodic as well in colour.

Regarding of the size and resolution of the image, how may I change them?

Many thanks.

shahvi, 2019-07-19:
That is what binning does. So binning by 2 reduces width and height by 1/2 and total resolution by 1/4

closed 2019-07-29 by alankwok97

alankwok97, 2019-07-31:
@shahvi I can see that you have included the method "void setBufferSize(int numBuf)" inside the camera.h file which is located in spinnaker_sdk_camera_driver-master/include/spinnaker_sdk_camera_driver. Therefore, I am wondering where have you called this method and would it be possible for me to change the number manually?
 
I want to do this because I keep getting EventBuffer error when I have a fps higher that 10 but the thing is, I need at least 20 fps. (binning = 1)

Many thanks.

P.S. I have verified that SpinView worked at 60 fps no problem at binning = 1

alankwok97, 2019-08-01:
@shahvi I have managed to solve the Event Buffer error by increasing the ring buffer of my NIC. I am currently trying to change the interpolation algorithm, do you how may I do that in the script?

shahvi, 2019-08-01:
If you are referring to the PixelFormat option,  see issue #10 I describe how to set up the format to BGR8.

If you are looking for a different setting, try finding it in SpinView and then you should be able to change the enumeration accordingly in the code.

alankwok97, 2019-08-02:
@shahvi Thanks for your reply, I managed to change the pixel format accordingly. I am now trying to change the parameter, buffer count, in the script but had no luck with it. Could you please point me in the right direction?

Many thanks.

Xulei-Zhai, 2021-06-24:
@alankwok97 @shahvi 
I have met the same problem as yours. I still can't slove this problem even though I have spend much time on it .  Here are the details :(ubuntu18.04 ,spinnaker version :2.2.0.48)

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [19085011]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/external_trigger: False
 * /acquisition_node/flip_horizontal: [False]
 * /acquisition_node/flip_vertical: [False]
 * /acquisition_node/frames: 3400
 * /acquisition_node/gain: 0
 * /acquisition_node/image_height: 1080
 * /acquisition_node/image_width: 1440
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 19085011
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/region_of_interest/height: 0
 * /acquisition_node/region_of_interest/width: 0
 * /acquisition_node/region_of_interest/x_offset: 0
 * /acquisition_node/region_of_interest/y_offset: 0
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/tf_prefix: 
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [3029]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0523b842-d4ed-11eb-9377-a40c66050bb4
process[rosout-1]: started with pid [3040]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [3047]
process[acquisition_node-3]: started with pid [3048]
[ INFO] [1624540002.950281771]: Initializing nodelet with 8 worker threads.
[ INFO] [1624540003.031144738]: Initializing nodelet
[ INFO] [1624540003.031688831]: [ OK ] USB memory: 1000 MB
[ INFO] [1624540003.031718356]: *** PARAMETER SETTINGS ***
[ INFO] [1624540003.031726592]: ** Date = 20210624
[ INFO] [1624540003.032100178]:   Save path set via parameter to: /home/good
[ INFO] [1624540003.032134688]:   Camera IDs:
[ INFO] [1624540003.032467966]:     19085011
[ INFO] [1624540003.032726638]:   Camera Aliases:
[ INFO] [1624540003.032747153]:     19085011 >> cam0
[... 27 lines cut ...]
[ INFO] [1624540003.038744298]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1624540003.038751559]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1624540003.038786791]: *** SYSTEM INFORMATION ***
[ INFO] [1624540003.038812289]: Creating system instance...
[ INFO] [1624540003.038833627]: spinnaker_sdk_camera_driver package version: 1.1.1
[ INFO] [1624540003.893396533]: Spinnaker library version: 2.2.0.48
[ INFO] [1624540003.893499457]: Retreiving list of cameras...
[ INFO] [1624540003.899294187]: Numer of cameras found: 1
[ INFO] [1624540003.899398326]:  Cameras connected: 1
[ INFO] [1624540003.901143681]:   -19085011 Blackfly S BFS-PGE-16S2C 1705.0.63.0
[ INFO] [1624540003.945383078]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1624540003.945967883]: *** FLUSH SEQUENCE ***
[ INFO] [1624540003.945995196]: Initializing cameras...
[ INFO] [1624540004.531815922]: Deinitializing cameras...
[ INFO] [1624540004.739228787]: All cameras deinitialized.
[ INFO] [1624540006.739689842]: Initializing cameras...

------------------------------------------
[FATAL] [1624540007.111634493]: Error: Spinnaker: GenICam::OutOfRangeException= Value = 1 must be equal or greater than Min = 2. : OutOfRangeException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 79) [-2002]
[ WARN] [1624540007.111672649]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1624540007.314172727]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
[acquisition_node-3] process has died [pid 3048, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/home/good/.ros/log/0523b842-d4ed-11eb-9377-a40c66050bb4/acquisition_node-3.log].
log file: /home/good/.ros/log/0523b842-d4ed-11eb-9377-a40c66050bb4/acquisition_node-3*.log
[vision_nodelet_manager-2] process has died [pid 3047, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/good/.ros/log/0523b842-d4ed-11eb-9377-a40c66050bb4/vision_nodelet_manager-2.log].
log file: /home/good/.ros/log/0523b842-d4ed-11eb-9377-a40c66050bb4/vision_nodelet_manager-2*.log

lantianran, 2021-12-28:
> 

I have the same problems , do you solve it ?

gschelske, 2022-05-25:
same error pls help