neufieldrobotics/spinnaker_sdk_camera_driver #49: Issue Increasing Frame Rate

kaalessi, 2019-07-25:
I am using an ORX-10G-51S5C camera with ROS and recording the data into a rosbag. I am consistently getting a frame rate in the mid 20s and would like to increase it to more than 30 fps. I am calculating the frame by dividing msgs in rosbag by the length in seconds of the rosbag. I have set binning to 2, and in SpinView set the GevSCPD to 1000 and the DeviceLinkThroughputLimit to 1200000000. What can be changed to increase the framerate and not impact the resolution significantly? When binning is set to 2 in SpinView, I am able to get 144 frames per second. I was able to get my GigE camera working by following this thread: https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/issues/10.

alankwok97, 2019-08-01:
@kaalessi I have exactly the same problem as you but just with a different camera, did you managed to increase the frame rate at the end?

alankwok97, 2019-08-01:
@kaalessi I changed the pixel format according to #10 and the fps went up but I am not sure would the quality suffer so might be worth checking it out. I have only tested BayerRG8 so far.

kaalessi, 2019-08-01:
@alankwok97 When setting the camera up, I also changed the pixel format like you to BayerRG8 and the frame rate was increased. I have not experimented with the other formats, but might give that a try today. Before changing the pixel format, I was getting like 6ish frames per second. I still haven't figured out the trick to get the frame rate up. I contacted FLIR/Point Grey Research last week and they seemed to think that if I can somehow increase the image buffer count for the ROS driver, then I may be able to get a higher frame rate. Changing the image buffer count in SpinView has helped temporarily, but it gets reset to the defaults even after saving. Their explanation was each individual program has its own buffer count. They also suggested it could be because the storage is too slow, but I have 6 solid state drives in RAID 0, so I doubt that is my problem. Other than that, they didn't have much of an idea. If you find anything out, please keep me up to date.

alankwok97, 2019-08-01:
@kaalessi It's strange that you are only get 6 fps with the setting you are currently in as I basically have the same setting as you, GevSCPD to 2000 and the DeviceLinkThroughputLimit to 1250000000, and I am getting 40-50 fps when I switch to BayerRG8 pixel format. Perhaps might be worth checking if your connection is really 1Gbit/s. Also, if you are still getting the EventBuffer error, increasing the ring buffer will also help and might be worth trying as well.

Regarding of the buffer count parameter, I tried altering it as well but as you said, mine didn't save either so I am not sure about that. 

I will update you if I find anything new.

kaalessi, 2019-08-01:
@alankwok97 Thanks for the response. I was getting 6 fps before I changed the pixel format to BayerRG8. Now that I am on that, I am getting between 20 and 30 something frames per second. I was getting the EventBuffer error previously, but fixed it by increasing both my ring buffer and UDP buffer. My network interface is also running at full 10G. There may be a spot in one of the .cpp files that we can manually edit the buffer count. If so, we may be in luck. I have used the grep command to search for both "frames" and "buffer" and other variants, but did not find anything too useful.

alankwok97, 2019-08-05:
@kaalessi I figured that you can install Autoware instead of this driver since you have SpinView installed. With autoware, you don't even need to change any buffer settings and I am getting 60 fps easily (check via typing in the command rostopic hz <topics> )

Here is the link: https://gitlab.com/autowarefoundation/autoware.ai/autoware/wikis/Source-Build

Just some head ups regarding of installing Autoware:
1. You need to configure Xorg as your default GNOME session
2. You need to fix the GUI bug 

But they are all readily available online so I am sure you would be able to configure/fix them in no time.

alankwok97, 2019-08-05:
[image]

kaalessi, 2019-08-05:
@alankwok97 Awesome! Thank you so much! I will look into it. I just botched my Linux install playing with some settings. Once I reinstall everything, I will definitely check this out!

closed 2020-06-29 by shahvi