neufieldrobotics/spinnaker_sdk_camera_driver #50: Cannot connect to GigE Blackfly S

atyshka, 2019-08-12:
I am trying to use a GigE Blackfly S with this driver and the nodelet is crashing without any error messages. I've correctly set the camera serial and experimented with both binning 1 and 2 and left all other parameters at default. I can confirm that the camera works perfectly fine in spinview. Here is the log output:

```
started roslaunch server http://alex-desktop:35887/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 2
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [19220523]
 * /acquisition_node/color: True
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1536
 * /acquisition_node/image_width: 2048
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 19220523
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [12907]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 07f42e1e-bd37-11e9-b8e2-7085c2ae64f9
process[rosout-1]: started with pid [12918]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [12922]
process[acquisition_node-3]: started with pid [12926]
[ INFO] [1565637967.211156336]: Initializing nodelet with 8 worker threads.
[ INFO] [1565637967.288913402]: Initializing nodelet
[ INFO] [1565637967.289453491]: [ OK ] USB memory: 1000 MB
[ INFO] [1565637967.289478199]: *** PARAMETER SETTINGS ***
[ INFO] [1565637967.289484704]: ** Date = 20190812
[ INFO] [1565637967.289844771]:   Save path set via parameter to: /home/alex
[ INFO] [1565637967.289859725]:   Camera IDs:
[ INFO] [1565637967.290148244]:   Camera Aliases:
[FATAL] [1565637967.363517338]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 12922, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/alex/.ros/log/07f42e1e-bd37-11e9-b8e2-7085c2ae64f9/vision_nodelet_manager-2.log].
log file: /home/alex/.ros/log/07f42e1e-bd37-11e9-b8e2-7085c2ae64f9/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 12926, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/alex/.ros/log/07f42e1e-bd37-11e9-b8e2-7085c2ae64f9/acquisition_node-3.log].
log file: /home/alex/.ros/log/07f42e1e-bd37-11e9-b8e2-7085c2ae64f9/acquisition_node-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

Any ideas what could be causing this? I know others have successfully used GigE Blackfly cameras with this driver just fine.

atyshka, 2019-08-13:
Updated sdk, rebuilt, and problem fixed

closed 2019-08-13 by atyshka

HTLife, 2021-04-21:
I already update the SDK to the latest `spinnaker-2.3.0.77-amd64`
But I still meet the same problem.

OS: Ubuntu20.04

lantianran, 2021-12-27:
> I already update the SDK to the latest `spinnaker-2.3.0.77-amd64` But I still meet the same problem.
> 
> OS: Ubuntu20.04

I met the same problem .  My SDK is spinnaker-2.2.0.48-amd64  ubuntu 18.04
[image]