neufieldrobotics/spinnaker_sdk_camera_driver #52: The acquisition.launch dose not run correctly

DowsonTseng, 2019-08-14:
Hi all 
I bought a new flir s camera , but I can't launch thisdrive node correctly according to the steps of readme . The error is as fllows:
     `[FATAL] [1565781001.434096093]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-1] process has died [pid 15410, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/pix/.ros/log/83dfd6f2-be6f-11e9-856c-1c1b0daf78ce/vision_nodelet_manager-1.log].
log file: /home/pix/.ros/log/83dfd6f2-be6f-11e9-856c-1c1b0daf78ce/vision_nodelet_manager-1*.log
[acquisition_node-2] process has died [pid 15411, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/pix/.ros/log/83dfd6f2-be6f-11e9-856c-1c1b0daf78ce/acquisition_node-2.log].
log file: /home/pix/.ros/log/83dfd6f2-be6f-11e9-856c-1c1b0daf78ce/acquisition_node-2*.log
`
How is this done?

DowsonTseng, 2019-08-14:
When I start the node alone(`rosrun spinnaker_sdk_camera_driver acquisition_node
`),I get the following error:
`[ INFO] [1565780958.474735279]: [ OK ] USB memory: 1000 MB
[ INFO] [1565780958.474790765]: *** PARAMETER SETTINGS ***
[ INFO] [1565780958.474803717]: ** Date = 20190814
[ INFO] [1565780958.475281176]:   Save path set via parameter to: /home/pix
[ INFO] [1565780958.475307916]:   Camera IDs:
[ INFO] [1565780958.475720143]:     19139183
[ INFO] [1565780958.476099151]:   Camera Aliases:
[ INFO] [1565780958.476114315]:     19139183 >> cam0
[ INFO] [1565780958.476872662]:   Unique time stamps for each camera: false
[ INFO] [1565780958.477221573]:   color set to: false
[ INFO] [1565780958.477526154]:   Exporting images to ROS: true
[ INFO] [1565780958.477816800]:   Showing live images setting: false
[ INFO] [1565780958.478188059]:   Showing grid-style live images setting: false
[ INFO] [1565780958.478481583]:   Max Rate Save Mode: false
[ INFO] [1565780958.478764699]:   Displaying timing details: false
[ INFO] [1565780958.479078578]:   No. of images to skip set to: 20
[ INFO] [1565780958.479369379]:   Init sleep delays set to : 1.00 sec
[ WARN] [1565780958.479721306]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1565780958.480008356]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1565780958.480325397]:   target_grey_value set to: 4.0
[ INFO] [1565780958.480632917]:   Binning set to: 1
[ INFO] [1565780958.480912075]:   Using Software rate control, rate set to: 20
[ INFO] [1565780958.481192091]:   Saving images set to: 0
[ INFO] [1565780958.481496225]:   Camera Intrinsic Paramters:
[ INFO] [1565780958.481574706]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1565780958.481873747]:   Camera Distortion Paramters:
[ INFO] [1565780958.481898372]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1565780958.482193114]:   Camera Rectification Paramters:
[ INFO] [1565780958.482219495]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1565780958.482519161]:   Camera Projection Paramters:
[ INFO] [1565780958.482549274]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1565780958.482561890]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1565780958.482579764]: Creating system instance...
[ INFO] [1565780958.483737567]: Retreiving list of cameras...
[ INFO] [1565780959.458669777]: Numer of cameras found: 1
[ INFO] [1565780959.458707879]:  Cameras connected: 1
[ INFO] [1565780959.458715488]:  Cameras come hare 
[ INFO] [1565780959.458735022]:   -19139183
[ INFO] [1565780959.482960217]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1565780959.483824102]: *** FLUSH SEQUENCE ***
[ INFO] [1565780959.483838286]: Initializing cameras...
[ INFO] [1565780959.483858911]: Camera inter init camera func1
Segmentation fault (core dumped)
`

DowsonTseng, 2019-08-14:
My camera model is : BFS-PGE-16S2C-CS

atyshka, 2019-08-14:
Had same problem. Try deleting build and devel folders in you catkin workspace, uninstall spinnaker and reinstall latest version from flir website, and re-build the catkin workspace. Fixed it for me

cjzhang916, 2019-11-01:
Have the same problem.Have you solved it?

ludvigls, 2019-11-08:
Some rosnodes are probably running in the background. Try this:
rosnode kill -a

EunbinSeo, 2020-05-10:
I have same problem.

[ INFO] [1589105014.902726761]: Initializing cameras...
[FATAL] [1589105015.167666185]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-1] process has died [pid 6530, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/dgist/.ros/log/22982268-92a5-11ea-a574-0c9d925bd54e/vision_nodelet_manager-1.log].
log file: /home/dgist/.ros/log/22982268-92a5-11ea-a574-0c9d925bd54e/vision_nodelet_manager-1*.log
[acquisition_node-2] process has died [pid 6532, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/dgist/.ros/log/22982268-92a5-11ea-a574-0c9d925bd54e/acquisition_node-2.log].
log file: /home/dgist/.ros/log/22982268-92a5-11ea-a574-0c9d925bd54e/acquisition_node-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

How can I fix it?

vik748, 2020-05-10:
@EunbinSeo can you start a new issue and fill out the details in there. We need more information to be able to say anything.

Hutaoie, 2020-06-03:
I got the error while roslaunch the acquisition_node.launch
[acquisition_node-1] process has died [pid 10822, exit code 127, cmd /home/***/catkin_ws/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/***/.ros/log/***/acquisition_node-1.log].
log file: /home/***/.ros/log/***/acquisition_node-1*.log
I've tried to reinstall the Spinnaker SDK, but still failed.

vik748, 2020-06-03:
@Hutaoie please start a new issue and fill out all the requested information.