neufieldrobotics/spinnaker_sdk_camera_driver #54: ERROR: Unable to set BinningHorizontal to 1 (ptr retrieval)

addext, 2019-09-11:
hi,

I installed the package as described in the readme file but run into following error when launching the acquisition.launch file:

`SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 0
 * /acquisition_node/cam_aliases: ['cam0', 'cam1']
 * /acquisition_node/cam_ids: [19196232, 18369848]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1200
 * /acquisition_node/image_width: 1920
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 19196232
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [3186]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c62ad446-d484-11e9-a8db-704d7b468a87
process[rosout-1]: started with pid [3197]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [3204]
process[acquisition_node-3]: started with pid [3205]
[ INFO] [1568200234.378092973]: Initializing nodelet with 8 worker threads.
[ INFO] [1568200234.469821027]: Initializing nodelet
[ INFO] [1568200234.470650278]: [ OK ] USB memory: 1000 MB
[ INFO] [1568200234.470738239]: *** PARAMETER SETTINGS ***
[ INFO] [1568200234.470766384]: ** Date = 20190911
[ INFO] [1568200234.471259372]:   Save path set via parameter to: /home/dflap3
[ INFO] [1568200234.471344509]:   Camera IDs:
[ INFO] [1568200234.471757675]:     19196232
[ INFO] [1568200234.471792758]:     18369848
[ INFO] [1568200234.472160588]:   Camera Aliases:
[ INFO] [1568200234.472199804]:     19196232 >> cam0
[ INFO] [1568200234.472227846]:     18369848 >> cam1
[ INFO] [1568200234.472832455]:   Unique time stamps for each camera: false
[ INFO] [1568200234.473213194]:   color set to: false
[ INFO] [1568200234.473500659]:   Exporting images to ROS: true
[ INFO] [1568200234.473786457]:   Showing live images setting: false
[ INFO] [1568200234.474056609]:   Showing grid-style live images setting: false
[ INFO] [1568200234.474328396]:   Max Rate Save Mode: false
[ INFO] [1568200234.474594420]:   Displaying timing details: false
[ INFO] [1568200234.474852531]:   No. of images to skip set to: 20
[ INFO] [1568200234.475111779]:   Init sleep delays set to : 1.00 sec
[ WARN] [1568200234.475356684]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1568200234.475595481]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1568200234.475838259]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1568200234.476077990]:   'binning'=1 invalid, Using defauly binning=
[ INFO] [1568200234.476317417]:   Using Software rate control, rate set to: 30
[ INFO] [1568200234.476556696]:   Saving images set to: 0
[ INFO] [1568200234.476817448]:   Camera Intrinsic Paramters:
[ INFO] [1568200234.477031244]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1568200234.477087424]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1568200234.477354952]:   Camera Distortion Paramters:
[ INFO] [1568200234.477408009]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1568200234.477423409]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1568200234.477650528]:   Camera Rectification Paramters:
[ INFO] [1568200234.477699540]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1568200234.477726995]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1568200234.478031901]:   Camera Projection Paramters:
[ INFO] [1568200234.478095465]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1568200234.478151048]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1568200234.478163537]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1568200234.478190427]: Creating system instance...
[ INFO] [1568200234.479567690]: Retreiving list of cameras...
[ INFO] [1568200235.951922580]: Numer of cameras found: 2
[ INFO] [1568200235.951970578]:  Cameras connected: 2
[ INFO] [1568200235.952035932]:   -18369848
[ INFO] [1568200235.952070280]:   -19196232
[ INFO] [1568200235.991937338]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1568200235.992759586]: *** FLUSH SEQUENCE ***
[ INFO] [1568200235.992779510]: Initializing cameras...
[ INFO] [1568200236.850680937]: Deinitializing cameras...
[ INFO] [1568200237.259045112]: All cameras deinitialized.
[ INFO] [1568200239.259414460]: Initializing cameras...
[FATAL] [1568200239.467671816]: Unable to set BinningHorizontal to 1 (ptr retrieval). Aborting...
[FATAL] [1568200239.467947258]: Error: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 77) [-2006]
[ WARN] [1568200239.467999702]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1568200239.661708138]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[acquisition_node-3] process has died [pid 3205, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/dflap3/.ros/log/c62ad446-d484-11e9-a8db-704d7b468a87/acquisition_node-3.log].
log file: /home/dflap3/.ros/log/c62ad446-d484-11e9-a8db-704d7b468a87/acquisition_node-3*.log
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'TriggerSoftware' while calling 'TriggerSoftware.Execute()' (file 'CommandT.h', line 61) [-2006]
[vision_nodelet_manager-2] process has died [pid 3204, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/dflap3/.ros/log/c62ad446-d484-11e9-a8db-704d7b468a87/vision_nodelet_manager-2.log].
log file: /home/dflap3/.ros/log/c62ad446-d484-11e9-a8db-704d7b468a87/vision_nodelet_manager-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

as you can see, binning was set to 0 in the params...
Also color mode was deactivated, so that can't be the issue - even though the cameras (BFLY-PGE-23S6-C) do support color.

Any suggestions?

Thanks a lot and best regards,
addext

shahvi, 2019-09-11:
Does spinview allow setting  binning to 1?

addext, 2019-09-11:
no, horizontal binning is locked to 1 in Spinview

addext, 2019-09-25:
apparently the issue was ROS melodic. Tried it with kinetic and it worked

closed 2019-09-25 by addext

impala454, 2020-01-27:
I am seeing this exact same error and am on ROS Kinetic.