neufieldrobotics/spinnaker_sdk_camera_driver #59: Could not find a package configuration file provided by   "image_exposure_msgs" with any of the following names

milmario, 2019-11-08:
Hi,
when I'm trying to "catkin build" the spinnaker_camera_driver, I get the following error message:

```
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "image_exposure_msgs" with any of the following names:

    image_exposure_msgsConfig.cmake
    image_exposure_msgs-config.cmake

  Add the installation prefix of "image_exposure_msgs" to CMAKE_PREFIX_PATH
  or set "image_exposure_msgs_DIR" to a directory containing one of the above
  files.  If "image_exposure_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:8 (find_package)

```

I'm new to the ROS-environment, so I don't really understand what that means?
I'm using Ubuntu 18.04 and ROS Melodic.

My catkin-configuration is:
```
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DBINDINGS_RUBY=OFF -DBUILD_SHARED_LIBS=ON -DCMAKE_CXX_FLAGS=-Wno-int-in-bool-context
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None

```

Please let me know if you need some further information

Thanks for your help! :)

vik748, 2019-11-08:
Try building again. Sometimes it doesn't build the messages in the first go.

milmario, 2019-11-08:
I tried building it many times actually... Error message stays the same

shahvi, 2019-11-10:
Which branch are you working off?  I don't see any mention of "image_exposure_msgs" anywhere in the master branch.

closed 2019-12-06 by milmario

nkhedekar, 2020-01-06:
As @shahvi said, it is not mentioned anywhere. Nonetheless, this can be solved by installing the `image_exposure_msgs` package as `sudo apt-get install ros-kinetic-image-exposure-msgs`