neufieldrobotics/spinnaker_sdk_camera_driver #61: AccessException thrown in node 'BinningHorizontal' while calling

LukeAI, 2019-11-26:
I have a BFLY-PGE-31S4C-C camera (colour camera) that works fine with SpinView but does not appear to work with this software. When I launch I get:

```
.
.
.
[ INFO] [1574766847.220783177]: *** FLUSH SEQUENCE ***
[ INFO] [1574766847.220853627]: Initializing cameras...
[ INFO] [1574766848.266822840]: Deinitializing cameras...
[ INFO] [1574766848.471857426]: All cameras deinitialized.
[ INFO] [1574766850.472168583]: Initializing cameras...
[FATAL] [1574766633.586155060]: Unable to set BinningHorizontal to 1 (ptr retrieval). Aborting...
[FATAL] [1574766633.586326700]: Error: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 77) [-2006]
[ WARN] [1574766633.586353570]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
```

I tried commenting out the lines that set binning as suggested here
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/issues/19

and I got similar complaints about "AutoExposureTargetGreyValueAuto" and then "LineSelector", "LineMode" etc. and eventually

```
[ INFO] [1574767506.657825357]: *** ACQUISITION ***
[FATAL] [1574767506.678578051]: Unable to execute trigger. Aborting...
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'TriggerSoftware' while calling 'TriggerSoftware.Execute()' (file 'CommandT.h', line 61) [-2006]

```

I guess this software doesn't support my camera? :)

vik748, 2020-01-21:
Sorry about the delayed response.  We only have Blackfly S USB cameras in our lab and so haven't been able to try with any other model.  Were you able to sort out your issues?

LukeAI, 2020-01-21:
hey no worries, no I wasn't able to resolve it. We are just using the flycap2 drivers which work fine.

closed 2020-06-29 by shahvi