neufieldrobotics/spinnaker_sdk_camera_driver #62: Hard framerate stop (Slow the camera framerate)

erikasnow, 2019-12-02:
What parameters do you change in the driver node to slow framerate? Changing the soft_framerate parameter causes the output raw image to delay images, not actually reducing frames from camera. 
Using the Flir Firefly S that outputs at 60 FPS. Trying to output at 30 FPS without any delay to the images.

erikasnow, 2019-12-02:
Also have there been issues on synchronization of camera_info and image_raw topics? This will cause problems when sending to image_proc part of image_pipeline

mithunvarma, 2019-12-02:
@erikasnow 
Which branch are you using (master, dev)?
We tested the code on Blackfly s cameras, where we do acquisition in continuous acquisition mode.
There are couple of ways to trigger:
1. software trigger: master cam is triggered based on soft_framerate, master cam triggers other slave cameras using gpio out as described in the Readme.
2. Hardware trigger: All connected cameras can be triggered using REF signal from sources like imu (available only in dev branch)
3. camera_outputs fixed frame rate: we implemented this in max_rate_save( by setting max_rate_save param to true), where we can set fixed frame rate on camera, uncomment line 654 and comment 653 to set the desired fps. This option doesn't export the images to ros, it just saves them to disk. check 
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/15803f032859341b2d1740dbce5300726b2ac8b8/src/capture.cpp#L654

The way to export images to ros mimicking third option(fixed frame rate) would be:
set, max_rate_save to 'False'(default)
1. go to capture.cpp/
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/15803f032859341b2d1740dbce5300726b2ac8b8/src/capture.cpp#L656

comment lines in 656-659 and add
```
cams[i].setEnumValue("LineSelector", "Line2");
cams[i].setEnumValue("LineMode", "Output");
// cams[i].setBoolValue("AcquisitionFrameRateEnable", false);
cams[i].setFloatValue("AcquisitionFrameRate", 30); // in your case
```
which will set camera to output images at fixed frame rate, now images are triggered on device instead of software or external hardware trigger.

Now, we need to change the code in lines 894, 934, 960 to comment the software triggering.
so it just grabs images when available.

comment lines with
```
 cams[MASTER_CAM_].trigger();
```
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/15803f032859341b2d1740dbce5300726b2ac8b8/src/capture.cpp#L894

https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/15803f032859341b2d1740dbce5300726b2ac8b8/src/capture.cpp#L934

https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/15803f032859341b2d1740dbce5300726b2ac8b8/src/capture.cpp#L960

Finally comment line 999, to remove sleep, grab_mat_frame() actually waits till a frame is available in buffer with a timeout of 2 sec.

https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/15803f032859341b2d1740dbce5300726b2ac8b8/src/capture.cpp#L999

I hope this setup should work, I haven't compiled or tested it.

mithunvarma, 2019-12-02:
Regarding sync issue, we use imagetransport camerapublisher which published image msg and info msg at same time and timestamps on msg headers of img_raw and cam_info are same.
Its in line https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/15803f032859341b2d1740dbce5300726b2ac8b8/src/capture.cpp#L792

can you tell me the steps you followed to reproduce the error.

If you have recorded images with rosbag with fixed buffer size, you might have dropped couple of msgs and that might be a reason for the issue. try recording with -b 0 in rosbag record.

Have you tried to subscribe using msg filters like in image_proc and does it throw warnings of mismatch. 
Also try with imagetransport camerasubscriber() instead of Subscriber() or imagetransport subscriber() 

http://docs.ros.org/melodic/api/image_transport/html/classimage__transport_1_1CameraSubscriber.html

erikasnow, 2019-12-03:
Thanks for your input! I compiled changes for fixed frame rate. No compile errors, but this error showed up on launch. 
`[FATAL] [1575400115.453809910]: Unable to set AcquisitionFrameRate to 30 (ptr retrieval). Aborting...
[FATAL] [1575400115.454385873]: Error: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'AcquisitionFrameRate' while calling 'AcquisitionFrameRate.SetValue()' (file 'FloatT.h', line 83) [-2006]
[ WARN] [1575400115.454478533]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event. [-1011]`

yzhou377, 2020-01-16:
@mithunvarma

Thank you for your reply above. We are trying to use a hardware trigger for this device, and you mentioned we need to change something in the dev branch. Would you kindly tell me how should I change this? 

Thank you for your time and effort. 

Best,
YZ

vik748, 2020-01-21:
@yzhou377 you question seems unrelated to this issue.  Can you open up a separate issue please with details on what you are trying to do.

dimaxano, 2020-02-24:
Not sure if the issue is still relevant for topic starter, but future user: note that to make @mithunvarma advice (fixed frame rate to ROS) work.
`cams[i].setBoolValue("AcquisitionFrameRateEnable", false);` line should be uncommented and set to `true`.
That work for me.

closed 2020-06-29 by shahvi