neufieldrobotics/spinnaker_sdk_camera_driver #63: Raspberry Pi4  Kernel Panic/Crash When Capturing Color

fultonms, 2019-12-12:
Hi! Thanks to all contributors for this driver!

I'm running a BlackFlyS FLIR camera on a Raspberry Pi4 (4GB), running Ubuntu 18.04 (kernel version 4.19.86-v8-24, most recent firmware, with usbfs_memory_mb set to 4096. While the camera works in SpinView dependably, there are some issues running the camera through this ROS driver. 

In my testing, I have only been able to get the ROS driver to successfully run under certain conditions:
1. Spinview has to be run first.
2. The capture needs to be set to non-color (The camera does support color)

If Spinview is not run (and stopped) before the ROS driver, or the ROS driver is set to use color (regardless of any other parameters), the driver crashes with the following error:

` Spinnaker could not open the device, try reconnecting the device.`

Not only that, but the crash appears to extend to crashing the entire system.  The GUI of the entire OS becomes unresponsive, and if running in a TTY, the TTY begins to get flooded by the following messages:

```
Dec 10 18:40:05 rpi kernel: [  778.197866] raspberrypi-clk raspberrypi-clk: Failed to get pllb frequency: -12
Dec 10 18:40:05 rpi kernel: [  778.213628] raspberrypi-clk raspberrypi-clk: Failed to change pllb frequency: -12
Dec 10 18:40:05 rpi kernel: [  778.213637] raspberrypi-clk raspberrypi-clk: Failed to get pllb frequency: -12
Dec 10 18:40:05 rpi kernel: [  778.289829] raspberrypi-clk raspberrypi-clk: Failed to change v3d frequency: -12
Dec 10 18:40:05 rpi kernel: [  778.289843] raspberrypi-clk raspberrypi-clk: Failed to get v3d frequency: -12
```

The error, as far as I can find in syslog, is the following:
```
Dec 10 17:46:19 rpi kernel: [  554.427395] xhci_hcd 0000:01:00.0: Ring expansion failed
Dec 10 17:46:19 rpi kernel: [  554.429241] usb 2-2: usbfs: usb_submit_urb returned -12
Dec 10 17:46:19 rpi kernel: [  554.609879] vc4_plane_set_blank: firmware call failed. Please update your firmware
Dec 10 17:46:19 rpi kernel: [  554.610098] WARNING: CPU: 3 PID: 7 at drivers/gpu/drm/vc4/vc4_firmware_kms.c:401 vc4_plane_set_blank+0x134/0x140 [vc4]
Dec 10 17:46:19 rpi kernel: [  554.610102] Modules linked in: rfcomm aes_neon_blk crypto_simd cryptd aes_arm64 bnep hci_uart btbcm serdev bluetooth ecdh_generic fuse joydev evdev vc4 brcmfmac snd_bcm2835(C) drm_kms_helper cec brcmutil snd_seq_midi snd_soc_core snd_seq_midi_event snd_compress snd_rawmidi snd_pcm_dmaengine cfg80211 snd_pcm snd_seq v3d bcm2835_codec(C) v4l2_mem2mem gpu_sched rfkill snd_seq_device bcm2835_v4l2(C) videobuf2_dma_contig snd_timer syscopyarea bcm2835_mmal_vchiq(C) v4l2_common sysfillrect videobuf2_vmalloc sysimgblt videobuf2_memops fb_sys_fops videobuf2_v4l2 drm videobuf2_common raspberrypi_hwmon snd hwmon drm_panel_orientation_quirks videodev media i2c_bcm2835 vc_sm_cma(C) rpivid_mem uio_pdrv_genirq uio fixed sch_fq_codel i2c_dev ip_tables x_tables ipv6 crc_ccitt
Dec 10 17:46:19 rpi kernel: [  554.610315] CPU: 3 PID: 7 Comm: kworker/u8:0 Tainted: G         C        4.19.86-v8-24 #24
Dec 10 17:46:19 rpi kernel: [  554.610322] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
Dec 10 17:46:19 rpi kernel: [  554.610342] Workqueue: events_unbound commit_work [vc4]
Dec 10 17:46:19 rpi kernel: [  554.610350] pstate: 00000005 (nzcv daif -PAN -UAO)
Dec 10 17:46:19 rpi kernel: [  554.610363] pc : vc4_plane_set_blank+0x134/0x140 [vc4]
Dec 10 17:46:19 rpi kernel: [  554.610376] lr : vc4_plane_set_blank+0x134/0x140 [vc4]
Dec 10 17:46:19 rpi kernel: [  554.610378] sp : ffffff800805bc50
Dec 10 17:46:19 rpi kernel: [  554.610381] x29: ffffff800805bc50 x28: ffffffac8ba66000 
Dec 10 17:46:19 rpi kernel: [  554.610388] x27: ffffffe9b701c070 x26: ffffffe96bb0fa00 
Dec 10 17:46:19 rpi kernel: [  554.610394] x25: ffffffac4c63a3f0 x24: ffffffac4c4c2310 
Dec 10 17:46:19 rpi kernel: [  554.610400] x23: ffffffac4c63a328 x22: 0000000000000000 
Dec 10 17:46:19 rpi kernel: [  554.610407] x21: ffffffe9b250e480 x20: ffffffac8ba686c8 
Dec 10 17:46:19 rpi kernel: [  554.610413] x19: 00000000fffffff4 x18: 0000000000000000 
Dec 10 17:46:19 rpi kernel: [  554.610420] x17: 0000000000000001 x16: 0000000000000007 
Dec 10 17:46:19 rpi kernel: [  554.610426] x15: 0000000000000000 x14: 69662072756f7920 
Dec 10 17:46:19 rpi kernel: [  554.610433] x13: 6574616470752065 x12: 7361656c50202e64 
Dec 10 17:46:19 rpi kernel: [  554.610440] x11: 656c696166206c6c x10: 616320657261776d 
Dec 10 17:46:19 rpi kernel: [  554.610446] x9 : 726966203a6b6e61 x8 : 6c625f7465735f65 
Dec 10 17:46:19 rpi kernel: [  554.610453] x7 : ffffffac8bb85000 x6 : 0000003d30790000 
Dec 10 17:46:19 rpi kernel: [  554.610461] x5 : 0000000000000001 x4 : ffffffac8ab318dc 
Dec 10 17:46:19 rpi kernel: [  554.610468] x3 : ffffffe9bbfaa098 x2 : 0000000000000007 
Dec 10 17:46:19 rpi kernel: [  554.610474] x1 : 7fdde3a69068f000 x0 : 0000000000000000 
Dec 10 17:46:19 rpi kernel: [  554.610481] Call trace:
Dec 10 17:46:19 rpi kernel: [  554.610498]  vc4_plane_set_blank+0x134/0x140 [vc4]
Dec 10 17:46:19 rpi kernel: [  554.610510]  vc4_plane_atomic_update+0x44/0x50 [vc4]
Dec 10 17:46:19 rpi kernel: [  554.610573]  drm_atomic_helper_commit_planes+0xf4/0x220 [drm_kms_helper]
Dec 10 17:46:19 rpi kernel: [  554.610587]  vc4_atomic_complete_commit+0x60/0x280 [vc4]
Dec 10 17:46:19 rpi kernel: [  554.610599]  commit_work+0x20/0x30 [vc4]
Dec 10 17:46:19 rpi kernel: [  554.610616]  process_one_work+0x1c4/0x420
Dec 10 17:46:19 rpi kernel: [  554.610620]  worker_thread+0x50/0x408
Dec 10 17:46:19 rpi kernel: [  554.610626]  kthread+0x108/0x138
Dec 10 17:46:19 rpi kernel: [  554.610633]  ret_from_fork+0x10/0x1c
Dec 10 17:46:19 rpi kernel: [  554.610638] ---[ end trace b99e203732373c60 ]---
```

I realize that this may be out of your capacity to help me debug, but if you have any idea of what might be going on, I'd appreciate some guidance, as this is very confusing to me.

mithunvarma, 2019-12-12:
@fultonms Can you share your launch file and camera config file

Is the param color set to false in launch file and config file?

We didn’t test the code on any raspberry variant before.

Based on your description, 

1. If total system memory is 4GB, you shouldn’t be setting usbfs_memory_mb to 4096 for one application.

Read more on 
https://www.flir.com/support-center/iis/machine-vision/application-note/understanding-usbfs-on-linux/

2. Ideally you don’t need to run spinview before running this driver. 
To see if camera is detected, check in $lsusb
Optionally you can run Acquisition sample binary installed by spinnaker sdk to check if spinnaker sdk is installed properly and to test image acquisition from camera.

I recommend setting usbfs limit to 1024 and param color to false and run the driver.

What FPS did you set in soft FPS param ?

Hope it helps

fultonms, 2019-12-12:
@mithunvarma Hi, thanks for your response. 

I am testing using the acquisition_node.launch and test_param.yaml launch file as they are in the repository at this point, with the only change being my camera's serial number. 

When I am testing without color, the color param is set to false. When I have tested with color, I have done so using the standard ROS method of passing parameters to the launch file, color:=true.  I have also tried setting the parameter default in the launch file, with the same results. 

A correction on my previous comment, the exact error message is as follows:

```
Terminate called after throwing and instance of 'Spinnaker :: Exception'
    what() : Spinnaker: Could not start acquisition. Please try reconnecting the device [-1010]
```

1. I've also tried at 1024, 2048, etc.  I'll modify my configuration to set the usbfs_memory_mb to 1024 in the future though, thanks.

2. That's what concerns me. When running this driver on other computers, it has worked fine. However, for the Raspberry Pi, the driver does not seem to work dependably without running Spinnaker first.  While doing some testing to confirm this, it did actually work (with no color) without running Spinnaker first, so maybe that problem isn't a consistent thing. I think the more persistent issue is the crashes when related to running the camera with color.

My camera is detected in lsusb.  When running with color param set false, the camera does work, however I need color for my applications.  The FPS is set to 20. 

Thanks for your help

PushyamiKaveti, 2019-12-12:
@fultonms Does the color mode work fine on other computers? If so, I think that the problem is with running color mode on Raspberry Pi. May be the Raspeberry Pi's USB is unable to handle the data coming through. You could may be try setting a lower FPS and see if that works?

vik748, 2019-12-13:
@fultonms Could you try a couple more things.  With color set to True can you try reducing the resolution by setting binning to 2 and/or reducing the soft_framerate to 5 or 10.  This should help us determine if it is a driver / spinnaker issue or a system limitation.

Also another thing that comes to mind is that in our experience when the power to the camera is unstable it can throw errors like this.  Can you connect the camera through a powered USB hub or give it external power via GPIO?

fultonms, 2019-12-13:
@PushyamiKaveti Yeah, I think that's the issue as well. I'm not expecting that this will work as well as it could on another system, just trying to get some level of performance out of this.

@vik748 I will try those things and report back. Thanks for the help!

fultonms, 2019-12-13:
@vik748 You are a lifesaver. Reducing the resolution by setting the binning to 2 was sufficient to get the camera working, at 20 FPS, in full color. Thank you so much! 

I had not been trying resolution reduction, but thinking about the problem now, it really should have been one of the first things I tried.

closed 2019-12-13 by fultonms

vik748, 2019-12-13:
@fultonms I still suspect that the root cause might be power.  When in full color and full resolution, the camera might have a spike in power draw causing issues. Lowering either greyscale, resolution or framerate might be helping it reduce that power spike.

fultonms, 2019-12-26:
@vik748 I think you might be right. I'm doing some testing, but progress is being made. 

A somewhat unrelated question: have you considered running cameras in a master/slave setup across multiple computers? So, several networked computers, one with the master cam, the others with slave cameras, all cameras connected via GPIO. The current status of this package does not allow for it. Is that a limitation in the Spinnaker SDK, or no?

vik748, 2019-12-27:
We haven't tried it but should be possible.  What you'd need to do is setup each computer as a single camera system.  Have the 'master' as a regular system and have all other be setup to 'external trigger' as in the external_trigger branch. We haven't done this, so it might need a few tweaks in the code.

fultonms, 2020-02-11:
@vik748 Thank you! That has been working well so far.

However, after approximately 30 seconds, the externally triggered camera stops capturing frames. It appears to get to the call for acquisition::Capture::get_mat_images and not return from the call. For clarity, it does make it through that call for some period of time (apparently random), but it does fail eventually. Any ideas on what this might be caused by?

shahvi, 2020-02-12:
How are the cameras powered?  We have seen if there is any power issues the cameras misbehave.  We have found that using a powered USB hub helps with stability.

fultonms, 2020-02-12:
I experience the same issue whether I am powering the camera off the USB3 from the Raspberry Pi, or off Line 3 (Pin 1), for which I have to switch my trigger source to Line 2 ( Pin 3). 

I have not experienced this issue when using SpinView on a laptop with either trigger/power setup. I'll be testing it with SpinView on the Pi4 tomorrow and will let you know what happens.

However at the moment, I don't think that it's a power issue, but I'm still testing to see if that could be the case.

The failure looks similar to the way that the camera stops capturing images if it does not have a trigger signal. However, I have checked with an oscilloscope to check that the trigger signal is being sent,  so I think my debugging is just not in-depth enough?

fultonms, 2020-02-13:
Having looked a little more into it, I see the same issue with Spinview on the RPI, both powered and unpowered. 

By looking a little more into it, I'm sometimes seeing warnings that frames are being skipped, then a warning: "Image incomplete with image status 9!" followed by a bad_alloc exception. Sometimes (without changing configs) I also see that warning and bad_alloc exception without first seeing the frame skip warnings, and still other times I see nothing, but the frame_grabbing function does not return.