neufieldrobotics/spinnaker_sdk_camera_driver #64: How to set the camera time counter to the system time at initialization?

olamarre, 2020-01-16:
Hi! Does this package set the device (camera) time at initialization? If so, where does it do it? I could not find it in this ROS package. Alternatively, how could I synchronize the camera with the system (computer) when starting the camera nodelet?

I noticed that the raw image timestamps (obtained using the `GetTimeStamp()` method) were in the past, and once they are properly adjusted, I would like to use them with the outgoing ROS messages instead of the current method, which uses ros::Time::Now(). I am using a FLIR Blackfly S BFS-U3-23S3C.

Thanks!

vik748, 2020-01-21:
@olamarre we did try doing that in the past but then decided against it. In this package we use a software trigger which is sent to the camera and then immediately log the time using now().  I believe this method should have an accuracy within 10ms of the true trigger time, however it would depend on the USB bus load and CPU utilization. If you need better accuracy, you'd be better of using an external trigger source and then logging its time.
If you do find a more elegant solution, please do share with us.

olamarre, 2020-01-21:
@vik748 thanks for the reply!
I am actually using 2 of those cameras in a stereo fashion, and one of them is configured to trigger the other one. The current methodology (using ros::Time::now() ) indeed works, but since we have 2 nodelets running independently (one for each camera), the timestamp error between two images can sometime become significant (we have a CPU with rather limited capabilities and a single USB port for both cameras). Using the `approximate_sync` ROS parameter works, but sometimes the timestamp error causes other issues in our vision pipeline, which is why we would like to try setting the camera's clock at startup.

Could you advice how you previously set the camera clock time at startup? I saw in the Spinnaker API that there is a `TimestampReset` method (documented [here](http://softwareservices.ptgrey.com/Spinnaker/latest/class_spinnaker_1_1_camera.html#a8a2596bb01c00aa231925b08884f8844)), but I'm unsure if it's the appropriate method or even how to properly use it (does it take in any argument, i.e. the CPU's current clock)?

Thank you very much! We'll be happy to report back positive results here.

shahvi, 2020-01-22:
@olamarre if you use our package as intended, you should be able to have a master slave setup with the master being software triggered and the master triggering the slave using hardware trigger via GPIO. Both the cameras then run in the same nodelet, using the timestamps of the software trigger and then publishing the exact same time stamp to both the topics.
This is described in the readme at https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver#multicamera-master-slave-setup

Hope this helps.

closed 2020-06-29 by shahvi