neufieldrobotics/spinnaker_sdk_camera_driver #66: Issue Running 2x Blackfly S

jimmyw404, 2020-02-04:
I have two Blackfly S cameras. I'm able to connect them all to a network with my laptop and stream images from them by specifying one cam_id at a time in the test_params.yaml file.

When I have two cameras the cam_ids test_params.yaml file I am unable to stream the data from the cameras.

Attached is a zip file with my launch file, the test_params.yaml file and a log. The only modifications I made are removing this assert message ROS_ASSERT_MSG(frameID_ == 0  and adding logging around the GetNextFrame() call to show that's where the Spinnaker exception is being thrown.

I see in https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver there are instructions on hooking a master/slave setup in the GPIO. I have not done this and don't need highly synchronized frames, I'm just looking to stream data from them and it's OK if they aren't synchronized.

Also note: I'm unable to stream two cameras from spinview, ebusplayer or software I wrote to interface with the spinnaker SDK.
[MultiCamPackage.zip](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/files/4156471/MultiCamPackage.zip)

Any ideas I can try out?

shahvi, 2020-02-05:
First step is to run two cams simulateously on Spinview. If that doesn't work, nothing else will. You might be running into bandwidth issues, try using binning to reduce resolution or reducing the frame rate.

Our package is not setup to use 2 cameras without the master / slave setup you would need to modify the code to make it work.

jimmyw404, 2020-02-05:
> First step is to run two cams simulateously on Spinview. If that doesn't work, nothing else will. You might be running into bandwidth issues, try using binning to reduce resolution or reducing the [...]

Thank you for your reply. When I reduced my image size on both cameras by increasing horizontal / vertical binning to 4 I was able to stream both cameras in spinview / ebusplayer / my software and the spinnaker SDK.

I'm still working through why I can't stream the full resolution, but it's clearly not a spinnaker SDK issue.

closed 2020-02-05 by jimmyw404