neufieldrobotics/spinnaker_sdk_camera_driver #70: Does this camera driver work with ROS Melodic?

michael5511b, 2020-03-04:
I have ROS Melodic running on Ubuntu 18.04. Since "ros-kinetic-cv-bridge" and "ros-kinetic-image-transport" in the prerequisites are not supported anymore, I substituted them with the melodic versions, which resulted in numerous catkin_make errors, such as not being able to find the "opencv" folder, after renaming the "opencv4" folder in the usr/include directory to "opencv", it resulted in more errors such as not being able to include <cv.h>.

Is there a workaround for this? Is there an updated version for this driver? Or is using Kinetic the only way?

Thank you!

shahvi, 2020-03-04:
So far we have only tested this on Kinetic, so not sure about issues with Melodic.  However, I believe Melodic comes with Opencv 3.x, so if your error message shows opencv4 it could be a conflict with a version of Opencv you installed separately from ROS. 
Also we haven't tried this code with the Chameleon model, but in theory it should be able to work with any camera that works with Spinnaker's SpinView app.
Please paste your error messages here so we might see what is going on...

ayuguchi, 2020-03-19:
FYI, I confirmed this code can work on ROS Melodic using a FLIR camera.

Janphr, 2020-04-02:
I'm using melodic on 18.04 as well and getting this error:

`[ INFO] [1585844738.262089787]: Creating system instance...
[ INFO] [1585844738.263719677]: Retreiving list of cameras...
[ INFO] [1585844741.656429504]: Numer of cameras found: 1
[ INFO] [1585844741.656512462]:  Cameras connected: 1
[ INFO] [1585844741.656626943]:   -2CDDA333400E
[ WARN] [1585844741.656732374]:    Camera 20135950 not detected!!!
[FATAL] [1585844741.656821839]: ASSERTION FAILED
	file = /home/janphr/Praktikum_Zft/ws/src/spinnaker_camera_driver/src/capture.cpp
	line = 317
	cond = cams.size()
	message = 
[FATAL] [1585844741.656889565]: None of the connected cameras are in the config list!
[FATAL] [1585844741.656946824]: 

[FATAL] [1585844741.782007163]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'`

I'm using a Blackfly S GigE. SpinView works fine.
Any idea what might cause this?

Thanks!

SlouchyCat, 2020-04-11:
> FYI, I confirmed this code can work on ROS Melodic using a FLIR camera.

Hi @shahvi and @ayuguchi ,

I cannot get this working on Melodic. Here is my error:

`SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [18012444]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1536
 * /acquisition_node/image_width: 2048
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 18012444
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [31628]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 53b0ceae-7ba3-11ea-9de2-94e6f7c21e33
process[rosout-1]: started with pid [31639]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [31646]
process[acquisition_node-3]: started with pid [31647]
[ INFO] [1586575200.930454169]: Initializing nodelet with 8 worker threads.
[ INFO] [1586575200.974120641]: Initializing nodelet
[ INFO] [1586575200.974997940]: [ OK ] USB memory: 1000 MB
[ INFO] [1586575200.975096490]: *** PARAMETER SETTINGS ***
[ INFO] [1586575200.975140196]: ** Date = 20200411
[ INFO] [1586575200.975602631]:   Save path set via parameter to: /home/lifeline
[ INFO] [1586575200.975632901]:   Camera IDs:
[ INFO] [1586575200.976008678]:     18012444
[ INFO] [1586575200.976343571]:   Camera Aliases:
[ INFO] [1586575200.976360056]:     18012444 >> cam0
[ INFO] [1586575200.977003973]:   Unique time stamps for each camera: false
[ INFO] [1586575200.977330360]:   color set to: false
[ INFO] [1586575200.977634905]:   Exporting images to ROS: true
[ INFO] [1586575200.977907535]:   Showing live images setting: false
[ INFO] [1586575200.978214523]:   Showing grid-style live images setting: false
[ INFO] [1586575200.978519801]:   Max Rate Save Mode: false
[... 16 lines cut ...]
[ INFO] [1586575200.983223830]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1586575200.983234622]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1586575200.983330352]: Creating system instance...
[ INFO] [1586575200.985263724]: Retreiving list of cameras...
[ INFO] [1586575203.423048827]: Numer of cameras found: 1
[ INFO] [1586575203.423160470]:  Cameras connected: 1
[ INFO] [1586575203.423359704]:   -18012444
[ INFO] [1586575203.498812230]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1586575203.499843281]: *** FLUSH SEQUENCE ***
[ INFO] [1586575203.499872748]: Initializing cameras...
[ INFO] [1586575204.268089470]: Deinitializing cameras...
[ INFO] [1586575204.272803693]: All cameras deinitialized.
[ INFO] [1586575206.273022376]: Initializing cameras...
[FATAL] [1586575206.478705421]: Error: Spinnaker: GenICam::OutOfRangeException= Value = 1 must be equal or greater than Min = 2. : OutOfRangeException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 79) [-2002]
[ WARN] [1586575206.478737825]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1586575206.661764244]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'TriggerSoftware' while calling 'TriggerSoftware.Execute()' (file 'CommandT.h', line 61) [-2006]
[acquisition_node-3] process has died [pid 31647, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/lifeline/.ros/log/53b0ceae-7ba3-11ea-9de2-94e6f7c21e33/acquisition_node-3.log].
log file: /home/lifeline/.ros/log/53b0ceae-7ba3-11ea-9de2-94e6f7c21e33/acquisition_node-3*.log
[vision_nodelet_manager-2] process has died [pid 31646, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/lifeline/.ros/log/53b0ceae-7ba3-11ea-9de2-94e6f7c21e33/vision_nodelet_manager-2.log].
log file: /home/lifeline/.ros/log/53b0ceae-7ba3-11ea-9de2-94e6f7c21e33/vision_nodelet_manager-2*.log
`

Any help would be greatly appreciated!

ayuguchi, 2020-04-11:
@SlouchyCat I recommend before starting to run a camera on this repo, you should confirm to work it through Spinview.

SlouchyCat, 2020-04-11:
> @SlouchyCat I recommend before starting to run a camera on this repo, you should confirm to work it through Spinview.

It is working in Spinview. Im using Blackfly S BFS-U3-13Y3C. Can u also post your camera model for the benefit of others?

Janphr, 2020-04-11:
I got it to work by simply programming a driver from scratch for my Blackfly S BFS-PGE-16S2C GigE. My model returns the device id instread of the serial number for example, that was one of the bugs and i needed some thing differently implemented anyway.
your error was caused by the combination of binning (1 or 2) and the pixel format. on my model binning 1 works only for 1 channel format (mono) and ISP needs to be enabled. binning 2 for everything else...

closed 2020-06-29 by shahvi