neufieldrobotics/spinnaker_sdk_camera_driver #74: Can multiple cameras be connected as slaves to a signal generator

mihir7714, 2020-03-13:
Can a signal generator (generating a standard pulse wave) be used with multiple cameras as slaves thereby making use of no master cam using this SDK?

I am trying to connect 3 cameras in sync but without having one of the cameras as a master. The setup instead will use a signal generator as a trigger for the 3 cameras.

shahvi, 2020-03-15:
Yes, this should be doable.  It is a feature we have added recently. You should be able to use the code in the branch called external_trigger. Hope this helps. Let us know if you have issues using it.

Cheers,
Vik

mihir7714, 2020-03-16:
Thanks Vik. I am trying to build the external-trigger code but getting the following CMake Error. 

Could not find a package configuration file provided by "trigger_msgs" with any of the following names:
trigger_msgsConfig.cmake
trigger_msgs-config.cmake

shahvi, 2020-03-16:
Try removing trigger_msgs from CMakeLists.txt under catkin required components (line 27).  The idea is that if you have the package it would use it otherwise it won't.

mihir7714, 2020-03-16:
Yes I did that, but it then gives me the following error:

/home/demo/catkin_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:1172:30: error: ‘sync_message_queue_vector_’ was not declared in this scope
         if(img_q->empty() || sync_message_queue_vector_.at(cam_no).empty()){
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/demo/catkin_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:1182:43: error: ‘sync_message_queue_vector_’ was not declared in this scope
         if (abs((int)img_q->size() - (int)sync_message_queue_vector_.at(cam_no).size()) > 100){
                  
Invoking "make -j8 -l8" failed

shahvi, 2020-03-16:
This is super wierd.  I just downloaded the repo fresh on a computer and removed line 27 from CMakeLists and was able to compile fine. Also, sync_message_queue_vector_ is declared in the capture.h file, so that error doesn't make sense.  Can you try starting from a clean slate. commenting line 27 and compiling.

mihir7714, 2020-03-16:
Yes tried it. Works! Thanks Vik. 
I am trying to generate live images through "rqt" but no images are seen, gray screen. While it says **ACQUISITION** on the terminal there are no updates after that to show images are being taken.
There is a particular warning about trigger_msgs. 

Pasting it below, is this expected to work despite that warning? is one of the acquisition.launch parameters configured wrong?

process[acquisition_node-2]: started with pid [25670]
[ INFO] [1584371320.197257931]: [ OK ] USB memory: 1500 MB
[ INFO] [1584371320.197986024]: *** PARAMETER SETTINGS ***
[ INFO] [1584371320.198020138]: ** Date = 20200316
[ INFO] [1584371320.198304735]:   Save path set via parameter to: /home/demo
[ INFO] [1584371320.198337380]:   Camera IDs:
[ INFO] [1584371320.198645886]:     17416721
[ INFO] [1584371320.198668764]:     17416723
[ INFO] [1584371320.198964673]:   Camera Aliases:
[ INFO] [1584371320.198986965]:     17416721 >> cam0
[ INFO] [1584371320.198997755]:     17416723 >> cam1
[ INFO] [1584371320.199284913]:   External trigger: true
**[ WARN] [1584371320.199306197]:   Using 'external_trigger'. Trigger msgs package not found, will use machine timestamps**
[ INFO] [1584371320.199575928]:   Unique time stamps for each camera: false
[ INFO] [1584371320.199814226]:   color set to: true
[ INFO] [1584371320.200046860]:   Exporting images to ROS: true
[ INFO] [1584371320.200267405]:   Showing live images setting: true
[ INFO] [1584371320.200450630]:   Showing grid-style live images setting: true
[ INFO] [1584371320.200704065]:   Max Rate Save Mode: false
[ INFO] [1584371320.200906774]:   Displaying timing details: true
[ INFO] [1584371320.201119020]:   No. of images to skip set to: 20
[ INFO] [1584371320.201335687]:   Init sleep delays set to : 1.00 sec
[ WARN] [1584371320.201547525]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1584371320.201755497]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1584371320.201965135]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1584371320.202170359]:   Binning set to: 2
[ INFO] [1584371320.202378549]:   Using Software rate control, rate set to: 30
[ INFO] [1584371320.202586208]:   Saving images set to: 0
[ INFO] [1584371320.202767523]:     save_type set as: bmp
[ INFO] [1584371320.203254184]:   tf_prefix set to: 
[ INFO] [1584371320.204002868]:   Camera coeffs not provided correctly, camera info messges will not be published.
[ INFO] [1584371320.204027871]: Creating system instance...
[ INFO] [1584371320.205332037]: Retreiving list of cameras...
[ INFO] [1584371321.269132461]: Numer of cameras found: 2
[ INFO] [1584371321.269229690]:  Cameras connected: 2
[ INFO] [1584371321.269373965]:   -17416721
[ INFO] [1584371321.269458658]:   -17416723
[ INFO] [1584371321.310479005]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1584371321.311123284]: *** FLUSH SEQUENCE ***
[ INFO] [1584371321.311143590]: Initializing cameras...
[ INFO] [1584371323.790295136]: Deinitializing cameras...
[ INFO] [1584371324.204446857]: All cameras deinitialized.
[ INFO] [1584371326.204972989]: Initializing cameras...
[ INFO] [1584371328.389388063]: *** ACQUISITION ***

shahvi, 2020-03-16:
First step to debug is making sure that your hardware setup work with SpinView with the same external trigger. Have you tried that?

closed 2020-06-29 by shahvi