neufieldrobotics/spinnaker_sdk_camera_driver #75: Encountering Spinnaker SDK issue: Spinnaker: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]

irispete, 2020-04-08:
**System Description**
Camera Name and Model: FLIR Blackfly S BFS-U3-88S6M
Operating System: Ubuntu 16.04
Spinnaker Version: 1.8.0.64, also tried 1.27.0.48
ROS Version: n/a

***Computer details***
Processor: Nvidia Jetson TX2
RAM: 8 Gb

***Multiple Cameras Only***
How is the triggering setup? Master/Slave or all slaves or all software triggered?
- software triggered

For USB 3.0 cameras:
On same USB 3.0 bus or different buses?
Is the USB 3.0 hub externally powered?
- 3 cameras plugged into an external USB 3.0 Hub that is an externally powered hub

**Do the cameras work with SpinView?**
Haven't tried this

**Describe the bug**
When I run multiple cameras at a rate of 10 fps or greater I encounter the error: Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]
Sometimes this happens right way, some times it can run as long as 3 hours before I encounter this error. I have only tried with 2 and 3 cameras. The higher the frame rate or the higher the number of cameras, the quicker I will usually see the problem.
I can duplicate the issue by running the AcquisitionMultipleCamera sample application, commenting out the code that saves the images to a file to speed it up, and increasing the number of loop iterations so it will run for a long duration.
When I use Spinnaker 1.8 I see the buffer error, when I use version 1.27 it usually just hangs the application rather than returning an error. I have not tried lower frame rates to see if there is a cutoff, I need to get at least 10 fps, 15 if possible.
The resolution of those cameras is 4096 x 2160

Here is the output of the device data from the sample app:
Application build date: Apr  8 2020 15:42:54

Spinnaker library version: 1.27.0.48

Number of cameras detected: 3

Running example for all cameras...

*** DEVICE INFORMATION ***

Printing device information for camera 0...

DeviceID : 19457313
DeviceSerialNumber : 19457313
DeviceVendorName : FLIR
DeviceModelName : Blackfly S BFS-U3-88S6M
DeviceType : U3V
DeviceDisplayName : FLIR
DeviceAccessStatus : OpenReadWrite
DeviceVersion : 1803.0.36.0
DeviceDriverVersion : none : 0.0.0.0
DeviceUserID :
DeviceIsUpdater : 0
DeviceInstanceId : 0128E521
DeviceLocation :
DeviceCurrentSpeed : SuperSpeed
GUIXMLLocation : Device
GUIXMLPath : Input.xml
GenICamXMLLocation : Device
GenICamXMLPath :
DeviceU3VProtocol : 1

Printing device information for camera 1...

DeviceID : 19421716
DeviceSerialNumber : 19421716
DeviceVendorName : FLIR
DeviceModelName : Blackfly S BFS-U3-88S6M
DeviceType : U3V
DeviceDisplayName : FLIR
DeviceAccessStatus : OpenReadWrite
DeviceVersion : 1803.0.36.0
DeviceDriverVersion : none : 0.0.0.0
DeviceUserID :
DeviceIsUpdater : 0
DeviceInstanceId : 01285A14
DeviceLocation :
DeviceCurrentSpeed : SuperSpeed
GUIXMLLocation : Device
GUIXMLPath : Input.xml
GenICamXMLLocation : Device
GenICamXMLPath :
DeviceU3VProtocol : 1

Printing device information for camera 2...

DeviceID : 19457314
DeviceSerialNumber : 19457314
DeviceVendorName : FLIR
DeviceModelName : Blackfly S BFS-U3-88S6M
DeviceType : U3V
DeviceDisplayName : FLIR
DeviceAccessStatus : OpenReadWrite
DeviceVersion : 1803.0.36.0
DeviceDriverVersion : none : 0.0.0.0
DeviceUserID :
DeviceIsUpdater : 0
DeviceInstanceId : 0128E522
DeviceLocation :
DeviceCurrentSpeed : SuperSpeed
GUIXMLLocation : Device
GUIXMLPath : Input.xml
GenICamXMLLocation : Device
GenICamXMLPath :
DeviceU3VProtocol : 1

**Error Messages**
"Failed waiting for EventData on NEW_BUFFER_DATA event [-1011]"

**Screenshots**
If applicable, add screenshots to help explain your problem.

shahvi, 2020-04-08:
@irispete this repo is for the ROS driver for these cameras. Are you using these with ROS?

Sounds like you are running 3 cameras on the same USB bus and might be running into bandwidth limits. I'd suggest you take a look at spinnaker's guide on multicameras at https://www.flir.com/support-center/iis/machine-vision/application-note/usb-3.1-multiple-camera-setup

irispete, 2020-04-08:
ahh, ok, my mistake. I am not using ROS to acquire the camera images, especially with that sample app. I have already read over that doc and everything else I could find. Thanks.

closed 2020-04-08 by irispete