neufieldrobotics/spinnaker_sdk_camera_driver #76: Cannot detect Camera

atifkarim, 2020-04-10:
***System Description***
Spinnaker Version: **spinnaker-2.0.0.109-amd64**
ROS Version: **Kinetic**

***Computer details***
Processor: Intel® Core™ i5-7200U CPU @ 2.50GHz × 4 
RAM: 15,5 GiB

Cameras work with **SpinView**

**Bug**
The package is built up successfully but while running the command it is not working by saying that camera is not detected and camera id is not in the config list.
```
roslaunch spinnaker_sdk_camera_driver acquisition.launch
```

**Error Messages**
Giving me the following error message --
```
[ WARN] [1586515189.562547025]:    Camera 18522234 not detected!!!
[FATAL] [1586515189.562635898]: ASSERTION FAILED
	file = /home/atif/spinnaker_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp
	line = 317
	cond = cams.size()
	message = 
[FATAL] [1586515189.562889674]: None of the connected cameras are in the config list!
[FATAL] [1586515189.562956459]: 

[FATAL] [1586515189.713728360]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 6468, exit code -5, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/atif/.ros/log/98d89c82-7b17-11ea-b98a-88d7f69a3204/vision_nodelet_manager-2.log].
log file: /home/atif/.ros/log/98d89c82-7b17-11ea-b98a-88d7f69a3204/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 6469, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/atif/.ros/log/98d89c82-7b17-11ea-b98a-88d7f69a3204/acquisition_node-3.log].
log file: /home/atif/.ros/log/98d89c82-7b17-11ea-b98a-88d7f69a3204/acquisition_node-3*.log

```
**Steps were taken**

I have gone into the file ```params/test_params.yaml``` and changed the first few lines:
```
cam_ids:
- 18522234
cam_aliases:
- cam0
master_cam: 18522234
```

**Information Screenshots**

I have changed the **cam_ids** and **master_cam** by using the information of the [image]

shahvi, 2020-04-11:
When you run the code, it should show all the cameras that are detected in the log.  Does it show anything there? If you see it in SpinView, it should work in the ROS node.  This is super weird, haven't come across this before.  @mithunvarma any ideas?

mithunvarma, 2020-04-11:
@shahvi No, I'm seeing this for first time.
@atifkarim could you post the full output from the terminal
thanks

SlouchyCat, 2020-04-11:
Dear all,

Im getting the same issue, no camera found. Im using latest Spinnaker SDK.
[image]

The full console error msg below:
`SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [18012444]
 * /acquisition_node/color: True
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1024
 * /acquisition_node/image_width: 1280
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 18012444
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6581]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3e46bdde-7bb8-11ea-8855-94e6f7c21e33
process[rosout-1]: started with pid [6592]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [6599]
process[acquisition_node-3]: started with pid [6600]
[ INFO] [1586584184.429629831]: Initializing nodelet with 8 worker threads.
[ INFO] [1586584184.531069287]: Initializing nodelet
[ INFO] [1586584184.531924743]: [ OK ] USB memory: 1000 MB
[ INFO] [1586584184.531961851]: *** PARAMETER SETTINGS ***
[ INFO] [1586584184.531975307]: ** Date = 20200411
[ INFO] [1586584184.532468081]:   Save path set via parameter to: /home/lifeline
[ INFO] [1586584184.532500317]:   Camera IDs:
[ INFO] [1586584184.532867492]:     18012444
[ INFO] [1586584184.533181001]:   Camera Aliases:
[ INFO] [1586584184.533205031]:     18012444 >> cam0
[ INFO] [1586584184.533850090]:   Unique time stamps for each camera: false
[ INFO] [1586584184.534128858]:   color set to: true
[ INFO] [1586584184.534458908]:   Exporting images to ROS: true
[ INFO] [1586584184.534785119]:   Showing live images setting: false
[ INFO] [1586584184.535048223]:   Showing grid-style live images setting: false
[ INFO] [1586584184.535317321]:   Max Rate Save Mode: false
[ INFO] [1586584184.535586977]:   Displaying timing details: false
[ INFO] [1586584184.535804473]:   No. of images to skip set to: 20
[ INFO] [1586584184.536103619]:   Init sleep delays set to : 1.00 sec
[ WARN] [1586584184.536377319]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1586584184.536679088]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1586584184.536934277]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1586584184.537216596]:   Binning set to: 1
[ INFO] [1586584184.537482979]:   Using Software rate control, rate set to: 30
[ INFO] [1586584184.537731719]:   Saving images set to: 0
[ INFO] [1586584184.537965113]:   Camera Intrinsic Paramters:
[ INFO] [1586584184.538152593]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1586584184.538432700]:   Camera Distortion Paramters:
[ INFO] [1586584184.538473921]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1586584184.538724906]:   Camera Rectification Paramters:
[ INFO] [1586584184.538759573]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1586584184.539077803]:   Camera Projection Paramters:
[ INFO] [1586584184.539136265]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1586584184.539150837]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1586584184.539180952]: Creating system instance...
[ INFO] [1586584184.540673318]: Retreiving list of cameras...
[ INFO] [1586584186.991283211]: Numer of cameras found: 1
[ INFO] [1586584186.991469688]:  Cameras connected: 1
[ INFO] [1586584186.991678049]:   -0112D91C
[ WARN] [1586584186.991806768]:    Camera 18012444 not detected!!!
[FATAL] [1586584186.991973482]: ASSERTION FAILED
	file = /home/lifeline/catkin_ws/src/spinnaker_sdk_camera_driver-master/src/capture.cpp
	line = 317
	cond = cams.size()
	message = 
[FATAL] [1586584186.992137957]: None of the connected cameras are in the config list!
[FATAL] [1586584186.992241573]: 

[FATAL] [1586584187.107472750]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 6599, exit code -5, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/lifeline/.ros/log/3e46bdde-7bb8-11ea-8855-94e6f7c21e33/vision_nodelet_manager-2.log].
log file: /home/lifeline/.ros/log/3e46bdde-7bb8-11ea-8855-94e6f7c21e33/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 6600, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/lifeline/.ros/log/3e46bdde-7bb8-11ea-8855-94e6f7c21e33/acquisition_node-3.log].
log file: /home/lifeline/.ros/log/3e46bdde-7bb8-11ea-8855-94e6f7c21e33/acquisition_node-3*.log
`

Could you please help take a look?

mithunvarma, 2020-04-11:
try changing cam_id to 0112D91C

SlouchyCat, 2020-04-11:
> try changing cam_id to 0112D91C

Hi @mithunvarma No thats not working. :(

mithunvarma, 2020-04-11:
> > try changing cam_id to 0112D91C
> 
> Hi @mithunvarma No thats not working. :(

did you set 0112D91C in both master_cam and cam_ids in yaml file ?

could you post the terminal output , if you don't mind.

atifkarim, 2020-04-11:
> @shahvi No, I'm seeing this for first time.
> @atifkarim could you post the full output from the terminal
> 
> Also, could you change the logger to debug in this line,3 of launch file
> you can set ROSCONSOLE_CONFIG_FILE to debug_console.conf
> 
> `  <env name="ROSCONSOLE_CONFIG_FILE" value="$(find spinnaker_sdk_camera_driver)/cfg/std_console.conf"/>`
> [link to launch file](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/521b72139c24e52416a5ed126219c78ee7e4cbca/launch/acquisition.launch#L3)
> 
> thanks

Full output from the terminal --
```
roslaunch spinnaker_sdk_camera_driver acquisition.launch
... logging to /home/atif/.ros/log/26fdeb30-7bd5-11ea-b32d-88d7f69a3204/roslaunch-atif-X556UQK-12559.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://atif-X556UQK:40273/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [18522234]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1536
 * /acquisition_node/image_width: 2048
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 18522234
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [12569]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 26fdeb30-7bd5-11ea-b32d-88d7f69a3204
process[rosout-1]: started with pid [12582]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [12599]
process[acquisition_node-3]: started with pid [12600]
[ INFO] [1586596600.770577304]: Initializing nodelet with 4 worker threads.
[... 40 lines cut ...]
	file = /home/atif/spinnaker_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp
	line = 317
	cond = cams.size()
	message = 
[FATAL] [1586596602.885606154]: None of the connected cameras are in the config list!
[FATAL] [1586596602.885684367]: 

[FATAL] [1586596603.039476916]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 12599, exit code -5, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/atif/.ros/log/26fdeb30-7bd5-11ea-b32d-88d7f69a3204/vision_nodelet_manager-2.log].
log file: /home/atif/.ros/log/26fdeb30-7bd5-11ea-b32d-88d7f69a3204/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 12600, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/atif/.ros/log/26fdeb30-7bd5-11ea-b32d-88d7f69a3204/acquisition_node-3.log].
log file: /home/atif/.ros/log/26fdeb30-7bd5-11ea-b32d-88d7f69a3204/acquisition_node-3*.log

```

I have also changed the logger to debug in the line,3 of launch file
Now the line is -- 
```
<launch>
  <!-- configure console output verbosity mode:debug_console.conf or std_console.conf -->
  <env name="ROSCONSOLE_CONFIG_FILE" value="$(find spinnaker_sdk_camera_driver)/cfg/debug_console.conf"/>
```

**PYTHON CODE**
I have also captured images with the Python code which has installed from [here](https://flir.app.boxcn.net/v/SpinnakerSDK/file/645666377169). If you need I can also give the **console output** of this code which clearly shows **DEVICE ID** and **SERIAL NUMBER**.

shahvi, 2020-04-12:
Basically 0112D91C is the hex for 18012444.  Looks like there is something goofy going on in this version of spinnaker, we haven't tested our code with this version.  Can you try to use version 1.24 and see if that helps.

SlouchyCat, 2020-04-12:
@shahvi Thanks! may I know where to get 1.24 version? I cannot seem to find it

shahvi, 2020-04-12:
@SlouchyCat you can get it from the FLIR's website 'archive' section at https://flir.app.boxcn.net/v/SpinnakerSDK/folder/74729015802

SlouchyCat, 2020-04-12:
@shahvi Thanks! But im running on 18.04 and i saw the archives were made for 16.04. maybe @atifkarim u want to try with 16.04 machine?

atifkarim, 2020-04-12:
> @shahvi Thanks! But im running on 18.04 and i saw the archives were made for 16.04. maybe @atifkarim u want to try with 16.04 machine?

@SlouchyCat  Yes, Ubuntu 16.04

atifkarim, 2020-04-21:
@shahvi  @mithunvarma @SlouchyCat  if you have any solution for this problem would be very helpful for me.

shahvi, 2020-04-21:
looks like v1.26 and v1.27 has packages for Ubuntu 18.04, have you tried either of those?

atifkarim, 2020-04-21:
> looks like v1.26 and v1.27 has packages for Ubuntu 18.04, have you tried either of those?

No, I haven't tried with Ubuntu 18.04 as right now I have to depend on Ubuntu 16.04 machine.

shahvi, 2020-04-21:
so have you tried Spinnaker 1.24 on your ubuntu 16.04 yet?

zgxsin, 2020-04-24:
The same issue here. I am using spinnaker-2.0.0.109-amd64 and Ubuntu18.04 over BFS-U3-13Y3C-C.

`setting /run_id to 9455e286-863f-11ea-ac88-f875a4453ab2
process[rosout-1]: started with pid [5258]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [5264]
process[acquisition_node-3]: started with pid [5266]
[ INFO] [1587741822.335660692] [/vision_nodelet_manager] [Loader::Impl::advertiseRosApi] [7f8a9fb81780]: Initializing nodelet with 12 worker threads.
[ INFO] [1587741822.406524842] [/vision_nodelet_manager] [capture_nodelet::onInit] [7f8a9fb81780]: Initializing nodelet
[ INFO] [1587741822.407303632] [/vision_nodelet_manager] [Capture::Capture] [7f8a9fb81780]: [ OK ] USB memory: 1000 MB
[ INFO] [1587741822.407332951] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]: *** PARAMETER SETTINGS ***
[ INFO] [1587741822.407341572] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]: ** Date = 20200424
[ INFO] [1587741822.407724409] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Save path set via parameter to: /home/guzhou
[ INFO] [1587741822.407742450] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Camera IDs:
[ INFO] [1587741822.408026437] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]: Debug Camera IDs size:19060447
[ INFO] [1587741822.408043182] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]: Debug camera id: 1
[ INFO] [1587741822.408053942] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:     19060447
[ INFO] [1587741822.408330900] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Camera Aliases:
[ INFO] [1587741822.408347023] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:     19060447 >> cam0
[ INFO] [1587741822.408880127] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Unique time stamps for each camera: false
[ INFO] [1587741822.409167748] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   color set to: false
[ INFO] [1587741822.409414343] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Exporting images to ROS: true
[ INFO] [1587741822.409600153] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Showing live images setting: false
[ INFO] [1587741822.409802112] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Showing grid-style live images setting: false
[ INFO] [1587741822.410066484] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Max Rate Save Mode: false
[ INFO] [1587741822.410268836] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Displaying timing details: false
[ INFO] [1587741822.410466516] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   No. of images to skip set to: 20
[ INFO] [1587741822.410639576] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Init sleep delays set to : 1.00 sec
[ WARN] [1587741822.410831809] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1587741822.411065932] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1587741822.411243179] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1587741822.411423714] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Binning set to: 1
[ INFO] [1587741822.411615228] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Using Software rate control, rate set to: 30
[ INFO] [1587741822.411819431] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Saving images set to: 0
[ INFO] [1587741822.411984526] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Camera Intrinsic Paramters:
[ INFO] [1587741822.412017495] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1587741822.412177020] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Camera Distortion Paramters:
[ INFO] [1587741822.412191028] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1587741822.412349382] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Camera Rectification Paramters:
[ INFO] [1587741822.412364113] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1587741822.412559752] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Camera Projection Paramters:
[ INFO] [1587741822.412576738] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1587741822.412584123] [/vision_nodelet_manager] [Capture::read_parameters] [7f8a9fb81780]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1587741822.412593476] [/vision_nodelet_manager] [Capture::Capture] [7f8a9fb81780]: Creating system instance...
[... cut ...]
[acquisition_node-3] process has died [pid 5266, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/guzhou/.ros/log/9455e286-863f-11ea-ac88-f875a4453ab2/acquisition_node-3.log].
log file: /home/guzhou/.ros/log/9455e286-863f-11ea-ac88-f875a4453ab2/acquisition_node-3*.log
^C[rosout-1] killing on exit
^C[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
`

byronwind, 2020-04-27:
1. change cam_ids and master_cam to 0112D91C. 
2. edit capture.cpp , chage cam_ids and master_cam's type to string

atifkarim, 2020-04-27:
> 1. change cam_ids and master_cam to 0112D91C.
> 2. edit capture.cpp , chage cam_ids and master_cam's type to string

Can you please attach here the snippet of the code where to change or just mention the line number

shahvi, 2020-04-27:
@byronwind  thanks for the workaround, in the long term we would need a better way to deal with this across different versions.

shahvi, 2020-04-28:
@atifkarim  @SlouchyCat  @zgxsin @byronwind  #84 should fix this.  Can you pls try it out.

Cheers,
Vik

zgxsin, 2020-04-29:
@shahvi I just tested it, i did not work. nh_pvt_.getParam() cannot read parameters inside  ROS_ASSERT_MSG(nh_pvt_.getParam("cam_ids", cam_id_vec), "..." ) .

shahvi, 2020-04-29:
This seems to be an issue with formatting in your yaml file.

PR #84 (2020-04-28): Fixing bug in spinnaker 2.x.x which causes serial number to be read in…
… hex
Fixes #76
closed 2020-06-29 by shahvi