neufieldrobotics/spinnaker_sdk_camera_driver #78: Warning of passing a NULL to non-pointer

Benefe, 2020-04-15:
While building with catkin_make I get a Warning:

`spinnaker_sdk_camera_driver/src/camera.cpp:5:13: warning: passing NULL to non-pointer argument 1 of ‘Spinnaker::CameraPtr::CameraPtr(long int)’ [-Wconversion-null]
     pCam_ = NULL;
`
Does not effect usability but isn't clean code or functionality.

shahvi, 2020-04-15:
This warning started showing up when we moved to c++11.  I couldn't figure out how to do this in a way to avoid the warning other than suppressing Warnings.  If you can think of a better way, please advise.

Benefe, 2020-04-20:
That's should be an easy fix, by using `nullptr` instead of `NULL`.

shahvi, 2020-04-20:
@Benefe I think I have tried this in the past and it does get rid of the compilation error.  However, if I remember correctly, it throws a runtime error.  Unfortunately I don't have a camera where I am quarantined to try this.  If you have one, can you try and confirm.

closed 2020-06-29 by shahvi