neufieldrobotics/spinnaker_sdk_camera_driver #85: spinview

soulslicer, 2020-05-12:
I have a Blackfly S GigE Camera. I changed the config parameters as suggested  as follows, where I input my camera serial number:

```
cam_ids:
- 19445015
cam_aliases:
- cam0
master_cam: 19445015
skip: 20
delay: 1.0
```

However, it crashes with the error:

```
[FATAL] [1589314567.671373073]: None of the connected cameras are in the config list!
```

I checked and it is because my serial number is actually `2CDDA328B517`  from the `string get_id() { return string(pCam_->GetUniqueID()); }` API of the driver. However, when i replace cam_ids with that, it crashes again with 

```
If cam_aliases are provided, they should be the same number as cam_ids and should correspond in order!
```

shahvi, 2020-05-12:
Duplicate of #76 

#84 should fix this. Can you pls try it out.

Cheers,

soulslicer, 2020-05-12:
Hi, unfortunately i think my blackfly S GigE camera may not work with your default settings. When I run it via spinview, it works, but via your driver it crashes with 

```
[FATAL] [1589318266.006069843]: Unable to execute trigger. Aborting...
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'TriggerSoftware' while calling 'TriggerSoftware.Execute()' (file 'CommandT.h', line 61) [-2006]
```

Is there any way to get this to work with the default settings?

vik748, 2020-05-12:
Your issue sounds similar to #80 , unfortunately we do not have one of this model to try.  Any feedback you can provide will be helpful.