neufieldrobotics/spinnaker_sdk_camera_driver #87: rqt_image_view get normal image, but ros get a flip image (use cv_bridge).

wwstory, 2020-05-17:
**System Description**
Camera Name and Model: PointGreyBFS-PGE-31S4C-C + FA0401C (lens)
Operating System:Ubuntu 18.04
Spinnaker Version:1.27
ROS Version: melodic

***Computer details***
Processor:E3-1275 V5
RAM:32G

**Do the cameras work with SpinView?**
yes, no problem

**my description:**
rqt_image_view get normal image, but ros get a flip image (use cv_bridge).
i have tried #41, but they still is different of 'rqt_image_view' and 'ros get', alway horizontal + vertical flip.
i don't know whose problem it is, spinnaker? lens? cv_bridge?

**other problem:**
frist, i use spinnaker 2.0, it indeed happen these problem: #76 , and i try #84, it can solve how to get cam id. meanwhile, it happen this problem of 'frame id == 0', i also try comment line 59 of 'camera.cpp', it can run only up to 2 minutes.
last, i try #63, set bining=4. it is ok. ( i need use color, so color = true.)
i want to know why it is great in spinview (color, bining=1), do you have any solutions? plz.

plz, help me, thank you.

shahvi, 2020-05-17:
@wwstory if rqt_image_view get it correctly, I would bet that the image being published is fine.  I am not familiar with ros get, but there might be something going on there.  The ROS image msgs don't have a 'flipped or reversed' tag to be causing an issue like this.  Can you provide more details on what exactly you are doing with `ros get`.

Also, what frame rate are you running at?  I believe the issue you are dealing with is about network traffic.  I think spin view probably handles a lot of these exceptions and keeps spitting out images at whatever rate it can.  I don't think it actually delivers the full rate, you can test by trying to record the frames in SpinView and checking the actual rate.  

For our robotics applications, we don't like having missed frames, so we don't just handle exceptions and move on.  Our lab doesn't have any ethernet cameras, so I don't have any experience with them but #10 has some information that might help.

As for the frame_id == 0 check.  We can implement a conditional check for non-ethernet cameras.
We should be able to use the enum called `DeviceType` and check if it is ETHERNET or GEV
```
enum DeviceTypeEnum
< Transport layer type of the device.

Enumerator	
DeviceType_Mixed	TL - Mixed
DeviceType_Custom 	TL - Custom
DeviceType_GEV 		TL - GEV
DeviceType_CL 		TL - CL
DeviceType_IIDC 		TL - IIDC
DeviceType_UVC 		TL - UVC
DeviceType_CXP 		TL - CXP
DeviceType_CLHS 		TL - CLHS
DeviceType_U3V 		TL - U3V
DeviceType_ETHERNET 	TL - ETHERNET
DeviceType_PCI 		TL - PCI	
```
It would be nice if someone with an ethernet camera can implement this and submit a PR.

wwstory, 2020-05-29:
@shahvi Hi,  Now I can run great, but i am not sure that it happen problem in the feature.

I try to compose this issue again.
I do this: 
1. install spinnaker sdk 2.0.
2. checkout spinnaker_sdk_camera_driver of dev branch.
3. set acquisition.launch binning = 2.
4. set net.core.rmem to 25mb.( bining = 2, if bining = 1, it maybe need set 100mb)
    sysctl -w net.core.rmem_max=26214400
    sysctl -w net.core.rmem_default=26214400

It's ok.

closed 2020-05-29 by wwstory