neufieldrobotics/spinnaker_sdk_camera_driver #88: No camera found while retrieving list of cameras

keerthana-kannan, 2020-06-19:
**System Description**
Camera Name and Model:
Operating System: Ubuntu 18.04
Spinnaker Version: Version 2.0 (tried versions 1.29 and 1.23 also)
ROS Version: Melodic

***Computer details***
Intel NUC NUC7i7DNB

Using single camera only. It's a USB 3.0 camera.

**Do the cameras work with SpinView?** YES 
When I follow the read me and execute the roslaunch, it says no cameras found while retrieving the list of cameras. 

**The error message is as follows:**

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://perception-NUC7i7DNKE:40367/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [19444631]
 * /acquisition_node/color: True
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1536
 * /acquisition_node/image_width: 2048
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 19444631
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [10007]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 216873d6-b288-11ea-af47-5c879cb840a7
process[rosout-1]: started with pid [10018]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [10025]
process[acquisition_node-3]: started with pid [10026]
[ INFO] [1592610834.347505333]: Initializing nodelet with 8 worker threads.
[ INFO] [1592610834.489925741]: Initializing nodelet
[ INFO] [1592610834.490876691]: [ OK ] USB memory: 1000 MB
[ INFO] [1592610834.490925100]: *** PARAMETER SETTINGS ***
[ INFO] [1592610834.490945474]: ** Date = 20200619
[ INFO] [1592610834.491441243]:   Save path set via parameter to: /home/perception
[ INFO] [1592610834.491483187]:   Camera IDs:
[ INFO] [1592610834.491828930]:     19444631
[ INFO] [1592610834.492170894]:   Camera Aliases:
[ INFO] [1592610834.492196892]:     19444631 >> cam0
[ INFO] [1592610834.492829254]:   Unique time stamps for each camera: false
[ INFO] [1592610834.493162301]:   color set to: true
[ INFO] [1592610834.493491119]:   Exporting images to ROS: true
[ INFO] [1592610834.493804454]:   Showing live images setting: false
[ INFO] [1592610834.494128064]:   Showing grid-style live images setting: false
[ INFO] [1592610834.494400664]:   Max Rate Save Mode: false
[ INFO] [1592610834.494669686]:   Displaying timing details: false
[ INFO] [1592610834.494929585]:   No. of images to skip set to: 20
[ INFO] [1592610834.495195745]:   Init sleep delays set to : 1.00 sec
[ WARN] [1592610834.495436787]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1592610834.495681570]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1592610834.495919506]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1592610834.496127323]:   Binning set to: 1
[ INFO] [1592610834.496345954]:   Using Software rate control, rate set to: 30
[ INFO] [1592610834.496555329]:   Saving images set to: 0
[ INFO] [1592610834.496797395]:   Camera Intrinsic Paramters:
[ INFO] [1592610834.496970523]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1592610834.497243447]:   Camera Distortion Paramters:
[ INFO] [1592610834.497284725]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1592610834.497557189]:   Camera Rectification Paramters:
[ INFO] [1592610834.497602440]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1592610834.497879625]:   Camera Projection Paramters:
[ INFO] [1592610834.497930061]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1592610834.497950080]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1592610834.497988918]: Creating system instance...
[ INFO] [1592610834.499554478]: Retreiving list of cameras...
[FATAL] [1592610835.582045522]: ASSERTION FAILED
	file = /home/perception/AS_Perception/spinnaker_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp
	line = 215
	cond = numCameras_
	message = 
[FATAL] [1592610835.582233508]: No cameras found!
[FATAL] [1592610835.582339655]: 

[FATAL] [1592610840.502818594]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 10025, exit code -5, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/perception/.ros/log/216873d6-b288-11ea-af47-5c879cb840a7/vision_nodelet_manager-2.log].
log file: /home/perception/.ros/log/216873d6-b288-11ea-af47-5c879cb840a7/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 10026, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/perception/.ros/log/216873d6-b288-11ea-af47-5c879cb840a7/acquisition_node-3.log].
log file: /home/perception/.ros/log/216873d6-b288-11ea-af47-5c879cb840a7/acquisition_node-3*.log

vik748, 2020-06-21:
Which camera model are you using?

keerthana-kannan, 2020-06-22:
@vik748 I'm using a FLIR Backfly S USB 3.0 Camera

shahvi, 2020-06-22:
If spinview can see the cameras I don't understand why the ROS driver would have an issue.  #48 also reported a similar issue.  I haven't tested on Melodic, so can't say what is going on.  

Similar to 48 can you try setting : `binning = 2` launch file to see if that helps.

All we do is `GetCameras()` to get list of attached cameras here: https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/2f9f14df0dcff7b4d5e1559abad8715f68d02170/src/capture.cpp#L211

This is very similar to the provided example at `/usr/src/spinnaker/src/Acquisition/Acquisition.cpp`
can you try running that with `Acquisition` and see if it detect the camera.

keerthana-kannan, 2020-06-23:
@vik748 binning = 2 gives a different error as mentioned in issue #48. The Acquisition.cpp detectd the camera and works fine and stores the 10 images. But the Acquisition.cpp runs as "sudo". I am not able to run roslaunch as sudo.

shahvi, 2020-06-23:
The camera in #48 was ethernet not USB.  Its interesting that you are having the same issue.

If your error is "First frame ID was not zero! Might cause sync issues later..."

try commenting out https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/2945642d00c16a0ff2174a2f840ed6432f9044d6/src/camera.cpp#L59

keerthana-kannan, 2020-06-23:
@shahvi After doing the above mentioned suggestion, I'm getting the following error. The output stops in *ACQUISITION* and doesn't come out of that. I cannot see any other output. 
[image]

vik748, 2020-06-23:
That is the normal behavior. It means it is working.

keerthana-kannan, 2020-06-23:
@vik748 But it's stuck there and I'm not seeing any image outputs as seen in actual acquisition.cpp example.

vik748, 2020-06-23:
It should be publishing to the appropriate  topic in ros. You can check by subscribing to it in rviz or rqtimageview.

keerthana-kannan, 2020-06-23:
@vik748 Sorry I didn't get you. What do you mean by that? After the Acquisition, I tried to do rqtimageview and there was no output. (Sorry I am new to using ROS).

shahvi, 2020-06-23:
Run `rostopic list` to see all message topics being publish
Run `rostopic echo <topic_name>` to see the messages on a topic
The images should be on something like `camera_array/cam0/xxx`

keerthana-kannan, 2020-06-23:
@shahvi 
[image]
No output with running rostopic list.

shahvi, 2020-06-23:
You need to run that on a separate terminal.  I'd suggest going through the ROS tutorials first.

keerthana-kannan, 2020-06-23:
@shahvi  Sure. Thank you so much for the help.

ZhouSongxin, 2020-06-25:
Camera Name and Model:
Operating System: Ubuntu 16.04
Spinnaker Version: Version 2.0
camera type: polarizationmono
ROS Version: Kinetic

roslaunch spinnaker_sdk_camera_driver acquisition_node.launch ... logging to /home/coostar/.ros/log/4a69339c-b6ac-11ea-bc5c-283a4d36ae91/roslaunch-coostar-5714.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://coostar:33323/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 2
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [19060278]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1536
 * /acquisition_node/image_width: 2048
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 19060278
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 20
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    acquisition_node (spinnaker_sdk_camera_driver/acquisition_node)

ROS_MASTER_URI=http://localhost:11311

process[acquisition_node-1]: started with pid [5736]
[ INFO] [1593068506.463634000]: [ OK ] USB memory: 1000 MB
[ INFO] [1593068506.463696770]: *** PARAMETER SETTINGS ***
[ INFO] [1593068506.463735883]: ** Date = 20200625
[ INFO] [1593068506.464419796]:   Save path set via parameter to: /home/coostar
[ INFO] [1593068506.464465352]:   Camera IDs:
[ INFO] [1593068506.464950227]:     19060278
[ INFO] [1593068506.465394835]:   Camera Aliases:
[ INFO] [1593068506.465410959]:     19060278 >> cam0
[ INFO] [1593068506.466225792]:   Unique time stamps for each camera: false
[ INFO] [1593068506.466605430]:   color set to: false
[ INFO] [1593068506.467045473]:   Exporting images to ROS: true
[ INFO] [1593068506.467535372]:   Showing live images setting: false
[ INFO] [1593068506.467966050]:   Showing grid-style live images setting: false
[ INFO] [1593068506.468340516]:   Max Rate Save Mode: false
[ INFO] [1593068506.468675322]:   Displaying timing details: false
[ INFO] [1593068506.468985598]:   No. of images to skip set to: 20
[ INFO] [1593068506.469375588]:   Init sleep delays set to : 1.00 sec
[ WARN] [1593068506.469767645]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1593068506.470117904]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1593068506.470457121]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1593068506.470786902]:   Binning set to: 2
[ INFO] [1593068506.471120730]:   Using Software rate control, rate set to: 20
[ INFO] [1593068506.471489898]:   Saving images set to: 0
[ INFO] [1593068506.471867903]:   Camera Intrinsic Paramters:
[ INFO] [1593068506.472021536]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1593068506.472436011]:   Camera Distortion Paramters:
[ INFO] [1593068506.472459806]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1593068506.472892459]:   Camera Rectification Paramters:
[ INFO] [1593068506.472944093]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1593068506.473398419]:   Camera Projection Paramters:
[ INFO] [1593068506.473426968]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1593068506.473462493]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1593068506.473510134]: Creating system instance...
[ INFO] [1593068506.475830768]: Retreiving list of cameras...
[ INFO] [1593068507.567492759]: Numer of cameras found: 1
[ INFO] [1593068507.567537031]:  Cameras connected: 1
[ INFO] [1593068507.567564419]:   -0122D636
[ WARN] [1593068507.567625342]:    Camera 19060278 not detected!!!
[FATAL] [1593068507.567648847]: ASSERTION FAILED
	file = /home/coostar/zsx_ros/src/spinnaker_sdk_camera_driver/src/capture.cpp
	line = 317
	cond = cams.size()
	message = 
[FATAL] [1593068507.567657675]: None of the connected cameras are in the config list!
[FATAL] [1593068507.567663171]: 

[acquisition_node-1] process has died [pid 5736, exit code -5, cmd /home/coostar/zsx_ros/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/coostar/.ros/log/4a69339c-b6ac-11ea-bc5c-283a4d36ae91/acquisition_node-1.log].
log file: /home/coostar/.ros/log/4a69339c-b6ac-11ea-bc5c-283a4d36ae91/acquisition_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I try  setting : binning = 2 launch file ,then got the results above.
Would you mind tell me what i can do, thanks.

ZhouSongxin, 2020-06-25:
@shahvi

shahvi, 2020-06-25:
@ZhouSongxin you error is different from the one in this issue.  See #76 , it should be fixed in the dev branch.  Can you try running that.

ZhouSongxin, 2020-06-26:
@shahvi 
I try the method of #76 ,like yesterday, set 0122D636 in both master_cam and cam_ids in yaml file.And also I set ROSCONSOLE_CONFIG_FILE to debug_console.conf.
I get the result as follow:
roslaunch spinnaker_sdk_camera_driver acquisition_node.launch ... logging to /home/coostar/.ros/log/fed23494-b74e-11ea-b573-283a4d36ae91/roslaunch-coostar-22401.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://coostar:38541/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: ['0122D636']
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1536
 * /acquisition_node/image_width: 2048
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 0122D636
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 20
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    acquisition_node (spinnaker_sdk_camera_driver/acquisition_node)

ROS_MASTER_URI=http://localhost:11311

process[acquisition_node-1]: started with pid [22418]
[ INFO] [1593137238.848531271]: [ OK ] USB memory: 1000 MB
[ INFO] [1593137238.848613292]: *** PARAMETER SETTINGS ***
[ INFO] [1593137238.848630755]: ** Date = 20200626
[ INFO] [1593137238.849254463]:   Save path set via parameter to: /home/coostar
[ INFO] [1593137238.849296032]:   Camera IDs:
[FATAL] [1593137238.849789527]: ASSERTION FAILED
	file = /home/coostar/zsx_ros/src/spinnaker_sdk_camera_driver/src/capture.cpp
	line = 358
	cond = nh_pvt_.getParam("cam_ids", cam_id_vec)
	message = 
[FATAL] [1593137238.849804572]: If cam_aliases are provided, they should be the same number as cam_ids and should correspond in order!
[FATAL] [1593137238.849824512]: 

[acquisition_node-1] process has died [pid 22418, exit code -5, cmd /home/coostar/zsx_ros/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/coostar/.ros/log/fed23494-b74e-11ea-b573-283a4d36ae91/acquisition_node-1.log].
log file: /home/coostar/.ros/log/fed23494-b74e-11ea-b573-283a4d36ae91/acquisition_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

I thought the question is in the C++ statement"ROS_ASSERT_MSG(nh_pvt_.getParam("cam_ids", cam_id_vec),"If cam_aliases are provided, they should be the same number as cam_ids and should correspond in order!");",
then I go to #84 , i try the operation like
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/pull/84/files/b323f2cf9fd1d2fffd7926638b82128b71612dbb

shahvi, 2020-06-26:
Does the camera work smoothly in SpinView? Also try setting binning to 2 to reduce bandwidth.

ZhouSongxin, 2020-06-26:
@shahvi 
It cannt work so smoothly with stream being aborted.
I set binning to 2.Unfortunately, the connection isnt stable.
[FATAL] [1593143068.343876371]: Excption: Spinnaker: Error writing to device at register address: 000C8824. Please try reconnecting the device. [-1010]

Thank you for your help very much.
best wishes for you!

keerthana-kannan, 2020-06-26:
@shahvi As mentioned above the camera worked fine with roslaunch instruction from the README (I was able to see the stream using rosrun rqt_image_view rqt_image_view). But when I do rosrun for the acquisition node, I'm getting the following error. (I'm running roscore in a parallel terminal)
[image]

shahvi, 2020-06-26:
this node needs a bunch of parameters to work correctly, that is why it is better to use the roslaunch file where you setup all the required parameters.  rosrun would require to independently setup all the parameters on the parameter server.  Is there a reason you don't want to use roslaunch ?

keerthana-kannan, 2020-06-26:
@shahvi Thank you for the clarification. I will use roslaunch. I was just confused as to why rosrun gave me an error while roslaunch worked fine.

closed 2020-06-29 by shahvi