neufieldrobotics/spinnaker_sdk_camera_driver #89: Building on Jetson AGX Xavier running Jetpack 4.3 (Ubuntu 18.04)

rishabhkjain, 2020-06-24:
I am trying setup this package on an Xavier which runs 18.04 (thus ROS melodic). As OpenCV is already installed with Jetpack (OpenCV 4 on Jetpack 4.3), and due to other differences in architecture (arm64) there were some issues with building this driver. Below are the issues I ran into when building along with the ways I was able to resolve them to get the package running. Hopefully this helps other people trying to use a similar compute. 

Note, this was using the dev branch of the repo as I was having the serial hex issue with the master branch. 

```
<code>
Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  spinnaker_sdk_camera_driver/CMakeLists.txt:18 (find_package)

-------------------------------------------------------------------------------

Solved by including this package in src for the workspace  

https://github.com/ros-perception/vision_opencv

===============================================================================
vision_opencv/cv_bridge/CMakeLists.txt:11 (find_package)
CMake Error at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1947 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.65.1

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_python37

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
--------------------------------------------------------------------------------------
Solved by changing 
    find_package(Boost REQUIRED python3) to     find_package(Boost REQUIRED python3)
 at /src/vision_opencv/cv_bridge/CMakeLists.txt:11

=========================================================================================

In file included from /home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/include/spinnaker_sdk_camera_driver/capture.h:4:0,
                 from /home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:1:
/home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/include/spinnaker_sdk_camera_driver/std_include.h:9:10: fatal error: cv.h: No such file or directory
 #include <cv.h

-------------------------------------------------------------------------------------------

Solved by changing 

#include <cv.h>
to #include <opencv2/opencv.hpp>

in /src/spinnaker_sdk_camera_driver/include/spinnaker_sdk_camera_driver/std_inc

==============================================================================================

/home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:927:42: error: ‘CV_WINDOW_NORMAL’ was not declared in this scope
     if (LIVE_)namedWindow("Acquisition", CV_WINDOW_NORMAL | CV_WINDOW_KEEPRATIO);
                                          ^~~~~~~~~~~~~~~~
/home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:927:42: note: suggested alternative: ‘CV_RAND_NORMAL’
     if (LIVE_)namedWindow("Acquisition", CV_WINDOW_NORMAL | CV_WINDOW_KEEPRATIO);
                                          ^~~~~~~~~~~~~~~~
                                          CV_RAND_NORMAL
/home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:927:61: error: ‘CV_WINDOW_KEEPRATIO’ was not declared in this scope
     if (LIVE_)namedWindow("Acquisition", CV_WINDOW_NORMAL | CV_WINDOW_KEEPRATIO);
                                                             ^~~~~~~~~~~~~~~~~~~
/home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:927:61: note: suggested alternative: ‘CV_MINOR_VERSION’
     if (LIVE_)namedWindow("Acquisition", CV_WINDOW_NORMAL | CV_WINDOW_KEEPRATIO);
                                                             ^~~~~~~~~~~~~~~~~~~
                                                             CV_MINOR_VERSION
/home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:976:23: error: ‘cvWaitKey’ was not declared in this scope
             int key = cvWaitKey(1);
                       ^~~~~~~~~
/home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:1003:21: error: ‘cvDestroyAllWindows’ was not declared in this scope
                     cvDestroyAllWindows();
                     ^~~~~~~~~~~~~~~~~~~
/home/avsensors/flir_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp:1032:21: error: ‘cvDestroyAllWindows’ was not declared in this scope
                     cvDestroyAllWindows();
------------------------------------------------------------------------------------------------------------------------------

solved by changing the usage to new CV definitions 

CV_WINDOW_NORMAL -> WINDOW_NORMAL 
CV_WINDOW_KEEPRATIO -> WINDOW_KEEPRATIO
cvWaitKey -> waitKey 
cvDestroyAllWindows -> destroyAllWindows 

in /src/spinnaker_sdk_camera_driver/src/capture.cpp

=======================================================================================================================================

/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: error: return-statement with no value, in function returning ‘void*’ [-fpermissive]
 #define import_array() {if (_import_array() < 0) {PyErr_Print(); PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }
                                                                                                                                                ^
/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: note: in definition of macro ‘import_array’
 #define import_array() {if (_import_array() < 0) {PyErr_Print(); PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }

----------------------------------------------------------------------------------------------------------------------------------
 solved by 
 changing #define NUMPY_IMPORT_ARRAY_RETVAL to 
 #define NUMPY_IMPORT_ARRAY_RETVAL NULL 

 in /usr/include/python2.7/numpy/__multiarray_api.h:1534

 ================================================================================================================================
</code>
```

shahvi, 2020-06-24:
@rishabhkjain This is helpful.  Thanks.

capitalPSA, 2020-07-21:
Works for me too. Thanks.

Estella12, 2020-12-07:
I met other failed issues while catkin_make (Ubuntu18.04 on Xavier):
"/usr/bin/ld: warning: libboost_system.so.1.65.1, needed by /opt/ros/melodic/lib/libroscpp.so, may conflict with libboost_system.so.1.57.0
/home/**/catkin_ws/devel/lib/libacquilib.so: undefined reference to `boost::this_thread::hiden::sleep_until(timespec const&)'
collect2: error: ld returned 1 exit status
"
There must be something wrong while linking. But I don't know how to fix.
There are two boost editions on my board, the 1.65 was installed with jetson pack ,and another 1.57 was installed with ros. And I have tried to specify paths of boost lib/include in CmakeLists.txt and *.h files, nothing works. 
Anybody knows why?
Thanks a lot.

Estella12, 2020-12-09:
Every step is working following the modifications, thx.

davidy, 2021-01-20:
Thanks for the contribution!

civerachb-cpr, 2021-06-03:
What version of libspinnaker did you have installed when you did this?  I'm currently trying to get the driver working on a AGX Xavier, but when I try launching either the acquisition or node_acquisition files I get this output:

```
[ INFO] [1622732321.532633307]: Initializing nodelet with 8 worker threads.
[ INFO] [1622732321.599024141]: Initializing nodelet
[ INFO] [1622732321.600669336]: [ OK ] USB memory: 1024 MB
[ INFO] [1622732321.600799453]: *** PARAMETER SETTINGS ***
[ INFO] [1622732321.600901602]: ** Date = 20210603
[ INFO] [1622732321.602936350]:   Save path set via parameter to: /home/administrator
[ INFO] [1622732321.603028738]:   Camera IDs:
[ INFO] [1622732321.604928312]:     17197559
[ INFO] [1622732321.606587010]:   Camera Aliases:
[ INFO] [1622732321.606656294]:     17197559 >> cam0
[ INFO] [1622732321.608168906]:   External trigger: false
[ INFO] [1622732321.611187442]:   Unique time stamps for each camera: false
[ INFO] [1622732321.612480460]:   color set to: false
[ INFO] [1622732321.613894604]:   17197559 flip_horizontal 0
[ INFO] [1622732321.615750592]:   17197559 flip_vertical 0
[ INFO] [1622732321.617835134]:   Exporting images to ROS: true
[ INFO] [1622732321.620221705]:   Showing live images setting: false
[ INFO] [1622732321.622145792]:   Showing grid-style live images setting: false
[ INFO] [1622732321.623622595]:   Max Rate Save Mode: false
[ INFO] [1622732321.626072369]:   Displaying timing details: false
[ INFO] [1622732321.629927519]:   No. of images to skip set to: 20
[ INFO] [1622732321.632968072]:   Init sleep delays set to : 1.00 sec
[ WARN] [1622732321.635225582]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1622732321.638004523]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1622732321.641253182]:   'gain' Parameter was zero or negative, using Auto gain based on target grey value
[ INFO] [1622732321.642783939]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1622732321.644422445]:   Binning set to: 1
[ INFO] [1622732321.646270048]:   Using Software rate control, rate set to: 20
[ INFO] [1622732321.648646283]:   Saving images set to: 0
[ INFO] [1622732321.653954619]:   tf_prefix set to: 
[ INFO] [1622732321.669014658]:   Region of Interest set to width: 0	height: 0	offset_x: 0 offset_y: 0
[ INFO] [1622732321.672111886]:   Camera Intrinsic Paramters:
[ INFO] [1622732321.673060505]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1622732321.676944456]:   Camera Distortion Paramters:
[ INFO] [1622732321.677288887]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1622732321.682414111]:   Camera Rectification Paramters:
[ INFO] [1622732321.682986456]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1622732321.687644522]:   Camera Projection Paramters:
[ INFO] [1622732321.688143713]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1622732321.688278375]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1622732321.688643928]: *** SYSTEM INFORMATION ***
[ INFO] [1622732321.688826400]: Creating system instance...
[ INFO] [1622732321.688994695]: spinnaker_sdk_camera_driver package version: 1.1.0
[ INFO] [1622732323.052746641]: Spinnaker library version: 2.2.0.48
[ INFO] [1622732323.053113377]: Retreiving list of cameras...
[FATAL] [1622732323.059755373]: ASSERTION FAILED
	file = /home/administrator/catkin_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp
	line = 185
	cond = numCameras_
	message = 
[FATAL] [1622732323.060003032]: No cameras found!
[FATAL] [1622732323.060117693]: 

[acquisition_node-1] process has died [pid 670, exit code -6, cmd /home/administrator/catkin_ws/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/administrator/.ros/log/6e7f22c6-c3e6-11eb-8b42-00044bcbdf38/acquisition_node-1.log].
log file: /home/administrator/.ros/log/6e7f22c6-c3e6-11eb-8b42-00044bcbdf38/acquisition_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```