neufieldrobotics/spinnaker_sdk_camera_driver #94: how to publish raw Img,not Compressed image

ZhouSongxin, 2020-06-29:
I found  the size of rostopic of image published by the project, a raw bmp image format is about 5M, but the img published is only 1M. In my study, I need raw Img published by ros. what should I do , if I want to pulish raw img(5M).thank you!

shahvi, 2020-06-29:
Can you check the resolution of the image being published by running `rostopic echo /camera_array/cam0/image_raw --noarr
` for the appropriate topic name.
Also make sure you have set `binning=1` to make sure you get the high resolution from the camera.

ZhouSongxin, 2020-06-29:
> Can you check the resolution of the image being published by running `rostopic echo /camera_array/cam0/image_raw --noarr ` for the appropriate topic name.
> Also make sure you have set `binning=1` to make sure you get the high resolution from the camera.

I got the response after entering the statement:
---
header: 
  seq: 436
  stamp: 
    secs: 1593438160
    nsecs: 400452116
  frame_id: "cam_0_optical_frame"
height: 1024
width: 1224
encoding: "mono8"
is_bigendian: 0
step: 1224
data: "<array type: uint8, length: 1253376>"

there is only an angle picture from the polarization mono.but in fact, the result should have a picture mixed with four angle polarized_picture ,the resolution should 2048*2448

shahvi, 2020-06-29:
I am not familiar with this camera and don't know what you mean by angle picture.

Sounds like your binning is set to 2, which is reducing your resolution.  if you set it to 1 it will increase the image size by factor of 4.

ZhouSongxin, 2020-06-29:
my test_params.yaml file as follow:
cam_ids:
- 19060278
cam_aliases:
- cam0
master_cam: 19060278
skip: 20
delay: 1.0

# Assign all the follwing via launch file to prevent confusion and conflict

save_path: /home/coostar/Desktop/polarizatonImg
save_type: .bmp #binary or .tiff or .bmp
binning: 1 # going from 2 to 1 requires cameras to be unplugged and replugged
#color: false
#frames: 50
#soft_framerate: 20 # this frame rate reflects to the software frame rate set using ros::rate
#exp: 997
#to_ros: true  #When to_ros is not selected, but live is selected, pressing 'space' exports single image to ROS

#Camera info message details
distortion_model: plumb_bob
image_height: 2048
image_width: 2448
distortion_coeffs:
- [-0.021141875266089592, -0.3733872931278025, 2.385982550579459, 3.2824571732099725]

#specified as [fx  0 cx 0 fy cy 0  0  1]
intrinsic_coeffs:
- [1886.9232141485886, 0.0, 604.7214878709341, 0.0, 1886.6668765711668, 493.47726714719823, 0.0, 0.0, 1.0]

rectification_coeffs:
- [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]

projection_coeffs:
- [913.700317, 0.000000, 953.448302, 0.000000, 0.000000, 1063.296631, 777.871993, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]

And the acquisition_node.launch:
 <arg name="binning"		    default="1"	doc="Binning for cameras, when changing from 2 to 1 cameras need to be unplugged and replugged"/>

shahvi, 2020-06-29:
What are the settings in your launch file.  The binning will be overwritten by the launch file as per the note "Assign all the following via launch file to prevent confusion and conflict"

ZhouSongxin, 2020-06-29:
@shahvi 
I use the defult setting:
<launch>
  
  <env name="ROSCONSOLE_CONFIG_FILE" value="$(find spinnaker_sdk_camera_driver)/cfg/std_console.conf"/>

  

  <arg name="binning"		    default="1"	doc="Binning for cameras, when changing from 2 to 1 cameras need to be unplugged and replugged"/>
  <arg name="color"		        default="false"	doc="Should color images be used (only works on models that support color images)"/>
  <arg name="exposure_time"     default="0" doc="Exposure_time setting for cameras"/>
  <arg name="target_grey_value"   default="0"   doc="AutoExposureTargetGreyValue min: 4 max: 99 if set below 4, AutoExposureTargetGreyValueAuto will be set to continuous(auto) also available as dynamic reconfigurable parameter" />
  <arg name="frames"            default="3400"	doc="Numer of frames to save/view 0=ON"/>
  <arg name="live"              default="false"	doc="Show images on screen GUI"/>
  <arg name="live_grid"         default="false"	doc="Show images on screen GUI in a grid"/>
  <arg name="output"            default="screen"	doc="display output to screen or log file"/>
  <arg name="save"              default="false"	doc="flag whether images should be saved or not"/>
  <arg name="save_path"         default="~"	doc="location to save the image data"/>
  <arg name="save_type"         default="bmp"	doc="Type of file type to save to when saving images locally: binary, tiff, bmp, jpeg etc." />
  <arg name="soft_framerate"    default="20"	doc="When hybrid software triggering is used, this controls the FPS, 0=as fast as possible"/>
  <arg name="time"              default="false"	doc="Show time/FPS on output"/>
  <arg name="to_ros"            default="true"	doc="Flag whether images should be published to ROS" />
  <arg name="utstamps"          default="false"	doc="Flag whether each image should have Unique timestamps vs the master cams time stamp for all" />
  <arg name="max_rate_save"     default="false"	doc="Flag for max rate mode which is when the master triggerst the slaves and saves images at maximum rate possible" />
  <arg name="config_file"       default="$(find spinnaker_sdk_camera_driver)/params/test_params.yaml" doc="File specifying the parameters of the camera_array"/>

  
  <node pkg="spinnaker_sdk_camera_driver" type="acquisition_node" name="acquisition_node" output="$(arg output)" args="" respawn="false" >

    
    <rosparam command="load"        file="$(arg config_file)" />

    
    <param name="exposure_time"     value="$(arg exposure_time)" />
    <param name="target_grey_value" value="$(arg target_grey_value)" />
    <param name="binning"           value="$(arg binning)" />
    <param name="color"             value="$(arg color)" />
    <param name="frames"            value="$(arg frames)" />
    <param name="live"              value="$(arg live)" />
    <param name="live_grid"         value="$(arg live_grid)" />
    <param name="save"              value="$(arg save)" />
    <param name="save_type"         value="$(arg save_type)" />
    <param name="save_path"         value="$(arg save_path)" />
    <param name="soft_framerate"    value="$(arg soft_framerate)" />
    <param name="time"              value="$(arg time)" />
    <param name="utstamps"          value="$(arg utstamps)" />
    <param name="to_ros"            value="$(arg to_ros)"/>
    <param name="max_rate_save"     value="$(arg max_rate_save)"/>

  </node>    
 
</launch>

ZhouSongxin, 2020-06-29:
[image]

ZhouSongxin, 2020-06-29:
my sensor's resolution should 2048*2448, but img published by ROS is 1024*1224

shahvi, 2020-06-29:
When you run the launch file, all the settings in the node are printed out.  Can you find the line with something like:
```
[ INFO] [1593139408.761924476]: Binning set to: 2
```

ZhouSongxin, 2020-06-29:
print [ INFO] [1593441202.661072730]:   Binning set to: 1 in my computer
roslaunch spinnaker_sdk_camera_driver acquisition_node.launch 
... logging to /home/coostar/.ros/log/54bf144e-ba15-11ea-a04d-283a4d36ae91/roslaunch-coostar-3031.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://coostar:46733/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [19060278]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 2048
 * /acquisition_node/image_width: 2448
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 19060278
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 20
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    acquisition_node (spinnaker_sdk_camera_driver/acquisition_node)

ROS_MASTER_URI=http://localhost:11311

process[acquisition_node-1]: started with pid [3054]
[ INFO] [1593441202.654365873]: [ OK ] USB memory: 1000 MB
[ INFO] [1593441202.654422152]: *** PARAMETER SETTINGS ***
[ INFO] [1593441202.654437815]: ** Date = 20200629
[ INFO] [1593441202.654912174]:   Save path set via parameter to: /home/coostar
[ INFO] [1593441202.654937384]:   Camera IDs:
[ INFO] [1593441202.655377400]:     19060278
[ INFO] [1593441202.655755872]:   Camera Aliases:
[ INFO] [1593441202.655768837]:     19060278 >> cam0
[ INFO] [1593441202.656464901]:   Unique time stamps for each camera: false
[ INFO] [1593441202.656777728]:   color set to: false
[ INFO] [1593441202.657222449]:   Exporting images to ROS: true
[ INFO] [1593441202.657701021]:   Showing live images setting: false
[ INFO] [1593441202.658161059]:   Showing grid-style live images setting: false
[ INFO] [1593441202.658578244]:   Max Rate Save Mode: false
[ INFO] [1593441202.658953972]:   Displaying timing details: false
[ INFO] [1593441202.659316011]:   No. of images to skip set to: 20
[ INFO] [1593441202.659673293]:   Init sleep delays set to : 1.00 sec
[ WARN] [1593441202.660035495]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1593441202.660399829]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1593441202.660738353]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1593441202.661072730]:   Binning set to: 1
[ INFO] [1593441202.661397138]:   Using Software rate control, rate set to: 20
[ INFO] [1593441202.661810970]:   Saving images set to: 0
[ INFO] [1593441202.662219593]:   Camera Intrinsic Paramters:
[ INFO] [1593441202.662302446]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1593441202.662703625]:   Camera Distortion Paramters:
[ INFO] [1593441202.662729865]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1593441202.663140210]:   Camera Rectification Paramters:
[ INFO] [1593441202.663169055]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1593441202.663561694]:   Camera Projection Paramters:
[ INFO] [1593441202.663591233]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1593441202.663600989]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1593441202.663616269]: Creating system instance...
[ INFO] [1593441202.665719969]: Retreiving list of cameras...
[ INFO] [1593441203.888901419]: Numer of cameras found: 1
[ INFO] [1593441203.888993232]:  Cameras connected: 1
[ INFO] [1593441203.889234773]:   -19060278
[ INFO] [1593441203.998725955]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1593441204.000991652]: *** FLUSH SEQUENCE ***
[ INFO] [1593441204.001039305]: Initializing cameras...
[ INFO] [1593441204.297724706]: Deinitializing cameras...
[ INFO] [1593441204.302247689]: All cameras deinitialized.
[ INFO] [1593441206.302449751]: Initializing cameras...
[ INFO] [1593441206.640303284]: *** ACQUISITION ***

shahvi, 2020-06-29:
Unfortunately without having a camera to play with, I can't be of much help. 

One thing I can suggest is to look at all the settings that work for you in SpinView and replicate it on the ROS driver.  All the settings are set on the camera at:
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/2f9f14df0dcff7b4d5e1559abad8715f68d02170/src/capture.cpp#L605

ZhouSongxin, 2020-06-29:
thank you!
@shahvi 
I find the resolution is set there!
in capture.cpp 264 rows:
// full resolution image_size
ci_msg->height = 2048; //image_height;
ci_msg->width =  2448; //image_width;

closed 2020-06-29 by ZhouSongxin