neufieldrobotics/spinnaker_sdk_camera_driver #95: unable to relaunch the node after shutting it down

rishabhkjain, 2020-06-30:
I am currently trying to configure the parameters for my camera and usually I need to shut down the current node, change something in the config file and then relaunch the node.  However, whenever I shut down the node (with ctrl+c) and try to relaunch the node, I always get an error stating that I may have the wrong binning number set. If I unplug and replug the camera from my computer and launch the node, it works fine. Is there a "cleaner" way to shut down the node so that I don't have to physically unplug and replug the camera?

shahvi, 2020-06-30:
I think for the binning, there is a bug in the firmware, where changing binning from 1 to 2 is fine, but when going 2 to 1 only shows a part of the image. The only way to fix this is to unplug and replug.  We reported that to FLIR as well.  You can try this even with SpinView and you should observe the issue.
What other settings are you having issues with?  Does it happen in SpinView as well?

rishabhkjain, 2020-06-30:
The problem I am having is not just for when I change the settings. For example, I will launch the node and it will work fine, and then I will shut it down using Ctrl + C in the terminal window where I had launched the node. However, if I try to launch the node again, it gives me an error. 

Here is an excerpt of the error that I get when I relaunch the node. I have  Blackfly USB camera with a color sensor. 
The entire log is available here [log.txt](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/files/4852596/log.txt)

```
[ INFO] [1593527743.988362537]: *** FLUSH SEQUENCE ***
[ INFO] [1593527743.988549487]: Initializing cameras...
[ INFO] [1593527744.831369380]: Deinitializing cameras...
[ INFO] [1593527744.871539190]: All cameras deinitialized.
[ INFO] [1593527746.872093483]: Initializing cameras...
[FATAL] [1593527747.563748855]: Error: Spinnaker: GenICam::OutOfRangeException= Value = 1 must be equal or greater than Min = 2. : OutOfRangeException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 79) [-2002]
[ WARN] [1593527747.564132962]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1593527747.772046425]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Camera is not streaming [-1010]
[acquisition_node-3] process has died [pid 9980, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/home/avsensors/.ros/log/f6a0b5e6-bade-11ea-aa4b-9abe2142ed9d/acquisition_node-3.log].
log file: /home/avsensors/.ros/log/f6a0b5e6-bade-11ea-aa4b-9abe2142ed9d/acquisition_node-3*.log
^C[vision_nodelet_manager-2] killing on exit
[vision_nodelet_manager-2] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

shahvi, 2020-06-30:
I am not able to reproduce the issue on the dev brach on Ubuntu 16.04, amd 64 system with Spinnaker 2.0.0.147.  Not sure what is going on.
Can you run spinview and take a look at which values you are allowed to set for binning on your camera. Also try the sequence in spinviewe to see if you can reproduce:
1. Set binning to 1
2. Start acquisition
3. Stop acquisition
4. Try setting the binning to 1 or 2

Separate from all this, another thing you could do, is put that line in a try catch block for this particular error.

Xulei-Zhai, 2021-06-13:
I have met the same problem.

`[ INFO] [1623570695.980551016]: *** FLUSH SEQUENCE ***
[ INFO] [1623570695.980569081]: Initializing cameras...
[ INFO] [1623570696.556836531]: Deinitializing cameras...
[ INFO] [1623570696.766786521]: All cameras deinitialized.
[ INFO] [1623570698.767073044]: Initializing cameras...
[FATAL] [1623570699.133026746]: Error: Spinnaker: GenICam::OutOfRangeException= Value = 1 must be equal or greater than Min = 2. : OutOfRangeException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 79) [-2002]
[ WARN] [1623570699.133060426]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1623570699.281441032]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
[acquisition_node-3] process has died [pid 8171, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/home/xulei/.ros/log/2c890876-cc1c-11eb-be58-378c21975153/acquisition_node-3.log].
log file: /home/xulei/.ros/log/2c890876-cc1c-11eb-be58-378c21975153/acquisition_node-3*.log
[vision_nodelet_manager-2] process has died [pid 8170, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/xulei/.ros/log/2c890876-cc1c-11eb-be58-378c21975153/vision_nodelet_manager-2.log].
log file: /home/xulei/.ros/log/2c890876-cc1c-11eb-be58-378c21975153/vision_nodelet_manager-2*.log
`

Xulei-Zhai, 2021-06-13:
I hane met the same problem .

` roslaunch spinnaker_sdk_camera_driver acquisition.launch
... logging to /home/xulei/.ros/log/2c890876-cc1c-11eb-be58-378c21975153/roslaunch-xulei-BATTLE-AX-B365M-HD-PRO-8131.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://xulei-BATTLE-AX-B365M-HD-PRO:39781/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [19085011]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/external_trigger: False
 * /acquisition_node/flip_horizontal: [False]
 * /acquisition_node/flip_vertical: [False]
 * /acquisition_node/frames: 3400
 * /acquisition_node/gain: 0
 * /acquisition_node/image_height: 1080
 * /acquisition_node/image_width: 1440
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 19085011
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/region_of_interest/height: 0
 * /acquisition_node/region_of_interest/width: 0
 * /acquisition_node/region_of_interest/x_offset: 0
 * /acquisition_node/region_of_interest/y_offset: 0
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/tf_prefix: 
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: noetic
 * /rosversion: 1.15.11

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [8146]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2c890876-cc1c-11eb-be58-378c21975153
process[rosout-1]: started with pid [8163]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [8170]
process[acquisition_node-3]: started with pid [8171]
[ INFO] [1623570694.961430684]: Initializing nodelet with 8 worker threads.
[ INFO] [1623570695.077213610]: Initializing nodelet
[ INFO] [1623570695.077938043]: [ OK ] USB memory: 1000 MB
[... 33 lines cut ...]
[ INFO] [1623570695.085800703]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1623570695.086108196]:   Camera Projection Paramters:
[ INFO] [1623570695.086158334]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1623570695.086169044]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1623570695.086189269]: *** SYSTEM INFORMATION ***
[ INFO] [1623570695.086198281]: Creating system instance...
[ INFO] [1623570695.086206270]: spinnaker_sdk_camera_driver package version: 1.1.0
[ INFO] [1623570695.919716940]: Spinnaker library version: 2.4.0.143
[ INFO] [1623570695.919825153]: Retreiving list of cameras...
[ INFO] [1623570695.925620902]: Numer of cameras found: 1
[ INFO] [1623570695.925729798]:  Cameras connected: 1
[ INFO] [1623570695.927520700]:   -19085011 Blackfly S BFS-PGE-16S2C 1705.0.63.0
[ INFO] [1623570695.979861596]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1623570695.980551016]: *** FLUSH SEQUENCE ***
[ INFO] [1623570695.980569081]: Initializing cameras...
[ INFO] [1623570696.556836531]: Deinitializing cameras...
[ INFO] [1623570696.766786521]: All cameras deinitialized.
[ INFO] [1623570698.767073044]: Initializing cameras...
[FATAL] [1623570699.133026746]: Error: Spinnaker: GenICam::OutOfRangeException= Value = 1 must be equal or greater than Min = 2. : OutOfRangeException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 79) [-2002]
[ WARN] [1623570699.133060426]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1623570699.281441032]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
[acquisition_node-3] process has died [pid 8171, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/home/xulei/.ros/log/2c890876-cc1c-11eb-be58-378c21975153/acquisition_node-3.log].
log file: /home/xulei/.ros/log/2c890876-cc1c-11eb-be58-378c21975153/acquisition_node-3*.log
[vision_nodelet_manager-2] process has died [pid 8170, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/xulei/.ros/log/2c890876-cc1c-11eb-be58-378c21975153/vision_nodelet_manager-2.log].
log file: /home/xulei/.ros/log/2c890876-cc1c-11eb-be58-378c21975153/vision_nodelet_manager-2*.log`

shahvi, 2021-06-13:
We have known issues with this version of Spinnaker, can you try one of the tested versions shown in the ReadME.