neufieldrobotics/spinnaker_sdk_camera_driver #98: Numer of cameras found: 1 but it can not detected.

708949503, 2020-07-02:
hello all
I run the program. But the problem occured:
[ INFO] [1593688277.763276319]: Creating system instance...
[ INFO] [1593688279.236575129]: Retreiving list of cameras...
[ INFO] [1593688280.675752245]: Numer of cameras found: 1
[ INFO] [1593688280.675863346]:  Cameras connected: 1
[ INFO] [1593688280.676020777]:   -01140DD4
[ WARN] [1593688280.676108796]:    Camera 18091476 not detected!!!
[FATAL] [1593688280.676174797]: ASSERTION FAILED
	file = /home/zwz/spinnaker_ws/src/spinnaker_camera_driver/src/capture.cpp
	line = 317
	cond = cams.size()
	message = 
[FATAL] [1593688280.676239237]: None of the connected cameras are in the config list!
[FATAL] [1593688280.676281138]: 

[FATAL] [1593688280.793412629]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 3623, exit code -5, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/zwz/.ros/log/bfb9caaa-bc54-11ea-a33f-b46921808f0e/vision_nodelet_manager-2.log].
log file: /home/zwz/.ros/log/bfb9caaa-bc54-11ea-a33f-b46921808f0e/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 3624, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/zwz/.ros/log/bfb9caaa-bc54-11ea-a33f-b46921808f0e/acquisition_node-3.log].
log file: /home/zwz/.ros/log/bfb9caaa-bc54-11ea-a33f-b46921808f0e/acquisition_node-3*.log

shahvi, 2020-07-02:
This is because of change in API from Spinnaker. Can you try the dev branch.

Sent from my Verizon, Samsung Galaxy smartphone

708949503, 2020-07-02:
> This is because of change in API from Spinnaker. Can you try the dev branch. Sent from my Verizon, Samsung Galaxy smartphone Get Outlook for Android<https://aka.ms/ghei36>
> […](#)

thank you 
I try another driver and i can pass it

GCChen97, 2020-10-25:
Because I only use one camera, I simply modify two lines in `acquisition::Capture::load_cameras()`:
```
            // if (cam.get_id().compare(cam_ids_[j]) == 0) {
            if (true) {
                current_cam_found=true;
                // if (cam.get_id().compare(master_cam_id_) == 0) {
                if (true) {
```

zlacelle, 2020-10-26:
I’m on the dev branch and am getting the same erro. It detects the camera and firmware, but then immediately throws a “Camera 0 not detected!!!” And quits.

This is with Spinnaker SDK 2.0.0.147 on Ubuntu 16.04, ROS Kinetic.

shahvi, 2020-10-26:
@zlacelle please post the full output from the node.

yanghan1ros, 2021-01-14:
[ INFO] [1610616468.237166437]: Initializing nodelet with 4 worker threads.
[ INFO] [1610616468.378173520]: Initializing nodelet
[ INFO] [1610616468.379202177]: [ OK ] USB memory: 1000 MB
[ INFO] [1610616468.379257104]: *** PARAMETER SETTINGS ***
[ INFO] [1610616468.379273247]: ** Date = 20210114
[ INFO] [1610616468.379832104]:   Save path set via parameter to: /home/a-y-x
[ INFO] [1610616468.379872753]:   Camera IDs:
[ INFO] [1610616468.380366480]:     20505630
[ INFO] [1610616468.380767377]:   Camera Aliases:
[ INFO] [1610616468.380797171]:     20505630 >> cam0
[ INFO] [1610616468.381608867]:   Unique time stamps for each camera: false
[ INFO] [1610616468.382052579]:   color set to: false
[ INFO] [1610616468.382496243]:   Exporting images to ROS: true
[ INFO] [1610616468.382825215]:   Showing live images setting: false
[ INFO] [1610616468.383148263]:   Showing grid-style live images setting: false
[ INFO] [1610616468.383479688]:   Max Rate Save Mode: false
[ INFO] [1610616468.383798346]:   Displaying timing details: false
[ INFO] [1610616468.384124399]:   No. of images to skip set to: 20
[ INFO] [1610616468.384472130]:   Init sleep delays set to : 1.00 sec
[ WARN] [1610616468.384900795]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1610616468.385412580]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1610616468.385812666]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1610616468.386161736]:   Binning set to: 1
[ INFO] [1610616468.386616163]:   Using Software rate control, rate set to: 30
[ INFO] [1610616468.387076840]:   Saving images set to: 0
[ INFO] [1610616468.387533310]:   Camera Intrinsic Paramters:
[ INFO] [1610616468.387785567]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1610616468.388188459]:   Camera Distortion Paramters:
[ INFO] [1610616468.388221906]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1610616468.390070162]:   Camera Rectification Paramters:
[ INFO] [1610616468.390124971]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1610616468.390540964]:   Camera Projection Paramters:
[ INFO] [1610616468.390597369]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1610616468.390619469]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1610616468.390666228]: Creating system instance...
[ INFO] [1610616468.394218049]: Retreiving list of cameras...
[FATAL] [1610616469.471596115]: ASSERTION FAILED
	file = /home/a-y-x/spinnaker_ws/src/spinnaker_camera_driver/src/capture.cpp
	line = 214
	cond = numCameras_
	message = 
[FATAL] [1610616469.471661554]: No cameras found!
[FATAL] [1610616469.471686322]: 

[FATAL] [1610616469.650931481]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 29030, exit code -5, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/a-y-x/.ros/log/c37cb162-564a-11eb-9215-54ab3a66b045/vision_nodelet_manager-2.log].
log file: /home/a-y-x/.ros/log/c37cb162-564a-11eb-9215-54ab3a66b045/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 29031, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=/home/a-y-x/.ros/log/c37cb162-564a-11eb-9215-54ab3a66b045/acquisition_node-3.log].
log file: /home/a-y-x/.ros/log/c37cb162-564a-11eb-9215-54ab3a66b045/acquisition_node-3*.log

closed 2021-01-14 by shahvi