neufieldrobotics/spinnaker_sdk_camera_driver #100: Unable to complete catkin_make, able to launch and connect camera to spinview

jbwurts, 2020-10-23:
**System Description**
Camera Name and Model: Blackfly S FFY-U3-16S2C-S
Operating System: Ubuntu 18.04
Spinnaker Version: 2.2.0
ROS Version: Melodic

For USB 3.0 cameras:
On same USB 3.0 bus or different buses? USB 3.0
Is the USB 3.0 hub externally powered? No

**Do the cameras work with SpinView?**
Yes

**Describe the bug**
A clear and concise description of what the bug is.

During catkin_make, the build cannot complete due to an error on libacquilib.so. verified spinview launches and connectes to camera

**Error Messages**
Linking CXX shared library /home/[...]/spinnaker_ws/devel/lib/libacquilib.so
/usr/bin/ld: cannot find -lSpinnaker
collect2: error: ld returned 1 exit status
spinnaker_camera_driver/CMakeFiles/acquilib.dir/build,make:238: recipe for target '/home/[...]/spinnaker_ws/devel/lib/libacquilib.so' failed

vik748, 2020-10-23:
Sounds like spinnaker might not be installed system-wide correctly. What happens if you open a fresh terminal and run `Acquisition` ?

jbwurts, 2020-10-26:
From what I can tell, the Acquisiton runs correctly. The saved images are accurate, albeit dark.

Application build date: Sep 30 2020 08:56:20

Spinnaker library version: 2.2.0.48

Number of cameras detected: 1

Running example for camera 0...

*** DEVICE INFORMATION ***

DeviceID : USB\VID_1E10&SRL_20129759\&2&3&1
DeviceSerialNumber : 20129759
DeviceVendorName : FLIR
DeviceModelName : Firefly FFY-U3-16S2C
DeviceType : USB3Vision
DeviceDisplayName : FLIR
DeviceAccessStatus : OpenReadWrite
DeviceVersion : 2002.0.17.0
DeviceDriverVersion : none : 0.0.0.0
DeviceUserID : 
DeviceIsUpdater : 0
DeviceInstanceId : USB\VID_1E10&SRL_20129759\&2&3&1
DeviceLocation : 
DeviceCurrentSpeed : SuperSpeed
GUIXMLLocation : Device
GUIXMLPath : Input.xml
GenICamXMLLocation : Device
GenICamXMLPath : 
DeviceU3VProtocol : 1

*** IMAGE ACQUISITION ***

Acquisition mode set to continuous...
Acquiring images...
Device serial number retrieved as 20129759...

Grabbed image 0, width = 1440, height = 1080
Image saved at Acquisition-20129759-0.jpg

Grabbed image 1, width = 1440, height = 1080
Image saved at Acquisition-20129759-1.jpg

Grabbed image 2, width = 1440, height = 1080
Image saved at Acquisition-20129759-2.jpg

Grabbed image 3, width = 1440, height = 1080
Image saved at Acquisition-20129759-3.jpg

Grabbed image 4, width = 1440, height = 1080
Image saved at Acquisition-20129759-4.jpg

Grabbed image 5, width = 1440, height = 1080
Image saved at Acquisition-20129759-5.jpg

Grabbed image 6, width = 1440, height = 1080
Image saved at Acquisition-20129759-6.jpg

Grabbed image 7, width = 1440, height = 1080
Image saved at Acquisition-20129759-7.jpg

Grabbed image 8, width = 1440, height = 1080
Image saved at Acquisition-20129759-8.jpg

Grabbed image 9, width = 1440, height = 1080
Image saved at Acquisition-20129759-9.jpg

Camera 0 example complete...

Done! Press Enter to exit...

shahvi, 2020-10-26:
Not sure what is going on, I haven't seen an issue like this before. 
The error still seems like it is related to the fact that catkin_make is not finding the Spinnaker SDK.  Did you install it to the standard location system-wide? Did you run `sudo sh install_spinnaker.sh` during installation?

jbwurts, 2020-10-26:
I ran `sudo sh install_spinnaker.sh` and it compiled properly.

however, I now concur that I believe the issue is with the Spinnaker SDK. Using Spinnaker-2.2.0, it appears the libacquilib.so is not created, thus catkin_make cannot link to Spinnaker SDK.
I was able to correct this by going to FLIR's download archive to download and install Spinnaker-2.0.0. Using version 2.0.0, libacquilib.so is linked and able to compile. I was able to go through the next few steps in the spinnaker_sdk_camera_driver and view the camera over ROS.

Thank you!

closed 2020-10-26 by jbwurts