neufieldrobotics/spinnaker_sdk_camera_driver #103: catkin_make

yanghan1ros, 2021-01-14:
/home/a-y-x/spinnaker_ws/src/spinnaker_camera_driver/include/spinnaker_sdk_camera_driver/std_include.h:5:10: fatal error: Spinnaker.h: 没有那个文件或目录
 #include "Spinnaker.h"
          ^~~~~~~~~~~~~
In file included from /home/a-y-x/spinnaker_ws/src/spinnaker_camera_driver/include/spinnaker_sdk_camera_driver/capture.h:4:0,
                 from /home/a-y-x/spinnaker_ws/src/spinnaker_camera_driver/src/capture.cpp:1:
/home/a-y-x/spinnaker_ws/src/spinnaker_camera_driver/include/spinnaker_sdk_camera_driver/std_include.h:5:10: fatal error: Spinnaker.h: 没有那个文件或目录
 #include "Spinnaker.h"
          ^~~~~~~~~~~~~

samsdolphin, 2021-01-14:
Yeah, I have the same error.

shahvi, 2021-01-14:
Please fill out a full bug report.

yanghan1ros, 2021-01-15:
> Yeah, I have the same error.

you should install the relevant driver.    SpinView_QT

rgov, 2021-01-26:
I filed this properly here: #105.

PushyamiKaveti, 2021-02-02:
closing this issue as it is being addressed in #105

closed 2021-02-02 by PushyamiKaveti