neufieldrobotics/spinnaker_sdk_camera_driver #105: Path to Spinnaker SDK is hardcoded, Spinnaker.h not found

rgov, 2021-01-26:
**System Description**
Camera Name and Model:  FLIR Blackfly S BFS-U3-50S5C-C
Operating System: Ubuntu 18.04 "Bionic"
Spinnaker Version: 2.3.0.77
ROS Version:  Melodic

***Computer details***
Processor:  Jetson TX2
RAM:  4 GB?

**Do the cameras work with SpinView?**

They work with the Python examples provided with the Spinnaker SDK.

**Describe the bug**

I am building from the `Feature/spinnaker_update` branch, using `catkin`.

When I compile the project, it is unable to find a key Spinnaker SDK header file. But the file exists, and is located at `/opt/spinnaker/include/Spinnaker.h`

The problem stems from the fact that `CMakeLists.txt` hardcodes the path to the Spinnaker SDK, rather than allowing CMake to try to discover the library.

```make
set(SPINNAKER_INCLUDE_DIR "/usr/include/spinnaker" CACHE PATH "Spinnaker Include Directory")
set(SPINNAKER_LIB_DIR "/usr/lib" CACHE PATH "Spinnaker Libs Directory")
```

You could invoke:

```
catkin build spinnaker_sdk_camera_driver --cmake-args -DSPINNAKER_INCLUDE_DIR=/opt/spinnaker/include -DSPINNAKER_LIB_DIR=/opt/spinnaker/lib
```

But the `CMakeLists.txt` should be changed to use [`find_package`](https://cmake.org/cmake/help/latest/command/find_package.html) to determine the path, rather than expect that it exists at the hardcoded location.

**Error Messages**

```
spinnaker_sdk_camera_driver/include/spinnaker_sdk_camera_driver/std_include.h:5:10: fatal error: Spinnaker.h: No such file or directory
 #include "Spinnaker.h"
```

shahvi, 2021-01-26:
@rgov thanks for pointing this out. It looks like spinnaker made this change in one of the recent updates where they started to install the driver in `/opt` instead of `/usr`.  We will test and update the master as per your recommendations.
Cheers.

PushyamiKaveti, 2021-02-02:
@rgov  this is fixed in the hotfix/spinnaker_cmake branch

closed 2021-02-04 by shahvi