neufieldrobotics/spinnaker_sdk_camera_driver #108: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'

dostyaaa, 2021-02-01:
**System Description**
Camera Name and Model: BFS-U3-28S5C-C
Operating System: Ubuntu 18.04
Spinnaker Version: 2.0.0.147
ROS Version: Melodic 1.14.10

***Computer details***
Processor: Intel i7-6600u 2.6 GHz
RAM: 16GB

***Multiple Cameras Only***
How is the triggering setup? Master/Slave or all slaves or all software triggered?

For USB 3.0 cameras:
On same USB 3.0 bus or different buses?
Is the USB 3.0 hub externally powered? 

For Ethernet cameras: 
How busy is the network, does the system work in grayscale, at a lower resoultion (binning) and framerate?

**Do the cameras work with SpinView?**
Yes, working perfectly fine

**Describe the bug**
Built sdk_driver with catkin without problems. Once i tried to launch acquisition.launch got the following error :

[FATAL] [1612185131.933308616]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 24036, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=ros/log/17e047e4-648f-11eb-863c-a44cc895af49/vision_nodelet_manager-2.log].
log file: ros/log/17e047e4-648f-11eb-863c-a44cc895af49/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 24038, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=ros/log/17e047e4-648f-11eb-863c-a44cc895af49/acquisition_node-3.log].
log file: ros/log/17e047e4-648f-11eb-863c-a44cc895af49/acquisition_node-3*.log

Tried both dev and master branch, same problem. Modify cam ids in the yaml file, same problem. Tried to upgrade sdk spinnaker but works with Ubuntu 20.04. Running out of ideas...

**Error Messages**
Full message:

`started roslaunch server http://etscan.local:36583/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [20285628]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1536
 * /acquisition_node/image_width: 2048
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 20285628
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [24019]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 17e047e4-648f-11eb-863c-a44cc895af49
process[rosout-1]: started with pid [24030]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [24036]
process[acquisition_node-3]: started with pid [24038]
[ INFO] [1612185131.671726176]: Initializing nodelet with 4 worker threads.
[ INFO] [1612185131.789179726]: Initializing nodelet
[ INFO] [1612185131.790039899]: [ OK ] USB memory: 1000 MB
[ INFO] [1612185131.790087765]: *** PARAMETER SETTINGS ***
[ INFO] [1612185131.790108465]: ** Date = 20210201
[ INFO] [1612185131.790670894]:   Save path set via parameter to: /home/etscan
[ INFO] [1612185131.790700317]:   Camera IDs:
[ INFO] [1612185131.791014838]:   Camera Aliases:
[FATAL] [1612185131.933308616]: Failed to load nodelet '/acquisition_node` of type `acquisition/capture_nodelet` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 24036, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=ros/log/17e047e4-648f-11eb-863c-a44cc895af49/vision_nodelet_manager-2.log].
log file: ros/log/17e047e4-648f-11eb-863c-a44cc895af49/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 24038, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/capture_nodelet vision_nodelet_manager __name:=acquisition_node __log:=ros/log/17e047e4-648f-11eb-863c-a44cc895af49/acquisition_node-3.log].
log file: ros/log/17e047e4-648f-11eb-863c-a44cc895af49/acquisition_node-3*.log
`

**Screenshots**
If applicable, add screenshots to help explain your problem.

Best regards !

dostyaaa, 2021-02-01:
Tried with rosrun:

[ INFO] [1612187576.893107793]: [ OK ] USB memory: 1000 MB
[ INFO] [1612187576.893972776]: *** PARAMETER SETTINGS ***
[ INFO] [1612187576.894008517]: ** Date = 20210201
[ WARN] [1612187576.894388261]:   Save path not provided, data will be saved to: /home/etscan
[ INFO] [1612187576.894421025]:   Camera IDs:
[ INFO] [1612187576.894712589]:   No camera aliases provided. Camera IDs will be used as names.
[ WARN] [1612187576.894996293]:   'utstamps' Parameter not set, using default behavior utstamps=false
[ WARN] [1612187576.895259437]:   'color' Parameter not set, using default behavior color=false
[ WARN] [1612187576.895515756]:   'to_ros' Parameter not set, using default behavior to_ros=false
[ WARN] [1612187576.895770646]:   'live' Parameter not set, using default behavior live=false
[ WARN] [1612187576.896022319]:   'live_grid' Parameter not set, using default behavior live_grid=true
[ WARN] [1612187576.896287614]:   'max_rate_save' Parameter not set, using default behavior max_rate_save=false
[ WARN] [1612187576.896532737]:   'time' Parameter not set, using default behavior time=false
[ WARN] [1612187576.896776853]:   'skip' Parameter not set, using default behavior: skip=20
[ WARN] [1612187576.897044228]:   'delay' Parameter not set, using default behavior: delay=1.000000
[ WARN] [1612187576.897302483]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1612187576.897541216]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1612187576.897787219]:   target_grey_value set to: 50.0
[ WARN] [1612187576.898023068]:   'binning' Parameter not set, using default behavior: Binning = 1
[ WARN] [1612187576.898254986]:   'soft_framerate' Parameter not set, using default behavior: No Software Rate Control 
[ WARN] [1612187576.898485811]:   'save' Parameter not set, using default behavior save=0
[ WARN] [1612187576.899322688]:   Camera coeffs not provided correctly, camera info messges intrinsics and distortion coeffs will be published with zeros.
[ INFO] [1612187576.899345569]: Creating system instance...
[ INFO] [1612187576.902054696]: Retreiving list of cameras...
[ INFO] [1612187580.301916718]: Numer of cameras found: 1
[ INFO] [1612187580.302106842]:  Cameras connected: 1
[ INFO] [1612187580.302329488]:   -013588BC
[ INFO] [1612187580.320076369]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1612187580.326351227]: *** FLUSH SEQUENCE ***
[ INFO] [1612187580.326522953]: Initializing cameras...
[ INFO] [1612187580.327205347]: Deinitializing cameras...
[ INFO] [1612187580.327368240]: All cameras deinitialized.
[ INFO] [1612187582.327833400]: Initializing cameras...
[ INFO] [1612187582.328109558]: *** ACQUISITION ***
Core dumped (Segmentation fault)

shahvi, 2021-02-01:
@dostyaaa One issue I noticed is that  `Camera IDs` is not being read.  Are you sure, you have formatted the yaml file correctly? In general I don't expect `rosrun` to work since our node needs a lot of parameters set correctly, which is done with the roslaunch files. All our testing is done with Ubuntu 16.04 and ROS kinetic, but others have reported being able to run with 18.04 and Ros Melodic.  Can you check your launch file to see which `yaml` file it is calling and paste the `yaml` file here.

dostyaaa, 2021-02-01:
Thanks for your answer shahvi. Launch file calls the defaut test_params.yaml.

cam_ids: -20285628
cam_aliases:
- cam0
master_cam: -20285628
skip: 20
delay: 1.0

# Assign all the follwing via launch file to prevent confusion and conflict

#save_path: ~/projects/data
#save_type: .bmp #binary or .tiff or .bmp
#binning: 1 # going from 2 to 1 requires cameras to be unplugged and replugged
#color: false
#frames: 50
#soft_framerate: 20 # this frame rate reflects to the software frame rate set using ros::rate
#exp: 997
#to_ros: true  #When to_ros is not selected, but live is selected, pressing 'space' exports single image to ROS

#Camera info message details
distortion_model: plumb_bob
image_height: 1536
image_width: 2048
distortion_coeffs:
- [-0.021141875266089592, -0.3733872931278025, 2.385982550579459, 3.2824571732099725]

#specified as [fx  0 cx 0 fy cy 0  0  1]
intrinsic_coeffs:
- [1886.9232141485886, 0.0, 604.7214878709341, 0.0, 1886.6668765711668, 493.47726714719823, 0.0, 0.0, 1.0]

rectification_coeffs:
- [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]

projection_coeffs:
- [913.700317, 0.000000, 953.448302, 0.000000, 0.000000, 1063.296631, 777.871993, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]

shahvi, 2021-02-01:
I think you have the yaml formatting incorrect.  Every space and line break matters.  Take a look at the example file.  You cam_ids is being read as "-20285628" instead of a list.

dostyaaa, 2021-02-01:
[image]

[image]

shahvi, 2021-02-01:
Reading the yaml is done outside the node. Roslaunch reads the yaml file and loads the parameters to the ROS parameter server. Our node reads it from the parameter server. You could check if the file is being read into the ROS parameter server by running rosparam list, you might have to start a separate roscore before running the launch file so that the ros master doesn't close when the node seg faults.

dostyaaa, 2021-02-01:
Seems to be read with the correct id:

[image]

robwaat, 2021-10-13:
For anyone else with this issue, the problem seems to have been a change in the ROS API between kinetic and melodic that prevents execution of code within `ROS_ASSERT()` macro calls.

These are used frequently in setting the parameters from `params/test_params.yaml`, preventing those parameters from being read into the nodelet.

A fix can be found in pull request #141 .

[More Info](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/issues/80#issuecomment-802862721)