neufieldrobotics/spinnaker_sdk_camera_driver #122: Failed to load nodelet

fys445089968, 2021-03-31:
1617176618.356471112  Node Startup
1617176618.447406042 ERROR /vision_nodelet_manager [/tmp/binarydeb/ros-melodic-nodelet-1.9.16/src/loader.cpp:301(Loader::load)] [topics: /rosout] The error before refreshing the cache was: Failed to load library /home/davy/14-spinnaker_ws/devel/lib//libacquilib.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/spinnaker/lib/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/local/lib/libopencv_highgui.so.3.4))
But I can find libacquilib.so under /home/davy/14-spinnaker_ws/devel/lib/, and my Qt's version is 5.12.

shahvi, 2021-03-31:
Please provide all the details in the issue template.

fys445089968, 2021-04-04:
> Please provide all the details in the issue template.

roslaunch spinnaker_sdk_camera_driver acquisition.launch
... logging to /home/davy/.ros/log/d58d7114-953a-11eb-9272-5405db7a4f45/roslaunch-davy-Lenovo-Legion-R7000P2020H-21758.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://davy-Lenovo-Legion-R7000P2020H:41359/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [17197559]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/external_trigger: False
 * /acquisition_node/flip_horizontal: [False]
 * /acquisition_node/flip_vertical: [False]
 * /acquisition_node/frames: 3400
 * /acquisition_node/gain: 0
 * /acquisition_node/image_height: 1080
 * /acquisition_node/image_width: 1440
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 17197559
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/region_of_interest/height: 0
 * /acquisition_node/region_of_interest/width: 0
 * /acquisition_node/region_of_interest/x_offset: 0
 * /acquisition_node/region_of_interest/y_offset: 0
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/tf_prefix: 
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [21768]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d58d7114-953a-11eb-9272-5405db7a4f45
process[rosout-1]: started with pid [21779]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [21795]
process[acquisition_node-3]: started with pid [21796]
[ INFO] [1617536549.680422198]: Initializing nodelet with 16 worker threads.
[ERROR] [1617536549.760465756]: Failed to load nodelet [/acquisition_node] of type [acquisition/Capture] even after refreshing the cache: Failed to load library /home/davy/14-blackfly_ws/devel/lib//libacquilib.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/spinnaker/lib/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/local/lib/libopencv_highgui.so.3.4))
[ERROR] [1617536549.760568136]: The error before refreshing the cache was: Failed to load library /home/davy/14-blackfly_ws/devel/lib//libacquilib.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /opt/spinnaker/lib/libQt5Core.so.5: version `Qt_5.9' not found (required by /usr/local/lib/libopencv_highgui.so.3.4))
[FATAL] [1617536549.760798458]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
[acquisition_node-3] process has died [pid 21796, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/home/davy/.ros/log/d58d7114-953a-11eb-9272-5405db7a4f45/acquisition_node-3.log].
log file: /home/davy/.ros/log/d58d7114-953a-11eb-9272-5405db7a4f45/acquisition_node-3*.log
^C[vision_nodelet_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
please help me

shahvi, 2021-04-04:
@jpsnir @PushyamiKaveti is this the issue you were talking about?