neufieldrobotics/spinnaker_sdk_camera_driver #123: Does the camera support PTP time synchronization?

fys445089968, 2021-04-06:
Does the camera support PTP time synchronization? Does it support external hardware triggering?Because I want to sync my lidar and this camera.

shahvi, 2021-04-06:
I am not sure about PTP sync, but the camera does support external hardware triggering.  You can setup the camera as a `Slave` as described in the [README](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver#readme) and use the `external_trigger` flag. However, the timestamp on the message will still be based on when the image is received to the computer.  Getting the correct timestamp is  more complicated.

dimaxano, 2021-04-06:
As a possible solution to obtaining correct timestamp, we can get camera time right before requesting the Image, compute the difference between camera time and the host (your computer) time. 
After the Image is received, we correct the timestamp from ImagePtr by the computed difference.

To obtain camera time we should execute [this](http://softwareservices.flir.com/Spinnaker/latest/class_spinnaker_1_1_camera.html#a9ffdfb6afd59669d5d1f8cb110581bc5) command, and then take value from [here](http://softwareservices.flir.com/Spinnaker/latest/class_spinnaker_1_1_camera.html#ac646b1ac33ee56b26c94735abe8c5f65)

shahvi, 2021-04-06:
This still would leave a lot of variability. You can try it and test the std dev between the image msgs. 
 The most reliable way would be to have a node from the sensor that is triggering the camera publish a trigger message with the time. The camera driver subscribes to this message and uses the trigger time to stamp the image message.  If trigger msgs package is available then Catkin make will build this functionality.

closed 2021-06-14 by vik748