neufieldrobotics/spinnaker_sdk_camera_driver #124: Camera calibration not work.

Keddie-yang, 2021-04-06:
System Description
Camera Name and Model: BFS-U3-16S2M-CS
Operating System: Ubuntu 18.04 Bionic
Spinnaker Version: 2.2.0.48
ROS Version: Melodic

Computer details
Processor: Xavier AGX
RAM: 32G

Do the cameras work with SpinView?
Yes.

problem describe: 

I use ROS camera_calibration [http://wiki.ros.org/camera_calibration] to find distortion, intrinsic and projection coefficient, then put into the test_params.yaml but output image  didn't make any changes.
Does anyone know how to make the camera calibration correctly ?

shahvi, 2021-04-06:
@Keddie-yang the camera driver does not correct for the camera distortion or intrinsic parameters. It just publishes that information as [CameraInfo](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/CameraInfo.html) messages. The node which uses the images can read the CameraInfo messages and correct for distortion etc.

Keddie-yang, 2021-04-06:
OK, I get it. Thank you.

closed 2021-06-14 by vik748