neufieldrobotics/spinnaker_sdk_camera_driver #125: Spinnaker SDK reinstall needed for each launch.

LincolnK, 2021-04-07:
**System Description**
Camera Name and Model: Blackfly S 1.3 MP
Operating System: Ubuntu 18.04
Spinnaker Version: spinnaker-2.3.0.77
ROS Version: melodic

***Computer details***
Processor: Intel Core i7-4720HQ CPU @ 2.60GHz
RAM: 16 GB

For USB 3.0 cameras:
I only use a single camera that is connected directly to one of my computers USB 3.0 Ports.

**Do the cameras work with SpinView?**
Yes

**Describe the bug**
I'm having an issue where attempting to launch the ROS Node fails. I've found that reinstalling Spinnaker SDK allows the ROS Node to properly launch once, however after the first node is closed, this issue will occur every time the node is launched. I've tried restarting my computer, and unplugging the camera, but the only thing that fixes is it uninstalling and reinstalling the Spinnaker SDK. I believe the issue to be at [capture.cpp Line 143](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/master/src/capture.cpp#L143), where this line causes something to go wrong. ROS doesn't log what goes wrong, and putting the line inside of a try catch block does not actually catch the error.

**Error Messages**
```
started roslaunch server http://lincoln-GS70-2QE:37889/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [18080149]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/external_trigger: False
 * /acquisition_node/flip_horizontal: [False]
 * /acquisition_node/flip_vertical: [False]
 * /acquisition_node/frames: 3400
 * /acquisition_node/gain: 0
 * /acquisition_node/image_height: 1080
 * /acquisition_node/image_width: 1440
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 18080149
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/region_of_interest/height: 0
 * /acquisition_node/region_of_interest/width: 0
 * /acquisition_node/region_of_interest/x_offset: 0
 * /acquisition_node/region_of_interest/y_offset: 0
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/tf_prefix: 
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [2640]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d9a9127c-97b9-11eb-86d6-acd1b8058dd7
process[rosout-1]: started with pid [2654]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [2657]
process[acquisition_node-3]: started with pid [2658]
[ INFO] [1617811004.725646594]: Initializing nodelet with 8 worker threads.
[ INFO] [1617811005.148638628]: Initializing nodelet
[ INFO] [1617811005.150921963]: [ OK ] USB memory: 1000 MB
[ INFO] [1617811005.150979213]: *** PARAMETER SETTINGS ***
[ INFO] [1617811005.150995037]: ** Date = 20210407
[ INFO] [1617811005.151570514]:   Save path set via parameter to: /home/lincoln
[ INFO] [1617811005.151613188]:   Camera IDs:
[ INFO] [1617811005.151965051]:     18080149
[ INFO] [1617811005.152231910]:   Camera Aliases:
[ INFO] [1617811005.152271860]:     18080149 >> cam0
[ INFO] [1617811005.152554830]:   External trigger: false
[ INFO] [1617811005.153011442]:   Unique time stamps for each camera: false
[ INFO] [1617811005.153260511]:   color set to: false
[ INFO] [1617811005.153547106]:   18080149 flip_horizontal 0
[ INFO] [1617811005.153805996]:   18080149 flip_vertical 0
[ INFO] [1617811005.154051995]:   Exporting images to ROS: true
[ INFO] [1617811005.154331249]:   Showing live images setting: false
[ INFO] [1617811005.154593622]:   Showing grid-style live images setting: false
[ INFO] [1617811005.154856956]:   Max Rate Save Mode: false
[ INFO] [1617811005.155106528]:   Displaying timing details: false
[ INFO] [1617811005.155349364]:   No. of images to skip set to: 20
[ INFO] [1617811005.155608939]:   Init sleep delays set to : 1.00 sec
[ WARN] [1617811005.155850526]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1617811005.156094446]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1617811005.156338075]:   'gain' Parameter was zero or negative, using Auto gain based on target grey value
[ INFO] [1617811005.156578115]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1617811005.156811612]:   Binning set to: 1
[ INFO] [1617811005.157031977]:   Using Software rate control, rate set to: 30
[ INFO] [1617811005.157265865]:   Saving images set to: 0
[ INFO] [1617811005.157891740]:   tf_prefix set to: 
[ INFO] [1617811005.159071217]:   Region of Interest set to width: 0	height: 0	offset_x: 0 offset_y: 0
[ INFO] [1617811005.159387416]:   Camera Intrinsic Paramters:
[ INFO] [1617811005.159646436]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1617811005.159962948]:   Camera Distortion Paramters:
[ INFO] [1617811005.160005321]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1617811005.160322781]:   Camera Rectification Paramters:
[ INFO] [1617811005.160363153]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1617811005.160645242]:   Camera Projection Paramters:
[ INFO] [1617811005.160686440]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1617811005.160703819]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1617811005.160747859]: *** SYSTEM INFORMATION ***
[ INFO] [1617811005.160764157]: Creating system instance...
[ INFO] [1617811005.160786439]: spinnaker_sdk_camera_driver package version: 1.1.0
[FATAL] [1617811005.309020455]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
[vision_nodelet_manager-2] process has died [pid 2657, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/lincoln/.ros/log/d9a9127c-97b9-11eb-86d6-acd1b8058dd7/vision_nodelet_manager-2.log].
log file: /home/lincoln/.ros/log/d9a9127c-97b9-11eb-86d6-acd1b8058dd7/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 2658, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/home/lincoln/.ros/log/d9a9127c-97b9-11eb-86d6-acd1b8058dd7/acquisition_node-3.log].
log file: /home/lincoln/.ros/log/d9a9127c-97b9-11eb-86d6-acd1b8058dd7/acquisition_node-3*.log
```

calum-blair-archangel, 2021-05-25:
I had the same issue. Not sure what the underlying cause is yet, but the item causing the problem is the config files inside `~/.config/spinnaker`. I deleted the `GenTLIfGevReg.ini , GenTLIfReg.ini,  GenTLSysReg.ini` and   `GenTLUsbReg.ini` files after each run and no longer needed to reinstall the SDK.

vik748, 2021-06-14:
@lincolnkinley are you still facing this issue?  What version of spinnaker are you using?

vik748, 2021-06-14:
@lincolnkinley I just read the bug report, can you try with a spinnaker 2.2.0.48

lincolnkinley, 2021-07-14:
I updated to the most recent version of the spinnaker driver and the issue was fixed. It must have been something on their end.

closed 2021-07-14 by lincolnkinley