neufieldrobotics/spinnaker_sdk_camera_driver #126: how to grab image in my project using this project?

HuangDihe, 2021-04-11:
Hello, I'm used to use dataset and not familiar with ros nodelet.  Could you tell me how to grab image in my project from Filr camera using this project?  Such as, put this project in my project and include sth.h? and run subscriber?  I would deeply appreciate your answer.

vik748, 2021-04-12:
I would suggest looking through the [ROS Tutorials](http://wiki.ros.org/ROS/Tutorials) and the README for this driver. You can add this driver to your ROS workspace and run it using the launch files. This node will publish the image as a ROS topic. To check if image is being published you can run `rqt_image_view`. In this case `rqt_image_view` acts as a subscriber, you basically need to create a subscriber like that. http://wiki.ros.org/image_transport/Tutorials/SubscribingToImages

HuangDihe, 2021-04-12:
hi，i just don‘t know the name of topic to subscribe

------------------ Original ------------------
From: vik748 ***@***.***&gt;
Date: Mon,Apr 12,2021 7:10 PM
To: neufieldrobotics/spinnaker_sdk_camera_driver ***@***.***&gt;
Cc: HuangDihe ***@***.***&gt;, Author ***@***.***&gt;
Subject: Re: [neufieldrobotics/spinnaker_sdk_camera_driver] how to grab image in my project using this project? (#126)

closed 2021-06-14 by vik748