neufieldrobotics/spinnaker_sdk_camera_driver #127: Error: unable to set BinningHorizontal to 1 (ptr retrieval)

nicewiz, 2021-04-14:
**_System Description_**
Camera Name and Model: Grasshopper3 GS3-U3-123S6C
Operating System: Ubuntu 16.04
Spinnaker Version: 2.0.0.147
ROS Version: kinetic 1.12.17

**Computer details**
Processor:Intel i7-7500U 2.70GHz
RAM: 8GB

**Multiple Cameras Only**
How is the triggering setup? Master/Slave or all slaves or all software triggered?

For USB 3.0 cameras:
On same USB 3.0 bus or different buses?
Is the USB 3.0 hub externally powered?

For Ethernet cameras: 
How busy is the network, does the system work in grayscale, at a lower resoultion (binning) and framerate?

**Do the cameras work with SpinView?**
Yes, perfectly.

**Describe the bug**
I have installed the spinnaker_sdk_camera_driver as described in the readme file, but run into following error(s) when launching the acquisition.launch file:
```
[FATAL] [1618388473.689578333]: Unable to set BinningHorizontal to 1 (ptr retrieval). Aborting...
[FATAL] [1618388473.689773938]: Error: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 77) [-2006]
[ WARN] [1618388473.689795455]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1618388473.890316354]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'TriggerSoftware' while calling 'TriggerSoftware.Execute()' (file 'CommandT.h', line 61) [-2006]
[acquisition_node-2] process has died [pid 4387, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/home/chandler/.ros/log/2bebec98-9cf7-11eb-8163-d0577bf7c33a/acquisition_node-2.log].
log file: /home/chandler/.ros/log/2bebec98-9cf7-11eb-8163-d0577bf7c33a/acquisition_node-2*.log
[vision_nodelet_manager-1] process has died [pid 4386, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/chandler/.ros/log/2bebec98-9cf7-11eb-8163-d0577bf7c33a/vision_nodelet_manager-1.log].
log file: /home/chandler/.ros/log/2bebec98-9cf7-11eb-8163-d0577bf7c33a/vision_nodelet_manager-1*.log
```
both cameras are recognized and stated as "connected" in the output.I have set Binning to 0 or other values, and color mode is deactivated and still got the following error.

**Error Messages**
```
SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [17392138]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/external_trigger: False
 * /acquisition_node/flip_horizontal: [False]
 * /acquisition_node/flip_vertical: [False]
 * /acquisition_node/frames: 3400
 * /acquisition_node/gain: 0
 * /acquisition_node/image_height: 1080
 * /acquisition_node/image_width: 1440
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 17392138
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/region_of_interest/height: 0
 * /acquisition_node/region_of_interest/width: 0
 * /acquisition_node/region_of_interest/x_offset: 0
[... 72 lines cut ...]
[ INFO] [1618388471.492295405]: *** FLUSH SEQUENCE ***
[ INFO] [1618388471.492324067]: Initializing cameras...
[ INFO] [1618388471.583439808]: Deinitializing cameras...
[ INFO] [1618388471.587834859]: All cameras deinitialized.
[ INFO] [1618388473.588079004]: Initializing cameras...
[FATAL] [1618388473.689578333]: Unable to set BinningHorizontal to 1 (ptr retrieval). Aborting...
[FATAL] [1618388473.689773938]: Error: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 77) [-2006]
[ WARN] [1618388473.689795455]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1618388473.890316354]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'TriggerSoftware' while calling 'TriggerSoftware.Execute()' (file 'CommandT.h', line 61) [-2006]
[acquisition_node-2] process has died [pid 4387, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/home/chandler/.ros/log/2bebec98-9cf7-11eb-8163-d0577bf7c33a/acquisition_node-2.log].
log file: /home/chandler/.ros/log/2bebec98-9cf7-11eb-8163-d0577bf7c33a/acquisition_node-2*.log
[vision_nodelet_manager-1] process has died [pid 4386, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/chandler/.ros/log/2bebec98-9cf7-11eb-8163-d0577bf7c33a/vision_nodelet_manager-1.log].
log file: /home/chandler/.ros/log/2bebec98-9cf7-11eb-8163-d0577bf7c33a/vision_nodelet_manager-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

Best Regards!

vik748, 2021-04-14:
Can you try commenting out the two lines where the binning is being set on the camera - https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/f6ab2909ba2a4e250d7838331c83d4b588badb0f/src/capture.cpp#L686

nicewiz, 2021-04-15:
We tried commenting out all the lines where fatal error occurred and it works out.
Then we went and checked the camera settings in spinview and found that some valuables like BinningHorizontal might have been locked in our camera so the function SetIntValue() can't work. Also could be other reasons but not sure.
Thanks for your advice!
BTW how can we save the image data from camera in ros?

vik748, 2021-04-15:
In ROS you can just use rosbag record to save all the ROS messages.

Looks like the right way to address this would be to put https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/f6ab2909ba2a4e250d7838331c83d4b588badb0f/src/camera.cpp#L173
line under `else`.  That should throw an error but let the code continue.

nicewiz, 2021-05-21:
we find it work if commenting out those lines throwing errors
but in this way we cannot set the camera to color mode nor we can set acquisition rate to other values
put `ptr->SetValue(val);` line under 'else' but  cannot set color to rgb(should not work)
still trying other ways to solve it

shahvi, 2021-05-21:
@nicewiz you can use SpinView to figure out exactly which variables you need to play with to get your desired settings and then try to replicate that in the driver.

nicewiz, 2021-05-23:
@shahvi This leads to a former question -> we find some variables in spinview are locked(cannot be changed or not in a correct operation?)
So are there any relations between the error aforementioned on the top of this issue and the following code?
` if (!IsAvailable(ptr) || !IsWritable(ptr))        ROS_FATAL_STREAM("Unable to set " << setting << " to " << val << " (ptr retrieval). Aborting...");
`

vik748, 2021-05-23:
@nicewiz yes, if something is locked in SpinView, most likely we would not be able to change it from the driver.

nicewiz, 2021-05-23:
Got it!
Are there any methods to unlock them in SpinView?
There are lots of important variables we wanna reset.

shahvi, 2021-05-23:
Usually I have seen that the variables that are  locked or grayed out are not supported by the camera hardware. Eg. If the camera is Grayslake then color option is grayed out. 

What variables are you trying to change?

nicewiz, 2021-05-25:
1. color as our camera Grasshopper3 GS3-U3-123S6C support the color mode
2. acquisition frame rate to 20 or more as the rate the driver grab the image with fps 5
and some other variables.

Strangely we cannot set color in driver though color mode supported while acquisition frame rate is locked

babakbch, 2025-05-07:
Has this been solved? I have the same issue with Grasshopper3 GS3-U3-123S6C , running a ros kinetic docker image I get : 

root@ubuntu:~/catkin_ws# roslaunch spinnaker_sdk_camera_driver acquisition.launch 
... logging to /root/.ros/log/fb9077a0-2b4d-11f0-8606-00e04c907eac/roslaunch-ubuntu-9536.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:40511/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [21236124]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/external_trigger: False
 * /acquisition_node/flip_horizontal: [False]
 * /acquisition_node/flip_vertical: [False]
 * /acquisition_node/frames: 3400
 * /acquisition_node/gain: 0
 * /acquisition_node/image_height: 1080
 * /acquisition_node/image_width: 1440
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 21236124
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/region_of_interest/height: 0
 * /acquisition_node/region_of_interest/width: 0
 * /acquisition_node/region_of_interest/x_offset: 0
 * /acquisition_node/region_of_interest/y_offset: 0
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/tf_prefix: 
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [9546]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fb9077a0-2b4d-11f0-8606-00e04c907eac
process[rosout-1]: started with pid [9559]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [9576]
[ INFO] [1746627454.832216279]: Initializing nodelet with 6 worker threads.
process[acquisition_node-3]: started with pid [9577]
[ INFO] [1746627459.523948143]: Initializing nodelet
[ INFO] [1746627459.527264979]: [ OK ] USB memory: 1000 MB
[... 40 lines cut ...]
[ INFO] [1746627461.616339643]: Retreiving list of cameras...
[ INFO] [1746627463.352861638]: Numer of cameras found: 1
[ INFO] [1746627463.353021159]:  Cameras connected: 1
[ INFO] [1746627463.355647620]:   -21236124 Grasshopper3 GS3-U3-15S5C FW:v2.10.3.00 FPGA:v2.02
[ INFO] [1746627463.415303310]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1746627463.423493799]: *** FLUSH SEQUENCE ***
[ INFO] [1746627463.423695434]: Initializing cameras...
[ INFO] [1746627463.784917418]: Deinitializing cameras...
[ INFO] [1746627463.812258900]: All cameras deinitialized.
[ INFO] [1746627465.812641280]: Initializing cameras...
[FATAL] [1746627466.026598877]: Unable to set BinningHorizontal to 1 (ptr retrieval). Aborting...
[FATAL] [1746627466.029100824]: Error: Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'BinningHorizontal' while calling 'BinningHorizontal.SetValue()' (file 'IntegerT.h', line 77) [-2006]
[ WARN] [1746627466.029259642]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1746627466.186162184]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'TriggerSoftware' while calling 'TriggerSoftware.Execute()' (file 'CommandT.h', line 61) [-2006]
[acquisition_node-3] process has died [pid 9577, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/root/.ros/log/fb9077a0-2b4d-11f0-8606-00e04c907eac/acquisition_node-3.log].
log file: /root/.ros/log/fb9077a0-2b4d-11f0-8606-00e04c907eac/acquisition_node-3*.log
[vision_nodelet_manager-2] process has died [pid 9576, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/root/.ros/log/fb9077a0-2b4d-11f0-8606-00e04c907eac/vision_nodelet_manager-2.log].
log file: /root/.ros/log/fb9077a0-2b4d-11f0-8606-00e04c907eac/vision_nodelet_manager-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done